evcc-io/evcc #21077: VW: missing charging status

TheNinth7, 2025-05-05:
### Describe the bug

Since end of last week, evcc is not able to recognize our VW ID.Buzz or ID.4.

One startup, there is no immediate error, but then evcc repeatedly reports `vehicle status: missing charging status` and does not automatically recognize the connected vehicle.

When manually setting the vehicle, evcc then continues to report `Carport: vehicle soc: missing fuel status`

Also validating the vehicle in the UI configuration fails:

[image]

This tracelog covers the whole startup, the manual vehicle setting and the first `missing fuel status`.

[evcc-missing-fuel-status-trace.txt](https://github.com/user-attachments/files/20040182/evcc-missing-fuel-status-trace.txt)

I updated to the latest 0.203.5 and the issue persists.

### Steps to reproduce

1. Setup vehicle, if helps I can provide the data (VIN, VW account)
2. Start evcc

### Configuration details

```yaml
{
"cache":"15m",
"capacity":77,
"icon":"car",
"maxCurrent":16,
"minCurrent":6,
"password":"removed",
"template":"vw",
"timeout":"10s",
"title":"ID.Buzz",
"user":"removed",
"vin":"removed"
}
```

### Log details

```text
Too long, attached above
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

TobiasHuber1980, 2025-05-05:
https://github.com/evcc-io/evcc/discussions/21004

TheNinth7, 2025-05-06:
I managed to resolve the issue. It appears that VW made some changes to how charging status is handled, and these updates had to be confirmed in the VW app. Notably, the confirmation prompts did *not* appear in the web portal.

https://github.com/evcc-io/evcc/discussions/21004#discussioncomment-13047075

closed 2025-05-06 by TheNinth7