evcc-io/evcc #21078: support for zaptec go 2

tbaehler, 2025-05-05:
It would be great to have support for the zaptec go 2. 
Seems to be quite equals to the zaptec.go charger.

At least I can charge and evcc sends out commands, but phase 3 -> 1 does not work. 
The zaptec2 automatic phase change should work out of the box by just, using

POST /api/installation/{id}/update and adjusting "availableCurrent".

I will try today if I can modify the zaptec.go to work with zaptec2

andig, 2025-05-06:
How should adjusting current lead to phase switching? I‘m afraid we‘ll need api docs. You‘re also welcome to contribute the required changes!

tbaehler, 2025-05-06:
Documentation is here. https://help.zaptec.com/hc/en-001/articles/33567180538001-Dynamic-load-balancing-with-the-zaptec-API

I started to implement it yesterday and added a zaptec2.go charger...

I will test it thisi evening... lets see

func (c *Zaptec2) installationUpdate(data zaptec.UpdateInstallation) error {
	uri := fmt.Sprintf("%s/api/installations/%s/update", zaptec.ApiURL, c.installationId)

	req, err := request.New(http.MethodPost, uri, request.MarshalJSON(data), request.JSONEncoding)
	if err == nil {
		_, err = c.DoBody(req)
		c.statusG.Reset()
	}

	return err
}

andig, 2025-05-06:
Laut Doku scheint es `519 SetPhases` weiter zu geben. Warum das bei Dir nicht funktioniert ist nicht klar.

tbaehler, 2025-05-06:
Ich habe dazu noch den Support bei Zaptec angefragt.
Sie haben mir bestätigt, dass bei der Zaptec Go dies über 

POST /api/installation/{id}/update zu erfolgen hat. 

Der SetPhase Call scheint nur bei der Zaptec Pro zu funktionieren.

tbaehler, 2025-05-06:
```
func (c *Zaptec2) installationUpdate(data zaptec.UpdateInstallation) error {
	uri := fmt.Sprintf("%s/api/installation/%s/update", zaptec.ApiURL, c.installationId)

	req, err := request.New(http.MethodPost, uri, request.MarshalJSON(data), request.JSONEncoding)
	if err == nil {
		_, err = c.DoBody(req)
		c.statusG.Reset()
	}

	return err
}
```

```
func (c *Zaptec2) Phases1p3p(phases int) error {
	phaseCurrent := 16
	noPower := 0
	data := zaptec.UpdateInstallation{
		AvailableCurrentPhase1: &phaseCurrent,
		AvailableCurrentPhase2: &phaseCurrent,
		AvailableCurrentPhase3: &phaseCurrent,
	}
	if phases == 1 {
		data = zaptec.UpdateInstallation{
			AvailableCurrentPhase1: &phaseCurrent,
			AvailableCurrentPhase2: &noPower,
			AvailableCurrentPhase3: &noPower,
		}
	}

	c.log.DEBUG.Printf("zaptec2: set phases to %d, max current %d", phases, phaseCurrent)
	return c.installationUpdate(data)
}
```

This is doing the trick. 

I can switch from 3 to 1 phase all works great. 

But switching from 1 to 3 does only work if i stop charging first and restart it again. It seems the wallbox does not interrupt here... 
Do you have any ideas?

andig, 2025-05-06:
You can do the staop/start in the code as well.

andig, 2025-05-07:
Let's close here, happy to look at a PR when you're ready to.

closed 2025-05-07 by andig