evcc-io/evcc #21090: evcc Android App gets a "bad request" response when playing with a load plan

mpw96, 2025-05-06:
### Describe the bug

[image]

I played with the plan while the car was loading. Suddenly I got this, but cannot repro. Please close this ticket if the information is not sufficient.

### Steps to reproduce

I tried but I couldn't. Obviously also not with the latest nightly. So my checkmark below is a lie... :disappointed:

### Configuration details

```yaml
# unique installation id
plant: ...

interval: 30s # control cycle interval

meters:
- type: template
  template: huawei-sun2000
  id: 1
  host: 192.168.96.165
  port: 502
  usage: grid
  modbus: tcpip
  timeout: 15s
  name: netz

- type: template
  template: huawei-sun2000
  id: 1
  host: 192.168.96.165
  port: 502
  usage: pv  
  modbus: tcpip
  timeout: 15s
  name: panels-dach

- type: template
  template: huawei-sun2000
  id: 1
  host: 192.168.96.165
  port: 502
  usage: battery
  modbus: tcpip
  timeout: 15s
  name: batterien
  capacity: 10

- type: template
  template: hoymiles-opendtu
  usage: pv
  host: 192.168.96.181
  name: panels-gartenhaus

vehicles:
- type: template
  template: enyaq
  user: ...
  password: 
  vin: ...
  capacity: 77
  phases: 3
  name: wattsman3
  title: Wattsman (3 Phasen)

- type: template
  template: enyaq
  user: ...
[... 159 lines cut ...]
[lp-1  ] DEBUG 2025/05/06 15:04:45 charger status: C
[lp-1  ] DEBUG 2025/05/06 15:04:45 soc estimated: 93.16% (vehicle: 93.00%)
[lp-1  ] DEBUG 2025/05/06 15:04:45 vehicle soc: 93%
[lp-1  ] DEBUG 2025/05/06 15:04:45 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/05/06 15:04:45 vehicle range: 392km
[lp-1  ] DEBUG 2025/05/06 15:04:45 pv charge current: 22.1A = 16A + 6.15A (-1414W @ 1p)
[site  ] DEBUG 2025/05/06 15:04:45 solar forecast: accumulated 36.425kWh, produced 35.330kWh, scale 0.970
[httpd ] TRACE 2025/05/06 15:05:00 GET /api/state
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.5 (c8ceefff8)

mpw96, 2025-05-06:
App version is 1.0.3 io.evcc.android.

naltatis, 2025-05-06:
Hi @mpw96, this PR https://github.com/evcc-io/evcc/pull/21080 includes fixes regarding the new late charging setting. Since you can't provide steps to reproduce I'd say please verify that this issue still happens with the upcoming nightly. I'll close this PR for now. Feel free to comment here if it isn't fixed. Then I'll reopen.

closed 2025-05-06 by naltatis