evcc-io/evcc #21097: tariffs/zones falsche Anzeige bei zeitabhängigem Strompreis (Octopus GO)

austrogaucho, 2025-05-06:
### Describe the bug

habe bei mir einen "zeitabhängigen Strompreis" (Octopus GO) in der config, obwohl die Preise zu den einzelnen Stunden richtig berechnet werden (siehe log) zeigt die UI unter "Vorhersage" aber die falschen Werte an 

[image]

gleicher Fehler in der Charger Ansicht

[image]

obwohl die günstigsten Stunden eigentlich von 0-5Uhr laut config wären, werden aber die günstigsten Stunden von 2-7Uhr angezeigt und wohl auch für den Ladezeitraum verwendet (das hab ich jetzt noch nicht ausprobiert) 

### Steps to reproduce

1.zeitabhängigen Strompreis mit "zones" konfigurieren
2. Menü "Vorhersage" und "Einstellungen Charger" öffnen
3.
...

### Configuration details

```yaml
grid: # price using energy from the grid
  type: fixed
  price: 0.3189 # EUR/kWh
  zones:
      - days: Mo-So
        hours: 0-5
        price: 0.2188 # EUR/kWh
```

### Log details

```text
´grid:
From                   To                     Price/Cost (EUR/kWh)
2025-05-06 00:00:00    2025-05-06 01:00:00    0.219
2025-05-06 01:00:00    2025-05-06 02:00:00    0.219
2025-05-06 02:00:00    2025-05-06 03:00:00    0.219
2025-05-06 03:00:00    2025-05-06 04:00:00    0.219
2025-05-06 04:00:00    2025-05-06 05:00:00    0.219
2025-05-06 05:00:00    2025-05-06 06:00:00    0.319
2025-05-06 06:00:00    2025-05-06 07:00:00    0.319
2025-05-06 07:00:00    2025-05-06 08:00:00    0.319
2025-05-06 08:00:00    2025-05-06 09:00:00    0.319
2025-05-06 09:00:00    2025-05-06 10:00:00    0.319
2025-05-06 10:00:00    2025-05-06 11:00:00    0.319
2025-05-06 11:00:00    2025-05-06 12:00:00    0.319
2025-05-06 12:00:00    2025-05-06 13:00:00    0.319
2025-05-06 13:00:00    2025-05-06 14:00:00    0.319
2025-05-06 14:00:00    2025-05-06 15:00:00    0.319
2025-05-06 15:00:00    2025-05-06 16:00:00    0.319
2025-05-06 16:00:00    2025-05-06 17:00:00    0.319
2025-05-06 17:00:00    2025-05-06 18:00:00    0.319
2025-05-06 18:00:00    2025-05-06 19:00:00    0.319
2025-05-06 19:00:00    2025-05-06 20:00:00    0.319
2025-05-06 20:00:00    2025-05-06 21:00:00    0.319
2025-05-06 21:00:00    2025-05-06 22:00:00    0.319
2025-05-06 22:00:00    2025-05-06 23:00:00    0.319
2025-05-06 23:00:00    2025-05-07 00:00:00    0.319
2025-05-07 00:00:00    2025-05-07 01:00:00    0.219
2025-05-07 01:00:00    2025-05-07 02:00:00    0.219
2025-05-07 02:00:00    2025-05-07 03:00:00    0.219
2025-05-07 03:00:00    2025-05-07 04:00:00    0.219
2025-05-07 04:00:00    2025-05-07 05:00:00    0.219
2025-05-07 05:00:00    2025-05-07 06:00:00    0.319
2025-05-07 06:00:00    2025-05-07 07:00:00    0.319
2025-05-07 07:00:00    2025-05-07 08:00:00    0.319
2025-05-07 08:00:00    2025-05-07 09:00:00    0.319
2025-05-07 09:00:00    2025-05-07 10:00:00    0.319
2025-05-07 10:00:00    2025-05-07 11:00:00    0.319
2025-05-07 11:00:00    2025-05-07 12:00:00    0.319
2025-05-07 12:00:00    2025-05-07 13:00:00    0.319
2025-05-07 13:00:00    2025-05-07 14:00:00    0.319
2025-05-07 14:00:00    2025-05-07 15:00:00    0.319
2025-05-07 15:00:00    2025-05-07 16:00:00    0.319
2025-05-07 16:00:00    2025-05-07 17:00:00    0.319
2025-05-07 17:00:00    2025-05-07 18:00:00    0.319
2025-05-07 18:00:00    2025-05-07 19:00:00    0.319
2025-05-07 19:00:00    2025-05-07 20:00:00    0.319
2025-05-07 20:00:00    2025-05-07 21:00:00    0.319
2025-05-07 21:00:00    2025-05-07 22:00:00    0.319
2025-05-07 22:00:00    2025-05-07 23:00:00    0.319
2025-05-07 23:00:00    2025-05-08 00:00:00    0.319
´
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

´root@evcc:~# evcc -v evcc version 0.203.5 root@evcc:~# ´

naltatis, 2025-05-06:
In welcher Zeitzone ist das evcc Server konfiguriert?

austrogaucho, 2025-05-07:
danke, da lag der Hase im Pfeffer!
Ich hab im lxc-Container mit "dpkg-reconfigure tzdata" die Zeitzone neu eingestellt, das wars.
Hätte ich auch selber draufkommen können....

closed 2025-05-07 by austrogaucho