evcc-io/evcc #21098: Unable to connect with Elecq Home H1 wallbox

CodeFinder2, 2025-05-06:
### Describe the bug

The Elecq Home H1 wallbox cannot establish a connection to my evcc server. According to the manufacturer and evcc's docs, there should be support for it.

### Steps to reproduce

1. I've setup evcc within a Proxmox Debian LXT container. My Elecq Home H1 is connected to the same network and I can ping the IP the wallbox.
2. I've entered the address `ws://my-evcc-host:8887` into the "Elecq Partner" app to specify the OCPP server. I left the other two fields blank (OCID, auth key). I also tried using the IP address and appended a slash (`/`) to the URL.
3. The wallbox then says "Connecting..." but that fails. (The OCPP server address seems to be accepted and stored just fine.) According to the Elecq App, the wallbox is using the latest firmware (v0.33.2).

Also rebooted the wallbox and restarted evcc without success.

### Configuration details

```yaml
My `/etc/evcc.yaml` file:

network:
  schema: http
  host: my-evcc-host.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: <removed>

interval: 30s # control cycle interval

site:
  title: MyCharger
  meters:

chargers:
  - name: my_elecq_charger
    type: template
    template: ocpp-elecq
```

### Log details

```text
May 06 23:03:24 my-evcc-host systemd[1]: Started evcc.
May 06 23:03:24 my-evcc-host evcc[2696]: [main  ] INFO 2025/05/06 23:03:24 evcc 0.203.5
May 06 23:03:24 my-evcc-host evcc[2696]: [main  ] INFO 2025/05/06 23:03:24 using config file: /etc/evcc.yaml
May 06 23:03:24 my-evcc-host evcc[2696]: [db    ] INFO 2025/05/06 23:03:24 using sqlite database: /var/lib/evcc/evcc.db
May 06 23:03:24 my-evcc-host evcc[2696]: [main  ] INFO 2025/05/06 23:03:24 listening at :7070
May 06 23:03:24 my-evcc-host evcc[2696]: [ocpp-1] DEBUG 2025/05/06 23:03:24 waiting for chargepoint: 5m0s
May 06 23:08:24 my-evcc-host evcc[2696]: [main  ] FATAL 2025/05/06 23:08:24 charger [my_elecq_charger] cannot create charger 'my_elecq_charger': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
May 06 23:08:24 my-evcc-host evcc[2696]: [main  ] FATAL 2025/05/06 23:08:24 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.5

StefanSchoof, 2025-05-07:
The error means that your Wallbox did not connect to evcc in 5 mins. 

The recommendation is to first try it without docker (you can do that on your window or Mac). Network is more difficult in docker and so there are more error sources. If it works without docker you can go in a next step into docker.

andig, 2025-05-07:
Nothing evcc can do if it doesn't receive a connection. You'll need to troubleshoot your network.