evcc-io/evcc #21134: Cupra Born Fahrzeuglimit immer 70%

RG130, 2025-05-08:
### Describe the bug

Auf dem EVCC-Dashboard wird für mein Fahrzeug Cupra Born Bj 2024 immer ein Fahrzeuglimit von 70% angezeigt und das Fahrzeug nur bis zu diesem Punkt geladen. Im Fahrzeug selbst bzw. in der Cupra-App ist ein Limit von 80%, 90% oder auch 100% eingestellt. Am Fahrzeuglimit in EVCC ändert sich nichts.
Die Folge ist, dass ich nur bis 70% laden kann, was sehr ärgerlich ist.
Was kann ich tun, um das Fahrzeuglimit zu verändern? Kann man den Wert evtl. in der Konfiguration überschreiben?

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
vehicles:
  - name: my_car
    type: template
    template: cupra 
    user: xxx
    password: xxx
    title: Cupra Born 
    capacity: 60 
    icon: car 
    phases: 3 
    mode: PV 
    minCurrent: 6 
    maxCurrent: 16
```

### Log details

```text
[lp-1 ] DEBUG 2025/05/08 20:35:53 vehicle soc: 67%
[lp-1 ] DEBUG 2025/05/08 20:35:53 vehicle soc limit: 70%
[lp-1 ] DEBUG 2025/05/08 20:35:53 vehicle range: 276km
[lp-1 ] DEBUG 2025/05/08 20:35:53 pv charge current: min 6A > 0A (4422W @ 3p, battery: true)
[site ] DEBUG 2025/05/08 20:36:21 ----
[lp-1 ] DEBUG 2025/05/08 20:36:21 charge power: 4013W
[lp-1 ] DEBUG 2025/05/08 20:36:22 charge currents: [5.84 5.87 5.92]A
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-05-08:
Was sagt

    evcc vehicle —log trace

RG130, 2025-05-09:
Die einzigen Log-Einträge, die ich dazu sehen kann sind die oben abgebildeten. 
Zu "vehicle" werden keine Einträge erzeugt, trotz trace-Einstellung.

[lp-1 ] DEBUG 2025/05/08 20:35:53 vehicle soc limit: 70%

Das ist die Zeile, wo der "falsche" Wert 70% irgendwie vom Fahrzeug kommt oder irgendwo festgelegt wird.

andig, 2025-05-09:
Prima. Bitte wie beschrieben vorgehen.

RG130, 2025-05-09:
Was heißt wie beschrieben vorgehen? Ich stehe sozusagen am Schlauch.

andig, 2025-05-09:
https://github.com/evcc-io/evcc/issues/21134#issuecomment-2864087782

RG130, 2025-05-09:
Es tut mir leid. Ich verstehe nicht, was ich machen soll.

andig, 2025-05-10:
Du musst einfach genau den Befehl ausführen so wie er da steht

RG130, 2025-05-10:
Das funktioniert bei mir leider nicht, da ich EVCC als Addon in Home Assistant betreibe. Wenn ich den Befehl im HA-Terminal eingebe kommt eine Fehlermeldung "zsh: command not found: evcc".

andig, 2025-05-10:
Dann machs halt ohne HA. Das Problem kannst nur du lösen.

RG130, 2025-05-10:
Wie soll ich den Befehl ohne HA ausführen, wo doch HA als OS installiert ist?

RG130, 2025-05-10:
Nicht so einfach für mich, aber ich glaube ich habe es geschafft. 
Im Anhang die benötigten Daten aus dem Trace-Befehl. Ich hoffe Sie finden eine Lösung. Aktuell muss ich zum Laden bis 80% oder höher zuerst evcc abschalten.

[evcc_log_trace_vehicle.txt](https://github.com/user-attachments/files/20138321/evcc_log_trace_vehicle.txt)

andig, 2025-05-10:
Es wird dir nicht gefallen:

```
[cupra ] TRACE 2025/05/10 11:14:02 {"engines":{"primary":{"fuelType":"electric","rangeKm":298.0,"levelPct":75.0}},"services":{"charging":{"status":"Charging","targetPct":70,"currentPct":75,"chargeSettings":"profile","chargeMode":"manual","preferredChargeMode":"manual","active":true,"remainingTime":15,"progressBarPct":110.0},"climatisation":{"status":"Off","targetTemperatureCelsius":22.0,"targetTemperatureFahrenheit":72.0,"climatisationTrigger":"off","active":false,"remainingTime":0,"progressBarPct":0.0},"windowHeating":
```

Da steht eindeutig

    "targetPct":70

Falls das nicht der Wert aus dem Auto ist dann müssten wir rausfinden, wo das Limit des Autos abgespeichert ist.

andig, 2025-05-10:
Ich sehe auch

    "chargeSettings":"profile"

Gibts da im Auto irgendwelche komischen Profile die da rein spucken und vllt. noch deaktiviert werden müssen? Ich verschiebs mal in die Discussions, vllt. hat noch jemand eine Idee.