evcc-io/evcc #21135: English language - please have a 24 hour time option

gr0mit1, 2025-05-08:
I see that the times in English language are shown in AM/PM.  We don't really use this in the UK any more - please can you enable a 24h time version?  I can see it's available as standartd in the German version.  Hope this is an easy fix?

Many thanks
Tim Robinson

closed 2025-05-09 by naltatis

gr0mit1, 2025-05-09:
that's an amazingly fast response - thank you! Do i need to wait for the next release to test it? Or can I pull down the fix and test it myself?  I'm a bit new to Github and Home Assistant so not sure what the process is for testing things.  Is there a howto somewhere?
BR
Tim

StefanSchoof, 2025-05-09:
You can install tomorrow the nightly or wait until the next release.