evcc-io/evcc #21143: PV Erzwuger / PV Generator

bluemlh-pe, 2025-05-09:
### Describe the bug

Hallo Zusammen,
nach dem Update auf V0.203.5 stimmt die Erzeugte PV-Leistung nicht mehr. Ich habe 3 Erzeuger. Einmal eine Sonnenbatterie (wird unter PV1 mit einem positiven Wert dargestellt) dann noch zwei kleinere PV-Erzeuger PV2 und PV3 die werden über einen Shelly gemessen und zeigen da Erzeugung schon immer minus Werte an. Diese minus Werte werden seit dem Update auch in der HMI von EVCC angezeigt. Und nicht mehr zur Gesamtleistung addiert.

Hello everyone,
After updating to V0.203.5, the generated PV power is no longer correct. I have 3 generators. One is a solar battery (displayed under PV1 with a positive value), then two smaller PV generators PV2 and PV3 which are measured via a Shelly and always display minus values for generation. Since the update, these minus values are also displayed in the EVCC HMI. And are no longer added to the total power.

Best Regards Peter

### Steps to reproduce

n.n

### Configuration details

```yaml
n.n
```

### Log details

```text
n.n
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

V0.203.5

VolkerK62, 2025-05-09:
bei Shelly gab es ein paar Anpassungen, aber ohne Konfig und Logfile wirst du hier kein Erfolg haben

premultiply, 2025-05-09:
Config und Logfile fehlt.

closed 2025-05-09 by premultiply

bluemlh-pe, 2025-05-09:
habe ich ergänzt

premultiply, 2025-05-09:
Eine `usage: Balkonkraftwerk` etc gibt es nicht.

bluemlh-pe, 2025-05-09:
Hat aber bis jetzt immer so funktioniert. Muss das dann jetzt "pv" heißen?

bluemlh-pe, 2025-05-09:
Es kommt aber der richtige Wert:
[site  ] DEBUG 2025/05/09 08:35:15 pv 2 power: -268W
[site  ] DEBUG 2025/05/09 08:35:15 pv 3 power: -269W

VolkerK62, 2025-05-09:
>Hat aber bis jetzt immer so funktioniert.

dann nur, weil es ignoriert wurde.
Bzgl `usage` siehe Doku https://docs.evcc.io/docs/devices/meters#shelly-1pm-em-plug-s-pro-3em-in-monophase-mode
Es gibt `grid, pv, charge`

>Es kommt aber der richtige Wert:
```
[site ] DEBUG 2025/05/09 08:35:15 pv 2 power: -268W
[site ] DEBUG 2025/05/09 08:35:15 pv 3 power: -269W
```
PV-Power muss positiv sein.