evcc-io/evcc #21144: New UI - Temp. Geräte Deaktivieren

Crajs20, 2025-05-09:
Hallo,

habe mir die neue UI angesehen und bin beigeistert. Schlage mich gerade auch mit der integration eines Heizstabes (Shelly) rum und das wird die Arbeit erheblich vereinfachen und auch viele Leute, die bisher sich nicht sich an das YAML-File editieren ran getraut haben, zu EVCC führen.

In ein paar Tagen kommt auch wieder die Pool Heizung in die Konfig, die ich mir aus der Datei vom Letzten Jahr wieder heraus kopieren darf. Hier wäre es toll in der neuen UI, wenn man Geräte, die man nicht benötigt auch deaktivieren kann, anstatt diese wieder zu löschen. Vielleicht habt ihr dies schon vorgesehen, konnte die Option aber nicht finden.
Einfach eine Option "aktiv / inaktiev" pro Gerät und man kann diese in der Konfig belassen, aber nicht nutzen.

Danke für das Tolle Projekt.

TobiasHuber1980, 2025-05-10:
Siehe hier https://github.com/evcc-io/evcc/issues/18141

andig, 2025-05-11:
@naltatis vielleicht wäre das die richtige Gelegenheit, mal aktiv/inaktiv zusammen mir "Fehler beim Start" zu betrachten?