evcc-io/evcc #21155: Unterstützung für eigene PV-Vorhersagedaten via MQTT

lawgiver0, 2025-05-09:
Beschreibung:
Ich möchte vorschlagen, dass evcc externe PV-Vorhersagedaten auch über MQTT entgegennehmen kann – zusätzlich oder alternativ zur bestehenden Solcast-Integration.

Anwendungsfall:
In meinem Setup erzeuge ich bereits PV-Vorhersagedaten (z. B. über Solcast, die DWD-API oder eigene Modelle) und veröffentliche diese regelmäßig per MQTT – etwa aus Home Assistant oder Node-RED.

Aktuell ist es nur möglich, interne Templates wie Solcast direkt in der Konfiguration unter tariffs > solar zu verwenden. Es gibt aber keine Möglichkeit, eigene Prognosedaten in der von evcc erwarteten Struktur über MQTT bereitzustellen – z. B. unter einem Topic wie evcc/forecast/solar:
[
  { "time": "2025-05-09T08:00:00Z", "value": 10000 },
  { "time": "2025-05-09T09:00:00Z", "value": 11000 }
]

Vorschlag zur Umsetzung:
Eine neue Konfigurationsmöglichkeit (z. B. type: mqtt unter tariffs > solar) wäre hilfreich, bei der der Pfad zum MQTT-Topic konfiguriert werden kann. Das Format der Daten könnte sich an der bestehenden Struktur orientieren (Array von Objekten mit Zeit- und Leistungswert).

Vorteile:

Flexibilität für Nutzer mit eigenen Datenquellen

Nutzung bestehender Automatisierungstools wie Home Assistant oder Node-RED

Entkopplung der Vorhersage vom internen Solcast-Modul und damit der notwendigen Solcast-Abfragen

Vielen Dank fürs Prüfen dieser Idee!

StefanSchoof, 2025-05-09:
Das geht. Unter https://github.com/evcc-io/evcc/discussions/18907#discussioncomment-12235450 findet sich ein Beispiel.

StefanSchoof, 2025-05-09:
Es gab seit dem ein Breaking Change und der Wert muss jetzt value und nicht mehr price heißen

andig, 2025-05-09:
Weitere Beispiele siehe jedes entsprechende Template.

closed 2025-05-09 by andig