evcc-io/evcc #21163: evcc hängt auf 1phasig fest (openWB als Ladepunkt)

sledge2019, 2025-05-10:
### Describe the bug

Hallo,

Gelegentlich passiert es, dass die EVCC auf 1phasig laden hängen bleiubt, obwohl REICHLICH PV-Überschuss vorhanden ist. Auch ein manuelles umstellen auf 3p erzielt keine Änderung.

Ausprobiert ohne Erfolg:

- Auto abstecken / anstecken
- EVCC rebooten
- EVCC Server Shutdown / Neustart
- openWB reboot
- openWB Strom aus / Strom an

Ich hatte das Phänomen schon einmal (vor ein paar Wochen), da hat denn EVCC aus / openWB stromlos, openWB an, EVCC an geholfen - aber das ist ja kein Zustand :-)

OpenWB Version 1.9.304.0 / als "Nur Ladepunkt" konfiguriert.

Bzgl. Nightly: Gerade nochmal getestet. Die Sonne im "Home" taucht auch kurz auf, dass er gleich auf 3p umschalten wird - bleibt aber auf einer Phase. Keine Veränderung.

Dankbar für jeden Schubs in die richtige Richtung.

Tom

### Steps to reproduce

1. Auto mit der Wallbox verbinden
2. Solar-Überschuss zum Abwinken
3. Warten, dass die Phasenumschaltung beginnt.

...

### Configuration details

```yaml
[main  ] INFO 2025/05/10 16:02:13 evcc 0.203.5
[main  ] INFO 2025/05/10 16:02:13 using config file: /etc/evcc.yaml
[db    ] INFO 2025/05/10 16:02:13 using sqlite database: /var/lib/evcc/evcc.db
[mqtt  ] INFO 2025/05/10 16:02:13 connecting evcc-1056994561 at tcp://mqtt.seconet.de:1883
[mqtt  ] DEBUG 2025/05/10 16:02:13 tcp://mqtt.seconet.de:1883 connected
[mqtt  ] DEBUG 2025/05/10 16:02:13 tcp://mqtt.seconet.de:1883 subscribe bridges/pool/heizung
[openwb] INFO 2025/05/10 16:02:13 connecting evcc-1452368680 at tcp://openwb.seconet.de:1883
[openwb] DEBUG 2025/05/10 16:02:13 tcp://openwb.seconet.de:1883 connected
[openwb] DEBUG 2025/05/10 16:02:13 tcp://openwb.seconet.de:1883 subscribe openWB/system/Timestamp
[tariff] ERROR 2025/05/10 16:02:15 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/05/10 16:02:15 creating tariff solar-template-1 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[tariff] ERROR 2025/05/10 16:02:15 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/05/10 16:02:15 creating tariff solar-template-0 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)[tariff] ERROR 2025/05/10 16:02:15 unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/05/10 16:02:15 creating tariff solar-template-2 failed: cannot create tariff type 'template': cannot create tariff type 'custom': unexpected status: 429 (Too Many Requests)
[lp-1  ] DEBUG 2025/05/10 16:02:15 charge total import: 13764.030kWh

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: *****
# trial token, valid until 2025-05-03

interval: 15s # control cycle interval

tariffs:
  currency: EUR
  grid:
    type: fixed
    price: 0.30
  feedin:
    type: fixed
    price: 0.12
  solar:
    - type: template
      template: forecast-solar
      lat: *****
      lon: *****
      dec: 25
      kwp: 4.3
      az: -135
    - type: template
      template: forecast-solar
      lat: *****
      lon: *****
      dec: 25
      kwp: 8.54
      az: 90
    - type: template
      template: forecast-solar
      lat: *****
      lon: *****
      dec: 25
      kwp: 5.2
      az: 30

meters:
- type: template
  template: victron-energy 
  usage: grid  
  host: victron.seconet.de  
  port: 502  
  name: Netz
- type: template
  template: victron-energy 
  usage: pv  
  host: victron.seconet.de  
  port: 502  
  name: Fronius
- type: template
  template: victron-energy 
  usage: battery  
  host: victron.seconet.de  
  port: 502  
  name: Victron
- type: custom
  name: Pool
  power:
    source: mqtt
    topic: "bridges/pool/heizung"

chargers:
- type: template
  template: openwb 
  host: openwb.seconet.de  
  phases1p3p: true
  name: OpenWB

vehicles:
- type: template
  template: tesla 
  title: Model 3  
  capacity: 55  
  clientId: 88dd3195-1eaf-4c42-8282-71af211c957a  
  accessToken: *****
  refreshToken: *****
  name: Model3

loadpoints:
- title: Garage
  charger: OpenWB
  vehicle: Model3
  mode: pv

site:
  title: Herrngiersdorf
  meters:
    grid: Netz
    pv:
    - Fronius
    battery:
    - Victron
    aux:
    - Pool

Version: `0.203.5`
```

### Log details

```text
[lp-1 ] DEBUG 2025/05/10 15:58:45 set charge current limit: 20A
[site ] DEBUG 2025/05/10 15:59:00 ----
[lp-1 ] DEBUG 2025/05/10 15:59:00 charge power: 4667W
[lp-1 ] DEBUG 2025/05/10 15:59:00 charge currents: [20.1 -0.1 0.1]A
[site ] DEBUG 2025/05/10 15:59:00 aux 1 power: 229W
[site ] DEBUG 2025/05/10 15:59:00 battery 1 power: -0W
[site ] DEBUG 2025/05/10 15:59:00 grid power: -4461W
[site ] DEBUG 2025/05/10 15:59:00 battery 1 soc: 100%
[site ] DEBUG 2025/05/10 15:59:01 pv 1 power: 10187W
[site ] DEBUG 2025/05/10 15:59:01 site power: -4690W
[lp-1 ] DEBUG 2025/05/10 15:59:01 vehicle odometer: 97585km
[lp-1 ] DEBUG 2025/05/10 15:59:01 detected active phases: 1p
[lp-1 ] DEBUG 2025/05/10 15:59:01 charge total import: 13763.790kWh
[lp-1 ] DEBUG 2025/05/10 15:59:01 charger status: C
[lp-1 ] DEBUG 2025/05/10 15:59:01 soc estimated: 60.03% (vehicle: 60.00%)
[lp-1 ] DEBUG 2025/05/10 15:59:01 vehicle soc: 60%
[lp-1 ] DEBUG 2025/05/10 15:59:01 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/10 15:59:01 vehicle range: 252km
[lp-1 ] DEBUG 2025/05/10 15:59:01 available power 9357W > 4140W min 3p threshold
[lp-1 ] DEBUG 2025/05/10 15:59:01 phase scale3p in 45s
[lp-1 ] DEBUG 2025/05/10 15:59:01 pv charge current: 40.4A = 20A + 20.4A (-4690W @ 1p)
[site ] DEBUG 2025/05/10 15:59:15 ----
[lp-1 ] DEBUG 2025/05/10 15:59:15 charge power: 4682W

Ein paar Minuten später wird das mit dem "3p treshold" nicht mehr angezeigt:

[lp-1 ] DEBUG 2025/05/10 16:08:16 vehicle range: 252km
[lp-1 ] DEBUG 2025/05/10 16:08:16 pv charge current: 38.7A = 20A + 18.7A (-4294W @ 1p)
[site ] DEBUG 2025/05/10 16:08:30 ----
[lp-1 ] DEBUG 2025/05/10 16:08:30 charge power: 4610W
[lp-1 ] DEBUG 2025/05/10 16:08:30 charge currents: [20 -0.1 0.1]A
[site ] DEBUG 2025/05/10 16:08:30 aux 1 power: 229W
[site ] DEBUG 2025/05/10 16:08:30 battery 1 power: -0W
[site ] DEBUG 2025/05/10 16:08:30 grid power: -4085W
[site ] DEBUG 2025/05/10 16:08:30 battery 1 soc: 100%
[site ] DEBUG 2025/05/10 16:08:30 pv 1 power: 9735W
[site ] DEBUG 2025/05/10 16:08:30 site power: -4314W
[lp-1 ] DEBUG 2025/05/10 16:08:30 detected active phases: 1p
[lp-1 ] DEBUG 2025/05/10 16:08:30 charge total import: 13764.520kWh
[lp-1 ] DEBUG 2025/05/10 16:08:30 charger status: C
[lp-1 ] DEBUG 2025/05/10 16:08:30 soc estimated: 61.23% (vehicle: 60.00%)
[lp-1 ] DEBUG 2025/05/10 16:08:30 vehicle soc: 61%
[lp-1 ] DEBUG 2025/05/10 16:08:30 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/10 16:08:30 vehicle range: 252km
[lp-1 ] DEBUG 2025/05/10 16:08:30 pv charge current: 38.8A = 20A + 18.8A (-4314W @ 1p)
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

# evcc -v evcc version 0.203.5

andig, 2025-05-11:
Der relevante Logteil in den Mitte fehlt.

sledge2019, 2025-05-11:
Hi,

es handelt sich um eine openWB mit 11kW. Das Problem scheint auf der openWB Seite zu liegen. Ich wollte heute morgen neue Logs "produzieren" und die openWB geht sofort hoch bis 4.6 kW Ladeleistung (also die 20A). Dazu kam dann die entsprechende Log-Meldung "charger out of sync....".

Sofort das Fahrzeug abgesteckt, openWB nochmal stromlos gemacht und wieder eingeschaltet. EVCC immer noch mit der gleichen "Fehlermeldung" und glaubte, trotz abgestecktem Fahrzeug in einem Ladevorgang zu sein.

Also habe ich die auf die letzte Stable zurückgesetzt (1.9.303) und jetzt läuft erstmal alles, wie es soll.

Aber irgendwas "veranlasst" die openWB, sich aufzuhängen und in diesen "ich lade dann mal mit einer Phase und 20A" Modus zu springen und dort hängenzubleiben.

Ich beobachte das weiter und werde beim nächsten Vorkommen das vollständige Log hochschieben.

Intermittierende Fehler sind super :-)

Danke soweit.

closed 2025-05-11 by sledge2019