evcc-io/evcc #21168: [RCT Template] Make usage of  rct.S0ExternalPowerW configurable

HWE70, 2025-05-11:
**Is your feature request related to a problem? Please describe.**
In a simple Master/Slave setup where the Save is a simple inverter only, using rct.S0ExternalPowerW makes sense, as it simply combines the Solar generators of the Slave in one value without the need to query the Slave separately.

If there is more then one Slave, in my case an PV Inverter and a Battery Inverter, rct.S0ExternalPowerW  is a mixture of Slave PV power, Slave Battery power and maybe grid or Master power loaded to the Slave battery as well. It makes no sense in this situation to use rct.S0ExternalPowerW. Instead all inverters need to be queried separately, while in all cases only  rct.SolarGenAPowerW and rct.SolarGenBPowerW are of interest. 

**Describe the solution you'd like**
The use of rct.S0ExternalPowerW  should be configurable in the RCT template. The default should be "true" to avoid a breaking change, but there should be an option to skip it in the calculation done in rct.go, eg. "return **a + b**, err" instead of "return **a + b + c**, err"

**Describe alternatives you've considered**
As a workaround I don't query "PV" currenty directly by evcc, but use HA and make the PV power values per inverter available to evcc by MQTT.

**Additional context**
Nothing more to mention

andig, 2025-05-11:
/cc @Maschga do you want to look into this?

Maschga, 2025-05-11:
Unfortunately, we don't know how`io_board.s0_external_power` is calculated. [rctlient](https://rctclient.readthedocs.io/en/latest/inverter_registry.html#io-board) doesn't have a description for it.

So with regard to the rct app, is this about the data from G?
Does the official rct app also mix the data?

HWE70, 2025-05-11:
Correct, there is no documentation for s0_external_power. But as far as I can see for the master all slave power is summed up  in this single OID. As long as there are slave inverters only, everything is fine. But if one salve is battery or a hybrid inverter it makes no sense any longer include s0_external_power into the PV generator formula. There is no "battery power" OID. That's why just created a helper in HA to make the attached picture more obvious (tomorrow). So far there is only the master's s0_external_power OID and the all generators OID of the slave inverter visible in the picture (I will update the picture tomorrow). But one can see already, that between 12:00 and 20:00 both graphs are identical, because there was no battery activity, while outside this time frame battery activity was happening (e.g. the odd export to grid last night), and this is visible in s0_external_power as well. So s0_external_power is not just about PV power, it sums up any power imported or exported by any slave, regardless if it is PV power or battery charge/discharge. That's why I ask for a switch to remove s0_external_power from the formula for system like mine. 

[image]

andig, 2025-05-12:
@HWE70 I would even assume that‘s an invalid installation situation. Since S0 ist just impulses the RCT cannot distinguish demand from charger. Even without evcc this would not work, at least for an Hybrid RCT

HWE70, 2025-05-12:
No, that's not an invalid installation, having slave systems with batteries is fully support and a common installation. Adding s0_external_power to the PV generator power by default is simply not a good idea, as it only works in exactly one setup, in which all slaves are plain PV inverters. And by the way the evcc battery controls works fine for slave batteries as well, as long as they are configured as distinct batteries.

[image]

Thus currently the batteries are directly connected  to evcc, but the PV plants needs HA input, as otherwise the values are unusable due to s0_external_power 

```
#
# [Meter] PV values gathered by Homeassistent
#
- name: pv_sued # MASTER (PS 10 DC)
  type: custom
  power: 
    source: mqtt
    broker: localhost:1883
    topic: rct/ps_sued/pv_power_total
    user: mqtt
    password: ******

- name: pv_ost_west # SLAVE INVERTER (PI 6)
  type: custom
  power: 
    source: mqtt
    broker: localhost:1883
    topic: rct/ps_ost_west/pv_power_total
    user: mqtt
    password: *******

#
# [Meter] Battery values directly discovered by evcc
#
- name: battery # MASTER (PS 10 DC)
  type: template
  template: rct-power
  usage: battery
  host: 192.168.40.18
  capacity: 11.5
  minsoc: 50
  maxsoc: 97

- name: battery2 # SLAVE BATTERY (PS 6 AC)
  type: template
  template: rct-power
  usage: battery
  host: 192.168.40.59
  capacity: 11.5
  minsoc: 7
  maxsoc: 97  
```

Maschga, 2025-05-12:
> And by the way the evcc battery controls works fine for slave batteries as well, as long as they are configured as distinct batteries.

If this is a bug, please create another issue.

I don't like the solution of making rct.S0ExternalPowerW configurable.
Before I think about this solution, I will write RCT to have a look at this problem and ask them to explain the calculation of rct.S0ExternalPowerW and if there is another registry that can be used.

andig, 2025-05-12:
@Maschga S0 will just show abs(power). 

> No, that's not an invalid installation, having slave systems with batteries is fully support and a common installation. 

@HWE70 can you explain what's the purpose of specific scenario in your case?

HWE70, 2025-05-12:
In each RCT PV inverter, regardless if we are talking about plain PV inverters like "Power Inverter 6.0" or hybrid PV inverters like "Power Storage 10 DC" there are two separated MPP trackers. So, if you are interested in the inverter's current PV power, one simply has to summarize the two power values provided by the two trackers (**SolarGenAPowerW + SolarGenBPowerW)**. Today evcc additionally adds **S0ExternalPowerW** by default, which is power provided externally by other slave inverters. This can be **PV** inverters, too, but this can be **battery** inverters like "Power Storage 6.0 AC" as well. For this reason summarizing **SolarGenAPowerW + SolarGenBPowerW + S0ExternalPowerW** leads to the correct total PV power of the **overall** PV system (master + slaves) in just two cases:

1. There are no slaves at all. In this case S0ExternalPowerW is always 0.
2. There are just plain PV inverter slaves (no battery inverter, no hybrid inverter). In this case the master's S0ExternalPowerW is the sum of PV power provided by the slaves, because the inverters can just export power, but never import power.

In my case it is different, because there are

1. a hybrid PV inverter (Power Storage 10 DC) as a master 
2. a plain PV inverter (Power Inverter 6.0) as a slave, and 
3. a battery inverter (Power Storage 6.0 AC) as a second slave

If evcc connects to the master, "SolarGenAPowerW + SolarGenBPowerW" describes the PV power provided by the master's MPPs. However, in this setup I can't just add the masters S0ExternalPowerW value, because now this includes the PV power of (2) and the battery power of (3). If the battery in charge mode  **S0ExternalPowerW = Power PV (2) - Power Bat (3).** If the battery is in discharge mode **S0ExternalPowerW = Power PV (2) + Power Bat (3)**.

For this reason in this setup taken S0ExternalPowerW into account makes no sense at all. Instead I just like to query  **SolarGenAPowerW + SolarGenBPowerW** from (1) and **SolarGenAPowerW + SolarGenBPowerW** from (2). This is exactly what I do in the configuration I already provided:

rct/ps_sued/pv_power_total  = SolarGenAPowerW + SolarGenBPowerW of the master (1)
rct/ps_ost_west/pv_power_total = SolarGenAPowerW + SolarGenBPowerW of the slave PV inverter (2)

But today HA queries **SolarGenAPowerW + SolarGenBPowerW** from both inverters and I publish the values to evcc using MQTT.

Please find the updated graph below:

1. the blue graph is the master's S0ExternalPowerW 
2. the yellow graph is the PV Power of the slave PV inverter (SolarGenAPowerW + SolarGenBPowerW)
3. the red graph is the power of the slave battery inverter 

As you can see S0ExternalPowerW  = "PV Power of the slave PV inverter" + "power of the slave battery inverter"

Just forget about the time frame between 4:00 and 6:00. But in the remaining time this is obvious, and this is the reason why S0ExternalPowerW can't be used for the PV power calculation if there is a slave battery inverter or a slave hybrid inverter. Instead "SolarGenAPowerW + SolarGenBPowerW" must be queried for **each** inverter separately.

Of course the default "SolarGenAPowerW + SolarGenBPowerW + S0ExternalPowerW" should persist, because if there is just one plain PV inverter slave, then there is no need to query the slave at all, because the master knows about the slave's power already, because (slave) "SolarGenAPowerW + SolarGenBPowerW" = (master) "S0ExternalPowerW" in this case.

[image]

1977michael, 2025-05-15:
Auch wenn es ein Issuse ist und keine Diskussion, erlaubt mir bitte, ein paar Sätze zu dem Theme verlieren (habe auch RCT).

Wie du schreibst, rufst du (eventuell nur zeitweise und zum testen) die Werte über HA ab und gibst diese via mqtt an evcc weiter. Somit hast Du es doch selbst in der Hand, welche (aus Deiner Sicht plausiblen) Werte Du evcc zur Verarbeitung weitergibst. 

Richtig ist, dass die RCT-WR miteinander verknüpft werden müssen, damit diese miteinander kommunizieren können. Ich frage mich aber gerade, ob die Leistung eines 2. und/oder 3. WR (Slave) innerhalb von RCT auch tatsächlich als external Power (G) angezeigt wird, wenn alles korrekt verdrahtet und in RCT konfiguriert ist. Details dazu sind in der Anleitung von RCT enthalten. Da müsste ich auch erst nachlesen...
External Power (in der RCT-App als "G" dargestellt) zeigt doch die Werte eines externen Energierezeugers (WR nicht von RCT, kleines Windrad etc.) an, der über einen S0-Zähler mit Infos versorgt wird. Das habe ich bei mir mit dem EZ1-D (1800 W) von Ap-Systems so gelöst, der dann als "G" in RCT angezeigt wird und über evcc automatisch mit ausgelesen wird.

Zugegeben stellt Deine Konfiguration nicht den Regelfall dar ;)

PR #21234 (2025-05-18): RCT: make S0ExternalPowerW configurable (BC)
Fix #21168
closed 2025-05-18 by andig