evcc-io/evcc #21170: Batterie entladen wird nicht angehalten, wenn günstig aus dem Netz geladen werden soll

svenp, 2025-05-11:
### Describe the bug

Batterie entladen wird nicht angehalten, wenn günstig aus dem Netz geladen werden soll.
Heute sind die Strompeise so negativ, das man für den Verbrauch bezahlt wird. 
Es wird aktuell Auto geladen und Heizpatrone, der Akkus soll auch mit günstigem Netz Strom geladen werden. 
Alles ist so eingestellt, aber es wird der Akku entladen.

[image]

### Steps to reproduce

Es ist nicht verständlich warum der Akku noch entladen wird

### Configuration details

```yaml
evcc 0.203.5
```

### Log details

```text
lp-2  ] DEBUG 2025/05/11 12:17:57 charger status: C
[lp-2  ] DEBUG 2025/05/11 12:17:58 smart cost active: -0.04
[site  ] DEBUG 2025/05/11 12:17:58 solar forecast: accumulated 56.589kWh, produced 114.450kWh, scale 2.022
[site  ] DEBUG 2025/05/11 12:18:12 ----
[lp-1  ] DEBUG 2025/05/11 12:18:12 charge power: 10390W
[lp-1  ] DEBUG 2025/05/11 12:18:12 charge currents: [14.5 14.5 14.5]A
[lp-2  ] DEBUG 2025/05/11 12:18:12 charge power: 3844W
[site  ] DEBUG 2025/05/11 12:18:12 pv 2 power: 7087W
[site  ] DEBUG 2025/05/11 12:18:12 pv 1 power: 3018W
[site  ] DEBUG 2025/05/11 12:18:12 grid power: 3998W
[site  ] DEBUG 2025/05/11 12:18:12 pv power: 10105W
[site  ] DEBUG 2025/05/11 12:18:12 battery 1 power: 4625W
[site  ] DEBUG 2025/05/11 12:18:12 battery 1 soc: 72%
[site  ] DEBUG 2025/05/11 12:18:12 site power: 8723W
[lp-1  ] DEBUG 2025/05/11 12:18:12 detected active phases: 3p
[lp-1  ] DEBUG 2025/05/11 12:18:12 charge total import: 1052.276kWh
[lp-1  ] DEBUG 2025/05/11 12:18:12 charger status: C
[mg    ] DEBUG 2025/05/11 12:18:12 StatInvaild. Starting query
[mg    ] DEBUG 2025/05/11 12:18:13 4: The remote control instruction failed, please try again later.
[mg    ] DEBUG 2025/05/11 12:18:13  No answer yet. Continue status query in background
[lp-1  ] DEBUG 2025/05/11 12:18:13 smart cost active: -0.04
[site  ] DEBUG 2025/05/11 12:18:13 solar forecast: accumulated 56.609kWh, produced 114.492kWh, scale 2.022
[mg    ] DEBUG 2025/05/11 12:18:15 Starting repeated query. Count: 0
[mg    ] DEBUG 2025/05/11 12:18:15 Exiting repeated query. Count: 1
[site  ] DEBUG 2025/05/11 12:18:27 ----
[lp-1  ] DEBUG 2025/05/11 12:18:27 charge power: 10390W
[lp-1  ] DEBUG 2025/05/11 12:18:27 charge currents: [14.5 14.5 14.5]A
[lp-2  ] DEBUG 2025/05/11 12:18:27 charge power: 3809W
[site  ] DEBUG 2025/05/11 12:18:27 pv 2 power: 7086W
[site  ] DEBUG 2025/05/11 12:18:27 battery 1 power: 4617W
[site  ] DEBUG 2025/05/11 12:18:27 grid power: 4031W
[site  ] DEBUG 2025/05/11 12:18:27 pv 1 power: 3038W
[site  ] DEBUG 2025/05/11 12:18:27 pv power: 10124W
[site  ] DEBUG 2025/05/11 12:18:27 battery 1 soc: 72%
[site  ] DEBUG 2025/05/11 12:18:27 site power: 8748W
[lp-2  ] DEBUG 2025/05/11 12:18:27 charge total import: 984.427kWh
[lp-2  ] DEBUG 2025/05/11 12:18:27 charger status: C
[lp-2  ] DEBUG 2025/05/11 12:18:27 smart cost active: -0.04
[site  ] DEBUG 2025/05/11 12:18:27 solar forecast: accumulated 56.629kWh, produced 114.534kWh, scale 2.023
[site  ] DEBUG 2025/05/11 12:18:42 ----
[lp-1  ] DEBUG 2025/05/11 12:18:42 charge power: 10390W
[lp-1  ] DEBUG 2025/05/11 12:18:42 charge currents: [14.5 14.5 14.5]A
[lp-2  ] DEBUG 2025/05/11 12:18:42 charge power: 3776W
[site  ] DEBUG 2025/05/11 12:18:42 pv 1 power: 3056W
[site  ] DEBUG 2025/05/11 12:18:42 pv 2 power: 7108W
[site  ] DEBUG 2025/05/11 12:18:42 battery 1 power: 4620W
[site  ] DEBUG 2025/05/11 12:18:42 pv power: 10164W
[site  ] DEBUG 2025/05/11 12:18:42 battery 1 soc: 72%
[... 502 lines cut ...]
[site  ] DEBUG 2025/05/11 12:25:42 solar forecast: accumulated 57.238kWh, produced 115.771kWh, scale 2.023
[site  ] DEBUG 2025/05/11 12:25:57 ----
[lp-1  ] DEBUG 2025/05/11 12:25:57 charge power: 10390W
[lp-1  ] DEBUG 2025/05/11 12:25:57 charge currents: [14.5 14.5 14.5]A
[lp-2  ] DEBUG 2025/05/11 12:25:57 charge power: 3887W
[site  ] DEBUG 2025/05/11 12:25:57 grid power: 5659W
[site  ] DEBUG 2025/05/11 12:25:57 pv 2 power: 7152W
[site  ] DEBUG 2025/05/11 12:25:57 pv 1 power: 3157W
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc 0.203.5

andig, 2025-05-11:
Im Log fehlt jeder Hinweis darauf, dass überhaupt ein battery mode gesetzt würde. Ansonsten: config fehlt.

closed 2025-05-11 by andig

svenp, 2025-05-11:
HIer noch die fehlende config

[config.log](https://github.com/user-attachments/files/20149516/config.log)

andig, 2025-05-11:
/cc @premultiply

maatinh, 2025-05-12:
Das Szenario hatten gestern sicherlich viele und ich habe dafür meinen WR manuell so eingestellt, dass sämtlicher Verbrauch nur aus dem Netz gezogen wurde und die PV-Leistung auf 0 reduziert wurde. Das maximiert den Ertrag durch negative Strompreise.

Ich denke, das ist etwas, was beim Netzladen auch in evcc abgebildet werden könnte:

1: Netzladen wenn die Kosten >0 (oder >EEG?) sind, dann soll auch die eventuell noch vorhandene PV-Leistung verwendet werden um den Netzbezug auf das minimum zu reduzieren
2: Netzladen/Auto laden/Heizen/... wenn Kosten <0 - dann sollte - je nach WR Fähigkeiten - die PV abgeschalten werden um Netzbezug zu maximieren und die Einspeisung netzdienlich zu reduzieren

andig, 2025-05-12:
> Batterie entladen wird nicht angehalten, wenn günstig aus dem Netz geladen werden soll.

> Alles ist so eingestellt, aber es wird der Akku entladen.

Mir ist nicht klar was Du da eingestellt hast oder was du erwartest. Bei entsprechender Konfiguration würde ich erwarten, dass die Batterie aus dem Netz geladen wird. Davon ist aber im Log nichts zu sehen, vmtl. ist es nicht konfiguriert.

> dass sämtlicher Verbrauch nur aus dem Netz gezogen wurde

Das gibts heute nicht. "Hold" ist eine Entladesperre die beim Fahrzeug-Schnelladen greift, aber keine Ladesperre. Das wäre dann eine zusätzliche Featurediskussion.

svenp, 2025-05-12:
> Mir ist nicht klar was Du da eingestellt hast oder was du erwartest. Bei entsprechender Konfiguration würde ich erwarten, dass die Batterie aus dem Netz geladen wird. Davon ist aber im Log nichts zu [...]

Das "Hold" Funktioniert beim bei schnellen laden des Autos auch, damit unterdrücke ich mit einem iobroker script, das in den SunnyIsland per modus register das entladen unterdrückt (maxdischarge = 0)  (ich habe kein HomeManager von SMA)
Ein  "Hold" hätte also für mich in dem Fall schon geholfen.

Meine Erwartung war, wenn ich z.B.  bei Batterie einstelle, lade aus dem Netz bei Preis < -5cent, das das hold gesetzt wird, damit der Akku nicht entladen wird, aber der Akku mit maximalem Strom geladen wird. (Ok habe gerade gesehen das sowohl laden als auch entladen auf hold gesetzt wird) Laden hatte ich nicht betachtet..... und in meinem Script anders verarbeitet.
 Das Auto hat in dem Fall ja auch geladen weil Preis < x cent, was dann aus Sonne un Netz kommen sollte. So meine Idee.
In Wirklichkeit kam es aber aus Sonne und Akku.

Es wäre ein Status schön, (Hold_discharge, Hold_charge und einfach nur "Hold" für beides wie jetzt schon) mit dem man dem Batteriewechselrichter sagen kann, lade ab Preis X  mit maximaler Leistung und sperre das entladen. (Idealerweise beim negativen Strompreis Planung des Akku berücksichtig, das der Akku genug Kapazität für die günstige Zeit hat, also vorher nicht schon  mit PV laden und schauen wie viel kriege ich in der Zeit geladen.)

premultiply, 2025-05-12:
> Das "Hold" Funktioniert beim bei schnellen laden des Autos auch, damit unterdrücke ich mit einem iobroker script, das in den SunnyIsland per modus register das entladen unterdrückt (maxdischarge = 0)

Wenn man es (in evcc) richtig konfiguriert macht evcc das selbst.
Das ist hier aber scheinbar (laut Log - da fehlen die entsprechenden Einträge) nicht der Fall und daher verstehe ich dieses Issue nicht.

svenp, 2025-05-13:
Also ist es nicht richtig konfiguriert?
Mein Verständnis war, ohne den SMA Homemanager funktioniert es nicht.

- name: sma-battery
  type: template
  template: sma-si-modbus
  usage: battery
# Modbus TCP
  modbus: tcpip
  id: 3
  host: 192.168.10.10 # Hostname
  port: 502 # Port
  capacity: 32 # Akkukapazität in kWh (optional)
  watchdog: 60s # optional
  chargepower: 4600 # Maximale Ladeleistung für erzwungenes Laden

Was habe ich falsch konfiguriert?

svenp, 2025-05-13:
[image]

[image]

Meinst Du das mit Batteriekontrolle, falls ja, das ist aktiv

premultiply, 2025-05-13:
Ok, spannend. Da stellt sich aber leider (immer noch) die Frage weshalb man davon im Log nichts sieht.