evcc-io/evcc #21181: Loggen und Kontrolle von Ladevorgängen mit Ladekabeln ohne Netzwerkverbindung über Fahrzeug API

AJ1979, 2025-05-12:
Zusätzlich zu meinem Ladepunkt zu Hause mit der Wallbox würde ich auch gerne die Ladevorgänge, welche mit dem mobilen Ladekabel am Arbeitsplatz, oder an diversen öffentlichen Ladestationen durchgeführt werden die geladene Energiemenge und CO2eq/kWh in evcc aufzeichnen und über evcc steuern.
Die Lademenge soll dafür aus dem SOC vom Auto berechnet werden. Die Autos sind in der vehicle section entsprechend konfiguriert.

Folgender Versuch hat leider nicht funktioniert:

```
chargers:
  - name: keba 
    type: template
    template: keba
    host: 10.0.0.42
    
  - name: Ladekabel
    type: template
    template: demo-charger
    status: A # optional
    power: 0 # optional
    enabled: true # optional

loadpoints:
  - title: Garage 
    charger: keba 
    mode: pv # charge mode (off, now, minpv, pv)
    soc:
      poll:
        mode: always
        interval: 60m
      estimate: true

  - title: Arbeit
    charger: Ladekabel 
    mode: now # charge mode (off, now, minpv, pv)
    soc:
      poll:
        mode: always
        interval: 60m
      estimate: true
```

naltatis, 2025-05-12:
Sehe ich als out-of-scope für evcc. Daten wir Preis/CO2/... hätten wir ja nicht bzw. müssten wir über weitere Konfiguration (Ladepreis Arbeitgeber) konfigurierbar machen. Bei öffentlichen Lademöglichkeiten hätte das vmtl. eine Pro-Ladevorgang-Datenpflege zur Konsequenz.

AJ1979, 2025-05-12:
es geht mir dabei nicht um den Preis, sondern nur um die geladenen kWh und CO2, damit man für jedes Fahrzeug den vollständigen CO2 Fußabdruck hat.