evcc-io/evcc #21184: Dokumentation zu Geräte -> Wärmepumpen, Heizstäbe -> über Relais-Kontakte

haexpo, 2025-05-12:
### Describe the bug

Mit Release 0.203.5 haben sich die setmode und getmode Werte verändert:

ALT: 
   Normal: 1
   Boost: 2
   Dimm: 3

NEU:
   Dimm: 1         
   Normal: 2       
   Boost: 3        

Die Dokumentation sollte angepasst werden und ein Hinweis in den Releasenotes für bestehende Nutzer wäre sinnvoll.

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
-
```

### Log details

```text
-
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.5

andig, 2025-05-12:
> Die Dokumentation sollte angepasst werden

Die ist auch open source, gerne PR :)

naltatis, 2025-05-12:
Wie @andig sagt, das gehört ins Doku Repo.

closed 2025-05-12 by naltatis