evcc-io/evcc #21199: Sporadic charger logic error: current mismatch

Sushifix, 2025-05-13:
### Describe the bug

[evcc-anonym-yaml.txt](https://github.com/user-attachments/files/20185019/evcc-anonym-yaml.txt)

[evcc-20250513-111402-debug.log](https://github.com/user-attachments/files/20185051/evcc-20250513-111402-debug.log)

Hallo, ich bekomme in meiner Umgebung immer mal wieder sporadisch die bekannte Meldung "charger logic error: current mismatch". Hier konkret:
[lp-1  ] WARN 2025/05/13 11:07:59 charger logic error: current mismatch (got 16.6A measured, expected 15.5A) - make sure your interval is at least 30s

Meine Version ist 0.203.5.
Da es hier in letzter Zeit bezüglich dem Lastmanagement deutliche Verbesserungen gab (vorher kam die Meldung sehr oft), wollte ich einmal melden, dass sie noch nicht ganz weg ist.

interval: 30s 

Eine Vergrößerung auf 60s brachte keine Verbesserung.

Mein Setup ist umfangreich. 4x Heidelberg, Zwei Lastkreise. Vielleicht liegt dort das Problem.
Ich habe Config und Log angehängt.

Wenn was fehlt oder ich helfen kann, dann gerne.
Dnake
Viele Grüße

### Steps to reproduce

1.Das angehängte Setup bei aktivem Laden (Ladeleistung muss nicht immer bei 11Kw liegen)
2.
3.
...

### Configuration details

```yaml
nterval: 30s # control cycle interval

sponsortoken: *****

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

circuits:
- name: main # eindeutiger Name, wird als Referenz für andere Circuits verwendet
  title: Hauptstromkreis # Anzeige in der UI (später)
  maxCurrent: 63
  meter: grid1 # optional
- name: WallboxenSaschaCircuit
  title: Wallboxen Sascha
  maxCurrent: 18
  parent: main
- name: WallboxSandraCircuit
  title: Wallbox Sandra
  maxCurrent: 18
  parent: main

vehicles:
- name: Idefix
  type: template
  template: mercedes
  title: Idefix
  icon: car
  user: *****
  region: EMEA # [EMEA, APAC, NORAM]
  accessToken: *****
  refreshToken: *****
  vin: *****
  capacity: 91
  
- name: Luigi
  type: template
[... 300 lines cut ...]
[lp-1  ] DEBUG 2025/05/13 11:07:59 charger status: C
[lp-1  ] DEBUG 2025/05/13 11:07:59 soc estimated: 51.97% (vehicle: 50.00%)
[lp-1  ] DEBUG 2025/05/13 11:07:59 vehicle soc: 52%
[lp-1  ] DEBUG 2025/05/13 11:07:59 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/05/13 11:07:59 vehicle range: 254km
[lp-1  ] WARN 2025/05/13 11:07:59 charger logic error: current mismatch (got 16.6A measured, expected 15.5A) - make sure your interval is at least 30s
[lp-1  ] DEBUG 2025/05/13 11:07:59 pv charge current: 17A = 16.6A + 0.37A (-255W @ 3p)
[lp-1  ] DEBUG 2025/05/13 11:07:59 set charge current limit:
```

### What type of operating system or environment does evcc run on?

Windows

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.5

VolkerK62, 2025-05-13:
Dein LP-1 lädt immer mit einem höheren Strom, als freigegeben. Warum auch immer? 
Wenn der Unterschied > 1A ist, kommt die Warnung.

Sushifix, 2025-05-13:
Wenn es eine Möglichkeit gäbe den Threshold einzustellen bin ich es los. Ein alternatives Strommeter (via Modbus, wenn es das gibt) finde ich jetzt etwas übertrieben, nur für die Warnung. Ich würde dann, wenn ich zu 99% alleine mit dem Thema (bei Lastmanagement aktiv) bin, davon ausgehen, dasss die amperfied / heidelberg energy control ganz und gar nicht genau misst. Gibt es hier Erfahrungen?

Oder gibt es andere Benutzer, die diese Meldung auch noch bekommen?