evcc-io/evcc #21206: Disable PV mode base on price

enboer, 2025-05-13:
You can set a price for energy in EVCC. Charger will switch to fast when tariff is below this. But what would be usefull is an option to disable solar charging if current price is above a certain level. So you can still plug in the care and it will only charge in PV if the price is in  this price range.

I created a screenshot how this could look like.

[image]

[image]

andig, 2025-05-13:
What is the use case for this feature?

andig, 2025-05-13:
From what I understand you have just described "smart grid charging". What's the difference?

enboer, 2025-05-13:
when I come home the price maybe 30ct. When plugging in the car it starts charging in PV mode, so not returning the left over to the grid. However during the night a normal charge could be cheaper (eg 20ct). It will be better then to keep feedin in the PV power to the grid at 30ct and pay during the night at 20ct. With the additional setting I can set the boundary until when I want it to allow charging in PV mode.
You are mentioning smart grid charging, do you mean I should set the charger to Off and set a plan for the next morning? Evcc with select the cheapest hours?

andig, 2025-05-13:
I'm sorry. I don't understand.

Lako-Unbekannt, 2025-05-13:
@enboer Why do you want to return electricity to the grid if the price is high? As far as I know the end user gains the same amount per kwh independ of the market price.