evcc-io/evcc #21211: Discharge wrong detected at solis inverter using rs485

humbs-roe, 2025-05-13:
### Describe the bug

[image]
[image]

The Discharge Power of 223W ist interpreted as charge Power by EVCC
Solis S6 Inverter ist connected to EVCC by RS485 USB converter
latest EVCC build is used

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
submitted later
```

### Log details

```text
submitted later
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

humbs-roe, 2025-05-13:
meters:
- type: template
  template: solis-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: grid  
  modbus: rs485serial  
  name: grid1
- type: template
  template: solis-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: pv  
  modbus: rs485serial  
  name: pv2
- type: template
  template: solis-hybrid 
  id: 1  
  device: /dev/ttyUSB0  
  baudrate: 9600  
  comset: 8N1  
  usage: battery  
  modbus: rs485serial  
  name: battery3

StefanSchoof, 2025-05-13:
Does the `solis-hybrid-s` template work?

andig, 2025-05-14:
Unfortunately your report is missing a detailed log file. Without log, we don‘t have the information for diagnosing the issue described. See https://docs.evcc.io/docs/faq#wie-kann-ich-ein-logfile-zur-fehleranalyse-erstellen.

closed 2025-05-14 by andig

humbs-roe, 2025-05-14:
With solis-hybrid-s template it seems better, thank you