evcc-io/evcc #21245: Custom Charger Konfiguration

peterle-k, 2025-05-15:
### Describe the bug

Die APT-Version von evcc (v0.203.5, installiert aus https://dl.evcc.io/public/evcc/stable/deb/debian) bricht beim Parsen einer gültigen custom-Charger-Konfiguration mit enable: und/oder enabled: ab – obwohl die Syntax laut Doku korrekt ist.

Ziel ist es, die Wallbox im Carport (einfacher Ladestecker) über den Shelly Pro 50EM zu schalten.

### Steps to reproduce

**In /etc/evcc.yaml folgenden charger definieren:**
- name: wbc
  type: custom
  power:
    source: script
    cmd: echo 0
  status:
    source: script
    cmd: echo 1
  enabled:
    source: http
    uri: http://192.168.8.9/rpc/Switch.GetStatus?id=0
    jq: .output
  enable:
    source: http
    uri: http://192.168.8.9/rpc/Switch.Set?id=0&on={{ . }}

**Erwartetes Verhalten**
Die Konfiguration wird korrekt geladen.
enabled: liest den Status des Schalters (z. B. Shelly Pro 50EM)
enable: schaltet das Gerät via HTTP

**Tatsächliches Verhalten**
evcc startet nicht.

Fehler im Log:
FATAL: charger [wbc] cannot create charger type 'custom': maxcurrent: invalid plugin type:

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: REDACTED
sponsortoken: REDACTED
telemetry: true
interval: 30s

meters:
# Netz
- type: template
  template: e3dc
  usage: grid
  host: 192.168.8.210
  port: 11516
  name: netz
# PV Leistung
- type: template
  template: e3dc
  usage: pv
  host: 192.168.8.210
  port: 11516
  name: pv
# Batterie
- type: template
  template: e3dc
  usage: battery
  host: 192.168.8.210
  port: 11516
  name: battery
# Shelly pro 50em im Carport
- name: shellyem50
  type: custom
  power:
    source: http
    uri: http://192.168.8.9/rpc/Shelly.GetStatus
    jq: .["em1:1"].act_power
  energy:
    source: http
    uri: http://192.168.8.9/rpc/Shelly.GetStatus
    jq: .["em1data:1"].total_act_energy
    scale: 0.001

chargers:
# TWC
- type: template
  template: twc3
  host: 192.168.8.124
  name: twc1
# Lader im Carport
- name: wbc
  type: custom
  power:
    source: script
    cmd: echo 0
  status:
    source: script
    cmd: echo 1
  enabled:
    source: http
    uri: http://192.168.8.9/rpc/Switch.GetStatus?id=0
    jq: .output
  enable:
    source: http
    uri: http://192.168.8.9/rpc/Switch.Set?id=0&on={{ . }}

vehicles:
# Tesla Model Y
- name: teslay
  type: template
  template: tesla
  title: Model Y
  clientID: REDACTED
  accessToken: REDACTED
  refreshToken: REDACTED
  vin: REDACTED
  capacity: 77
  phases: 3
  commandProxy: https://api.myteslamate.com
  proxyToken: REDACTED
  priority: 3

# Hyundai Ioniq per Bluelink
- type: template
  name: ioniq
  template: hyundai
  vin: REDACTED
  title: IONIQ
  user: REDACTED
  password: REDACTED
  language: de
  phases: 1
  priority: 2

loadpoints:
# TWC
- title: TWC
  charger: twc1
  mode: pv
  vehicle: teslay
  priority: 3
# Carport
- title: Carport
  charger: wbc
  meter: shellyem50
  mode: now
  vehicle: ioniq
  priority: 2

tariffs:
# Awattar
  currency: EUR
  grid:
    type: template
    template: awattar
    region: AT
    charges: 0.1725
    tax: 0.2
  feedin:
    type: fixed
    price: 0.09

mqtt:
# mosquitto
  broker: 192.168.8.164:1884
  topic: evcc
  clientid: evcc
  user: REDACTED
  password: REDACTED

site:
  title: Daheim
  meters:
    grid: netz
    pv: pv
    battery: battery
  residualPower: 300
```

### Log details

```text
Mai 15 20:27:58 raspberrypi systemd[1]: Stopping evcc.service - evcc...
Mai 15 20:27:58 raspberrypi systemd[1]: evcc.service: Main process exited, code=exited, status=1/FAILURE
Mai 15 20:27:58 raspberrypi systemd[1]: evcc.service: Failed with result 'exit-code'.
Mai 15 20:27:58 raspberrypi systemd[1]: Stopped evcc.service - evcc.
Mai 15 20:27:58 raspberrypi systemd[1]: Started evcc.service - evcc.
Mai 15 20:27:58 raspberrypi evcc[91340]: [main  ] INFO 2025/05/15 20:27:58 evcc 0.203.5
Mai 15 20:27:58 raspberrypi evcc[91340]: [main  ] INFO 2025/05/15 20:27:58 using config file: /etc/evcc.yaml
Mai 15 20:27:58 raspberrypi evcc[91340]: [db    ] INFO 2025/05/15 20:27:58 using sqlite database: /var/lib/evcc/evcc.db
Mai 15 20:27:58 raspberrypi evcc[91340]: [mqtt  ] INFO 2025/05/15 20:27:58 connecting evcc at tcp://192.168.8.164:1884
Mai 15 20:27:58 raspberrypi evcc[91340]: [main  ] INFO 2025/05/15 20:27:58 listening at :7070
Mai 15 20:27:58 raspberrypi evcc[91340]: [main  ] FATAL 2025/05/15 20:27:58 charger [wbc] cannot create charger 'wbc': cannot create charger type 'custom': maxcurrent: invalid plugin type:
Mai 15 20:27:58 raspberrypi evcc[91340]: [main  ] FATAL 2025/05/15 20:27:58 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.5

mfuchs1984, 2025-05-15:
Da fehlt maxcurrent, was laut Doku obligatorisch ist
https://docs.evcc.io/docs/devices/chargers#manuell

andig, 2025-05-15:
Die Fehlermeldung wird verbessert wenn custom Geräte ins UI kommen 👍🏻

closed 2025-05-15 by andig

naltatis, 2025-05-16:
> Die Fehlermeldung wird verbessert wenn custom Geräte ins UI kommen 👍🏻

see https://github.com/evcc-io/evcc/pull/20966

VolkerK62, 2025-05-16:
Warum nutzt du nicht das Template?
https://docs.evcc.io/docs/devices/smartswitches#shelly

peterle-k, 2025-05-17:
geht das template denn für den shelly pro em50 ?

VolkerK62, 2025-05-17:
Bei Shelly gab es vor kurzem umfassende "Aufräumarbeiten".
Mit dem aktuellen Release sollte es funktionieren.

peterle-k, 2025-05-17:
Ok, danke, dann nehm ich die Shelly template