evcc-io/evcc #21258: Mercedes not longer usable

chris9o, 2025-05-16:
### Describe the bug

Both the existing vehicle and a newly added vehicle do not work. The parallel running Mercedes integration in HA shows no error (with the same parameterization)

### Steps to reproduce

1.Add a new car to EVCC (Mercedes Benz)
2.klick on "check"
3.Result: cannot create vehicle type 'template': cannot create vehicle type 'mercedes': cannot get vehicles: Get "https://bff.emea-prod.mobilesdk.mercedes-benz.com/v2/vehicles": unexpected status: **400 (Bad Request)**
...

### Configuration details

```yaml
vehicles:
  - name: EQA
    type: template
    template: mercedes
    title: EQA # Wird in der Benutzeroberfläche angezeigt (optional)
    capacity: 77 # Akkukapazität in kWh (optional)
    user: username@mydomain
    region: EMEA # [EMEA, APAC, NORAM]
    accessToken: **REMOVED**    
    refreshToken: **REMOVED**
    vin: **REMOVED**  # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
```

### Log details

```text
No entries were found in the log 
only:
[mercedes] DEBUG 2025/05/16 12:57:08 identity.NewIdentity - token found in instance store
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.5

chris9o, 2025-05-21:
Hey @ReneNulschDE ,
tested and seems to work. Well done. 

I've tested both - new Token and adding a new car. Both OK.

Thank you very much.

chris9o, 2025-05-24:
During first load at the wallbox ...

Alfen_Eve: vehicle soc: Get \"[https://widget.emea-prod.mobilesdk.mercedes-benz.com/v1/vehicle/W1Nxxxxxxxxxx/vehicleattributes\](https://widget.emea-prod.mobilesdk.mercedes-benz.com/v1/vehicle/W1Nxxxxxxxxxx/vehicleattributes/)": unexpected status: 400 (Bad Request)

Sorry thought that all issues where closed ... there is one remaining :-(
Please have a look on that.  VIN  removed because of privacy.

andig, 2025-05-24:
There is, if configured.

ReneNulschDE, 2025-06-07:
and please describe the complete case. Questions:

Do you start evcc on the comandline in parallel to the running service like `evcc vehicle`?
Do you have multiple cars? if yes, please try to include only one car and check again?