evcc-io/evcc #21273: Improve smart charging with dynamic tarifs

Current implementation does always charge if below the configured limit and doesn't take all known prices into account. 

For cost saving smart charge should use the cheapest timeslot (calculated with the needed time to reach the target) that is below the limit.

With this change it's not required to adapt the limit all the time to use the best tarif.

notz, 2025-05-17:


andig, 2025-05-17:
What you‘re describing is the charge planner.

closed 2025-05-17 by andig

notz, 2025-05-18:
@andig Perhaps, I'm to stupid. But I don't think so.

I'm speaking of "Günstiges Netzladen".

Also if I create a repeating schedule for the weekend it doesn't seem to use the cheapest tarif that is under the price limit.

It's waiting until he knows every price and then chooses the cheapest slot. So the schedule is waiting for the last day (which is fine).

Then come smart charge limit into account. But there he doesn't take the cheapest prices. It starts on the first timeslot which is below the price limit.

andig, 2025-05-19:
> I'm speaking of "Günstiges Netzladen".

...und ich von https://docs.evcc.io/docs/features/plans

notz, 2025-05-19:
> Gibst du eine Abfahrtszeit und ein Ladeziel ein, sucht dir der Planungsalgorithmus automatisch die besten Ladezeiten raus.

That isn't the case, if you use the ev rarely (weekends).

I have today this prices:

| Hour | Price |
|----|----|
| 11 | 0,12 |
| 12 | 0,05 |
| 13 | 0,01 |
| 14 | -0,05 |

My car is 60% charged and the charging limit is 80%. So it needs 2h to charge.

- With a schedule for Saturday morning, it's waiting to know all prices

- With a smart charge limit of € 0,08 it's starting at 12 o'clock. But it would be cheaper to start at 13 o'clock

andig, 2025-05-19:
This is a kind of contradiction due to missing data. You're basically saying "I want to charge now but I also want to charge cheapest". Planner will ensure cheapest with the assumption that later data might provide better rates. If you want a better algorithm for charging _now_ without waiting for the plan you can adjust the smart charging limit by api.

notz, 2025-05-19:
> If you want a better algorithm for charging now without waiting for the plan you can adjust the smart charging limit by api

Yeah, that's the feature request about. Use the smart charging limit more intelligent to get the cheapest tariff.

Another cost function with the current planner implementation can't provide the functionality I need (AI can't forecast the price for 5 or more days in detail).

andig, 2025-05-19:
> AI can't forecast the price for 5 or more days in detail

It doesn't need to. All you need to do is to make sure that all "unknown" time slots have costs and are expensive so the planner doesn't choose them.

notz, 2025-05-19:
No, because the planner is making the wrong choice.

It would charge my ev, if today is the cheapest tariff of the known range (using energyforecast.de which has a bit pessimistic forecast). But I don't need to charge, because I can wait for better prices. Which often occurs, if the weather report changes to a more sunny day.

It only needs to charge if I plan a long drive (which works with the current planner).

notz, 2025-05-20:
@andig An optional max price (or an option to use only rates below the smart charging limit) per repeating plan would solve the issue.

Then it's possible to create a repeating plan to charge the car whenever the prices are cheap and optional creating another plan (manual or repeating) if a specific soc is required.

But probably that's not a common use case. At least for the ev smart charging ha plugin, someone requested the same feature.

niklaswa, 2025-06-14:
Hmm, okay, that's nice, then I'll probably just don't understand the planner right - as stated I'm using it without a vehicle, so I don't have a target percentage available and the goal of 5kWh would also be way too much 😅 I'll take a look at that though!

andig, 2025-06-21:
Basically you have a plan now that is „opportunistic“ in that it‘s unclear if that plan will be able reach its goal due to additional constraints. These could be stacked now: the lower the price limit, the higher the charge goal for the same time. 

This is a bit complex though. @naltatis would we want to add this?

andig, 2025-06-21:
I dont understand. What are „the cheapest slots“ if time doesn‘t matter? Chrapest compared to what?

naltatis, 2025-06-21:
I also dont understand. Looking at your https://github.com/notz/evcc/commit/b39a2417febf118751682e7ff5f787752674236c it looks like this change removes the guarantee, that the vehicle will reach the specific soc which goes against the whole concept of our charging plan.

We had some discussions about making the currently fixed smart cost limit more dynamic. Automatically update the value daily under specific constraints (cheapest n-hours between 18-6 o'clock, never higher then 20ct/kWh, ...). This would work for any kind of loadpoint (no soc required). But I'd see this as a feature that's independent of the charger.

notz, 2025-06-21:
@andig I have a dynamic tariff and on forecasts with a lot of sun or wind the prices will go down below zero. Mostly the cheapest slots are between 13:00 and 15:00. On current smart charging it will charge for example from 9 to 11 because the price is already low and under the limit. But from 13:00 to 15:00 it's much cheaper. Once I had -0,28ct/kWh. The smart charging won't use that because the soc reached the limit before. To be smart, I have to adapt the smart cost limit every time.

My implementation only replicates the smart charging functionality with more intelligence (which the planner already has) to use that cheaper times. I have no requirement to reach the limit on time. Therefore i can add another plan.