evcc-io/evcc #21281: Add support for additional Shelly devices

gr0mit1, 2025-05-17:
My previous request for additional hot water control was closed - I suspect I didn't put enough useful information in and I apologise for my hasty request.

Background
-----------
I have an EV charger, PV, Victron battery storage all running happily using evcc.

I recently added a Shelly PM1 which toggles my hot water heater on and off if the car is charged (or not connected) and there is excess solar being generated and the battery is full.  My heater is approx 3kW.  So turning it on or off is a bit of a 'blunt instrument'.

Requirement
-------------
I would like to add some proportional heater control to the hot water heating element.  I bought a Shelly Dimmer 0-10v PM, which, using a Carlo Gavazzi phase angle controller (conveniently with 0-10v input) can control the power to my heater from 0-3kW from Home Assistant. It appears as a dimmable light in the Home Assistant control panel.

I would like evcc to control the power to the hot water heater, (in the same way as it does to my car) so that it balances the available excess PV and dumps it into the hot water tank.  e.g. if there's 1kW excess, evcc would dump 1kW into the water tank by manipulating the 'brightness' of the hot water heater.

Does this make more sense?  I'd be happy to help clarify requirements - I think there may be others who would like to do something similar.  I'm not a coder but I'm happy to try things and test things.

Many thanks!
BR
Tim Robinson

andig, 2025-05-17:
I don‘t see how a dimmer fits the evcc device model. Seems you‘ll need to implement a custom Go charger.

RenatusRo, 2025-05-19:
AFAIK using a phase angle controller makes not much sense: 
Yes, you can dim the power going into the heater.
But you cannot control the energy flow: part of the cycle you will export energy (when switching off the heater), part of the cycle you will import. So your grid meter will see both export AND import.

andig, 2025-05-19:
What you're doing seems to be what `type: heatpump` does- controlling output power. If you're able to create a custom configuration that has

	SetMaxPower             plugin.Config
	GetMaxPower             *plugin.Config // optional

what you're trying to do should already work. 

Note I didn't check if your particular Shelly can be configured using evcc templates. If not, a `shelly` plugin might be required. Since this is a bit outside mainstream use the core team likely won't work on this.

RenatusRo, 2025-05-19:
> It does work! I've been doing this for many years with a self-built phase-angle controller, controlled by an arduino to dump power to the hot water based on the state if import/export current. 

Ok, lucky case then, the meter ignores the micro-exports and -imports.

If the heatpump solution doesn't pan out, try this:
1. define a custom charger like

```
  - name: charger_boiler_esp
    type: custom
    status:
       source: mqtt
       topic: esp_boiler/state
    enabled:
       source: mqtt
       topic: esp_boiler/enabled
    enable:
       source: mqtt
       topic: esp_boiler/enable
       payload:
    maxcurrent:
       source: mqtt
       topic: esp_boiler/current
    maxcurrentmillis:
       source: mqtt
       topic: esp_boiler/current
    power:
       source: mqtt
       topic: esp_boiler/power_measured
    features:
      #no vehicle
      - integrateddevice
      #SoC as temperature
      - heating
    icon: waterheater

```

And in home assistant make an automation based on the enable and maxcurrentmillis values to control the shelly dimmer.
In the automation set the values of status, enabled and power to reflect the state of your controller.