evcc-io/evcc #21289: Phasenumschaltung wird trotz phases: 3 und phasesSwitchable: false versucht (Alfen + Modbus)

manes86, 2025-05-18:
### Describe the bug

title: "Phasenumschaltung wird trotz phases: 3 und phasesSwitchable: false versucht (Alfen + Modbus)"

labels:
  - bug
  - modbus
  - charger
  - alfen

area: Charger

version: "0.203.5"

description: |
  Ich verwende evcc 0.203.5 mit einer Alfen Eve Single S-line Wallbox über Modbus TCP.

  Die Wallbox ist **fest auf 3 Phasen konfiguriert**, und ich möchte ausschließlich 3-phasiges Überschussladen betreiben – ohne Phasenumschaltung.

  In meiner `evcc.yaml` ist korrekt `phases: 3` gesetzt und es wird ein angepasstes Template `alfen-fixed` verwendet, das explizit `phasesSwitchable: false` und `phasesConfigured: 3` übergibt.

  Trotzdem versucht evcc weiterhin regelmäßig die Phasen umzuschalten und wirft folgenden Fehler:
switch phases: modbus: exception '3' (illegal data value), function '6'

Dieser Fehler tritt **auch auf**, wenn:

- Die Wallbox in der ACE Installer Software korrekt auf 3-Phasen-Betrieb ohne Umschaltung konfiguriert ist
- Der Ladepunkt korrekt `phases: 3` gesetzt hat
- Das Template korrekt geladen und als `alfen-fixed` eingebunden ist
- Die Datei `alfen-fixed.yaml` korrekt in `/etc/evcc/templates/` liegt **oder** in `~/.evcc/templates/`
- evcc korrekt neu gestartet wurde

### Steps to reproduce

steps_to_reproduce: |
1. evcc v0.203.5 installieren
2. Alfen Eve Single S-line per Modbus TCP (Port 502, ID 1) konfigurieren
3. Wallbox über ACE Installer auf feste 3 Phasen einstellen, Umschaltung deaktivieren
4. Lokales Template anlegen (`alfen-fixed`) mit:
   ```yaml
   templates:
     - name: alfen-fixed
       parent: alfen
       params:
         phasesConfigured: 3
         phasesSwitchable: false
   ```
5. evcc.yaml konfigurieren:
   ```yaml
   chargers:
   - type: template
     template: alfen-fixed
     host: 192.168.178.96
     port: 502
     modbus: tcpip
     id: 1
     name: wallbox5

   loadpoints:
   - title: Garage
     charger: wallbox5
     phases: 3
     mode: pv
   ```
6. evcc starten, Fahrzeug verbinden
7. Fehler erscheint regelmäßig im Log/Dashboard

expected_behavior: |
evcc sollte **keine Umschaltversuche** starten, wenn:
- `phasesSwitchable: false` im Template
- `phases: 3` im Ladepunkt gesetzt ist

actual_behavior: |
evcc versucht dennoch `Function 6`-Modbus-Befehle zur Phasenumschaltung zu senden. Die Wallbox lehnt diese mit `exception '3' (illegal data value)` ab. Dies führt zu fehlerhaften Statusmeldungen im Dashboard und unnötigen Wiederholversuchen.

system_info: |
- evcc version: 0.203.5
- Raspberry Pi OS 32 Bit
- Wallbox: Alfen Eve Single S-line 11kW
- Kommunikation: Modbus TCP (ID 1, Port 502)
- Wechselrichter: Huawei SUN2000-10KTL-M1
- evcc läuft als systemd-Service

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: GESCHWÄRZT

interval: 30s # control cycle interval

sponsortoken: GESCHWÄRZT

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.178.67  
  port: 502  
  usage: grid  
  modbus: tcpip  
  timeout: 15s  
  storageunit: 1  
  name: grid1
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.178.67  
  port: 502  
  usage: pv  
  modbus: tcpip  
  timeout: 15s  
  storageunit: 1  
  name: pv2
- type: template
  template: huawei-sun2000 
  id: 1  
  host: 192.168.178.67  
  port: 502  
  usage: battery  
  modbus: tcpip  
  timeout: 15s  
  storageunit: 1  
  name: battery3

chargers:
- type: template
  template: alfen 
  id: 1  
  host: 192.168.178.96  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: fiat 
  user: GESCHWÄRZT 
  password: GESCHWÄRZT 
  vin: GESCHWÄRZT 
  title: Fiat  
  icon: car  
  capacity: 42  
  phases: 3  
  mode: pv  
  minCurrent: 6  
  maxCurrent: 16  
  identifiers:
  - rfid 
  cache: 30s  
  coarsecurrent: true  
  welcomecharge: false  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv
  phases: 3

site:
  title: GESCHWÄRZT
  meters:
    grid: grid1
    pv:
    - pv2
    battery:
    - battery3
```

### Log details

```text
switch phases: modbus: exception '3' (illegal data value), function '6'
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.5

andig, 2025-05-18:
Zu deinem Template können wir nichts sagen. Wenn im Ladepunkt phases=3 eingestellt ist wird nicht geschaltet. Allerdings versucht der Ladepunkt einmalig beim Start den Zielwert (=3p) einzustellen. Das lässt sich _nur_ verhindern wenn der Charger selbst die Anzahl der aktiven Phasen zurück gibt so dass der Ladepunkt prüfen kann, ob das überhaupt notwendig ist.

manes86, 2025-05-18:
Wie kann ich denn verhindern, dass diese Fehlermeldung kommt und evcc versucht, die Phasen zu switchen?
Auch mit dem Alfen-Standard-Template erschien die Fehlermeldung.
Die Wallbox ist so konfiguriert, dass er fix auf 3 Phasen geschaltet ist.

andig, 2025-05-18:
Einfach der WB das schalten erlauben und den Ladepunkt auf 3p stellen. Den Fehler provozierst Du selbst mit der Einstellung...

closed 2025-05-18 by andig