evcc-io/evcc #21293: Ladung kann nach pause nicht jedes mal gestartet werden

Jrlohni, 2025-05-18:
### Describe the bug

Moin zusammen,

wenn eine längere Ladepause eingelegt wird, beginnt der Dacia an mehreren Wallboxen nicht mehr zu laden.
Das Verhalten ist vor allem aufgefallen, da der Ladeplan nicht mehr absolviert wird und das Auto morgens nicht voll ist.
Soweit ich das noch weiß, war immer ein zweites Auto an einer anderen Wallbox angeschlossen, hat aber nie geladen. Außerdem ist ein Lastmanagement konfiguriert.
1.	Ladung steht auf PV, Ladeplan ist eingestellt auf 4 Uhr morgens, Ladung ist um 18 Uhr abends pausiert worden. Ladung beginnt nicht mit Ladeplan.
2.	Ladung steht auf PV, Ladung ist ca. 4 Stunden vor dem Test pausiert worden, Ladung startet nicht beim Umstellen auf "Schnell".
3.	Ladung steht auf PV, Ladung ist ca. 4 Stunden vor dem Test pausiert worden, Ladung startet bei dem Umstellen auf "Min+PV" und lädt auch auf "Schnell" weiter. (Konnte allerdings nur einmal erreicht werden.)
Anbei ist das Log von einem der Test. Dort wird der Wake-Up-Mode vermutlich nicht lange genug aufrechterhalten, oder? Das war zumindest der einzige Unterschied, welchen ich feststellen konnte bei den Tests.

Grüße Johannes

### Steps to reproduce

1. Zwei Autos an je eine Wallbox anschließen (eventuell reicht auch eins)
2. Laden mit dem Dacia Spring kurz starten, dann pausieren.
3.1 Laden über Ladeplan in 8 Stunden wieder starten lassen.
3.2 Laden nach einigen Stunden auf "Schnell" stellen

### Configuration details

```yaml
charger
---
db:8 {Type:template Title: Icon: Product:Juice Charger Me} map[host:192.168.178.109 port:502 template:bender]
db:13 {Type:template Title: Icon: Product:go-e Charger Gemini, HOME+, HOMEfix (V3)} map[host:192.168.178.31 template:go-e-v3]
db:14 {Type:template Title: Icon: Product:go-e Charger Gemini, HOME+, HOMEfix (V3)} map[host:192.168.178.31 template:go-e-v3]

vehicle
---
db:3 {Type:template Title: Icon: Product:} map[cache:15m capacity:28.3 icon:car maxCurrent:16 minCurrent:6 mode:pv password:***** phases:1 priority:3 template:dacia title:Gret user:*****]
db:6 {Type:template Title: Icon: Product:} map[cache:6e+11 capacity:62 icon:shuttle maxCurrent:16 minCurrent:6 mode:minpv password:***** priority:1 template:enyaq timeout:10s title:Muddi user:*****]
db:7 {Type:template Title: Icon: Product:} map[accessToken:******* cache:15m capacity:79 clientId:******* commandProxy:https://api.myteslamate.com icon:van maxCurrent:16 minCurrent:6 mode:minpv priority:2 refreshToken:******************* template:tesla title:Guido]

loadpoint
---
db:9 {Type: Title: Icon: Product:} map[charger:db:8 circuit:main defaultMode:pv limitSoc:0 maxCurrent:16 meter: mode:pv phasesConfigured:3 soc:map[estimate:true poll:map[interval:9e+11 mode:connected]] thresholds:map[disable:map[delay:3e+11 threshold:0] enable:map[delay:9e+10 threshold:-1400]] title:Hof (Schwarz) vehicle:]
db:15 {Type: Title: Icon: Product:} map[charger:db:14 circuit:main defaultMode:pv limitSoc:0 maxCurrent:16 meter: minCurrent:6 mode:pv phasesConfigured:1 priority:0 soc:map[estimate:true poll:map[interval:9e+11 mode:connected]] thresholds:map[disable:map[delay:3e+11 threshold:0] enable:map[delay:9e+10 threshold:-1400]] title:Garage (Gelb) vehicle:]
```

### Log details

```text
[site  ] DEBUG 2025/05/15 05:18:13 pv 1 power: 0W
[site  ] DEBUG 2025/05/15 05:18:13 grid power: 4465W
[site  ] DEBUG 2025/05/15 05:18:13 site power: 4465W
[lp-2  ] DEBUG 2025/05/15 05:18:13 charge total import: 18685.901kWh
[lp-2  ] DEBUG 2025/05/15 05:18:13 charger status: B
[lp-2  ] DEBUG 2025/05/15 05:18:13 next soc poll remaining time: 11m56s
[lp-2  ] DEBUG 2025/05/15 05:18:13 plan: charge 54m21s between 2025-05-15 05:18:13 +0200 CEST until 2025-05-15 06:30:00 +0200 CEST (power: 11040W, avg cost: 0.290)
[circuit-main] DEBUG 2025/05/15 05:18:13 validate current: 16A + (0A -> 16A) > 17A capped at 1A
[lp-2  ] DEBUG 2025/05/15 05:18:13 charger disable
[lp-2  ] DEBUG 2025/05/15 05:18:13 wake-up timer: stop
[site  ] DEBUG 2025/05/15 05:18:13 solar forecast: accumulated 619.386kWh, produced 997.883kWh, scale 1.611
[site  ] DEBUG 2025/05/15 05:18:14 ----
[lp-1  ] DEBUG 2025/05/15 05:18:14 charge power: 0W
[lp-2  ] DEBUG 2025/05/15 05:18:15 charge power: 0W
[lp-2  ] DEBUG 2025/05/15 05:18:15 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/15 05:18:15 power: 0W
[circuit-main] DEBUG 2025/05/15 05:18:15 current: 16A
[site  ] DEBUG 2025/05/15 05:18:15 pv 1 power: 0W
[site  ] DEBUG 2025/05/15 05:18:15 grid power: 4463W
[site  ] DEBUG 2025/05/15 05:18:15 site power: 4463W
[lp-1  ] DEBUG 2025/05/15 05:18:15 charger status: B
[lp-1  ] DEBUG 2025/05/15 05:18:15 next soc poll remaining time: 2m19s
[lp-1  ] DEBUG 2025/05/15 05:18:15 plan: charge 25m18s between 2025-05-15 05:18:15 +0200 CEST until 2025-05-15 06:00:00 +0200 CEST (power: 3680W, avg cost: 0.284)
[circuit-main] DEBUG 2025/05/15 05:18:15 validate current: 16A + (0A -> 16A) > 17A capped at 1A
[lp-1  ] DEBUG 2025/05/15 05:18:15 charger disable
[... 881 lines cut ...]
[circuit-main] DEBUG 2025/05/15 05:20:03 power: 0W
[circuit-main] DEBUG 2025/05/15 05:20:03 current: 0A
[site  ] DEBUG 2025/05/15 05:20:03 pv 1 power: 0W
[site  ] DEBUG 2025/05/15 05:20:03 grid power: 4220W
[site  ] DEBUG 2025/05/15 05:20:03 site power: 4220W
[lp-1  ] DEBUG 2025/05/15 05:20:03 charger status: B
[lp-1  ] DEBUG 2025/05/15 05:20:03 next soc poll remaining time: 32s
[lp-1  ] DEBUG 2025/05/15 05:20:03 plan: charge 25m18s between 2025-05-15 05:20:03 +0200 CEST until 2025-05-15 06:00:00 +0200 CEST (power: 3680W, avg cost: 0.284)
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

203.5

Jrlohni, 2025-05-18:
Zur Vollständigkeit noch mal die andere Ausgabe der Config (mit der konnte ich den Issue nicht erstellen):

`[main  ] INFO 2025/05/18 16:55:39 evcc 0.203.5
[main  ] INFO 2025/05/18 16:55:39 using config file: /etc/evcc.yaml
[lp-1  ] DEBUG 2025/05/18 16:55:42 set title: Hof (Schwarz)
[lp-1  ] DEBUG 2025/05/18 16:55:44 set priority: 0
[lp-1  ] DEBUG 2025/05/18 16:55:44 set smart cost limit: <nil>
[lp-1  ] DEBUG 2025/05/18 16:55:44 set thresholds: {Enable:{Delay:1m30s Threshold:-1400} Disable:{Delay:5m0s Threshold:0}}
[lp-1  ] DEBUG 2025/05/18 16:55:46 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-1  ] DEBUG 2025/05/18 16:55:46 set session energy limit: 0
[lp-1  ] DEBUG 2025/05/18 16:55:46 set session soc limit: 0
[lp-1  ] DEBUG 2025/05/18 16:55:46 set soc config: {Poll:{Mode:connected Interval:15m0s} Estimate:0xc000f7f619}
[lp-1  ] DEBUG 2025/05/18 16:55:47 set default mode: pv
[lp-1  ] DEBUG 2025/05/18 16:55:47 set phases: 3
[lp-1  ] DEBUG 2025/05/18 16:55:47 set max current: 16
[lp-2  ] DEBUG 2025/05/18 16:55:47 set title: Garage (Gelb)
[lp-2  ] DEBUG 2025/05/18 16:55:49 set priority: 0
[lp-2  ] DEBUG 2025/05/18 16:55:49 set smart cost limit: <nil>
[lp-2  ] DEBUG 2025/05/18 16:55:49 set thresholds: {Enable:{Delay:1m30s Threshold:-1400} Disable:{Delay:5m0s Threshold:0}}
[lp-2  ] DEBUG 2025/05/18 16:55:51 set plan energy: 0kWh @ 0001-01-01 00:53:28 +0053 LMT
[lp-2  ] DEBUG 2025/05/18 16:55:51 set session energy limit: 0
[lp-2  ] DEBUG 2025/05/18 16:55:51 set session soc limit: 0
[lp-2  ] DEBUG 2025/05/18 16:55:51 set soc config: {Poll:{Mode:connected Interval:15m0s} Estimate:0xc000c83d49}
[lp-2  ] DEBUG 2025/05/18 16:55:53 set default mode: pv
[lp-2  ] DEBUG 2025/05/18 16:55:53 set phases: 1
[lp-2  ] DEBUG 2025/05/18 16:55:53 set min current: 6
[lp-2  ] DEBUG 2025/05/18 16:55:53 set max current: 16
[lp-2  ] DEBUG 2025/05/18 16:55:55 charge total import: 18783.976kWh

Konfiguration (/etc/evcc.yaml):

network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

#interval: 2s # control cycle interval. Interval <30s can lead to unexpected behavior, see https://docs.evcc.io/docs/reference/configuration/interval

# database configuration for persisting charge sessions and settings
# database:
#   type: sqlite
#   dsn: <path-to-db-file>

# sponsor token: *****
# trial token, valid until 2024-08-23
#sponsortoken: *****
sponsortoken: *****

# telemetry enables aggregated statistics
#
# Telemetry allows collecting usage data (grid and green energy, charge power).
# Data is aggregated, no individual charging sessions are tracked. The collected,
# anonymous data can be retrieved using https://api.evcc.io.
#
# See https://github.com/evcc-io/evcc/pull/4343 or details.
#
# For time being, this is only available to sponsors, hence data is associated with
# the sponsor token's identity.
#
# telemetry: true

# log settings
log: info
[... 366 lines cut ...]
      msg: Car disconnected after ${connectedDuration}
    soc: # vehicle soc update event
      title: Soc updated
      msg: Battery charged to ${vehicleSoc:%.0f}%
    guest: # vehicle could not be identified
      title: Unknown vehicle
      msg: Unknown vehicle, guest connected?
  services:
  # - type: pushover
  #   app: *****
  #   recipients: *****
  #   - # list of recipient ids
  # - type: telegram
  #   token: *****
  #   chats: *****
  #   - # list of chat ids
  # - type: email
  #   uri: smtp://<user: *****
  # - type: ntfy
  #   uri: https://<host>/<topics>
  #   priority: <priority>
  #   tags: <tags>

Version: `0.203.5`
`

Jrlohni, 2025-05-18:
Ich hätte noch eine größere Konfigurationsdatei, welche ich direkt im Issue nicht Posten konnte. Leider bekomme ich die Formatierung hier auch nicht vernünftig hin, damit Sie hier gut zu lesen ist (nicht als "Code", nicht sonst wie...) daher hier als Datei.

[Konfig_EVCC.txt](https://github.com/user-attachments/files/20273556/Konfig_EVCC.txt)

mfuchs1984, 2025-05-18:
Du hast wie es aussieht ein interval von 2 Sekunden. So schnell reagiert keine Wallbox. Stell das mal auf 30 wie empfohlen.

Jrlohni, 2025-05-18:
Guter Hinweis! Das teste ich mal!
Jedoch hat es jetzt ein Jahr ohne Probleme auf 2 Sekunden und auch einer Sekunde funktioniert. Außerdem funktioniert es auch jetzt immer wieder so, nur nach einer längeren Ladepause nicht mehr.

VolkerK62, 2025-05-18:
Für beide Loadpoints ist ein Ladeplan hinterlegt
```
[lp-1  ] DEBUG 2025/05/15 05:19:55 plan: charge 25m18s between 2025-05-15 05:19:55 +0200 CEST until 2025-05-15 06:00:00 +0200 CEST (power: 3680W, avg cost: 0.284)
[lp-2  ] DEBUG 2025/05/15 05:19:57 plan: charge 54m21s between 2025-05-15 05:19:57 +0200 CEST until 2025-05-15 06:30:00 +0200 CEST (power: 11040W, avg cost: 0.291)
```
Beide wollen laden, können aber nicht, wegen Lastmanagement.
`[circuit-main] DEBUG 2025/05/15 05:19:57 validate current: 16A + (0A -> 16A) > 17A capped at 1A`

Jrlohni, 2025-05-18:
Oh sorry Leute, da hab ich ein Log zu spät gegriffen... Dort habe ich weitere Tests durchgeführt.

Hier ist eins, wo nur der LP-1 einen Plan hat. Gleiches Verhalten in EVCC natürlich.

```[site  ] DEBUG 2025/05/07 22:42:40 update repeating plans for db:3 to: [{[1 2 3 4 5] 06:00 Europe/Berlin 45 0 true} {[1 2 3 4 5] 23:12 Europe/Berlin 35 0 true}]
[site  ] DEBUG 2025/05/07 22:42:40 ----
[lp-1  ] DEBUG 2025/05/07 22:42:40 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:41 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:41 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/07 22:42:41 power: 0W
[circuit-main] DEBUG 2025/05/07 22:42:41 current: 0A
[site  ] DEBUG 2025/05/07 22:42:41 pv 1 power: 0W
[site  ] DEBUG 2025/05/07 22:42:41 grid power: 1564W
[site  ] DEBUG 2025/05/07 22:42:41 site power: 1564W
[lp-2  ] DEBUG 2025/05/07 22:42:41 charge total import: 18589.852kWh
[lp-2  ] DEBUG 2025/05/07 22:42:41 charger status: B
[lp-2  ] DEBUG 2025/05/07 22:42:41 pv charge current: 0A = 0A + -6.8A (1564W @ 1p)
[site  ] DEBUG 2025/05/07 22:42:41 solar forecast: accumulated 97.240kWh, produced 182.542kWh, scale 1.877
[site  ] DEBUG 2025/05/07 22:42:42 ----
[lp-1  ] DEBUG 2025/05/07 22:42:42 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:43 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:43 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/07 22:42:43 power: 0W
[circuit-main] DEBUG 2025/05/07 22:42:43 current: 0A
[site  ] DEBUG 2025/05/07 22:42:43 pv 1 power: 0W
[site  ] DEBUG 2025/05/07 22:42:43 grid power: 1583W
[site  ] DEBUG 2025/05/07 22:42:43 site power: 1583W
[lp-1  ] DEBUG 2025/05/07 22:42:43 charger status: B
[lp-1  ] DEBUG 2025/05/07 22:42:43 next soc poll remaining time: 14m15s
[lp-1  ] DEBUG 2025/05/07 22:42:43 plan: charge 1h1m31s between 2025-05-07 22:42:43 +0200 CEST until 2025-05-07 23:12:00 +0200 CEST (overruns by 32m14s, power: 3680W, avg cost: 0.284)
[lp-1  ] DEBUG 2025/05/07 22:42:43 set charge current limit: 16A
[lp-1  ] DEBUG 2025/05/07 22:42:43 charger enable
[lp-1  ] DEBUG 2025/05/07 22:42:43 wake-up timer: start
[lp-1  ] DEBUG 2025/05/07 22:42:43 pv timer elapse
[lp-1  ] DEBUG 2025/05/07 22:42:43 pv timer inactive
[site  ] DEBUG 2025/05/07 22:42:43 solar forecast: accumulated 97.240kWh, produced 182.542kWh, scale 1.877
[site  ] DEBUG 2025/05/07 22:42:44 ----
[lp-1  ] DEBUG 2025/05/07 22:42:44 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:45 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:45 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/07 22:42:45 power: 0W
[circuit-main] DEBUG 2025/05/07 22:42:45 current: 16A
[site  ] DEBUG 2025/05/07 22:42:45 pv 1 power: 0W
[site  ] DEBUG 2025/05/07 22:42:45 grid power: 1571W
[site  ] DEBUG 2025/05/07 22:42:45 site power: 1571W
[lp-2  ] DEBUG 2025/05/07 22:42:45 charge total import: 18589.852kWh
[lp-2  ] DEBUG 2025/05/07 22:42:45 charger status: B
[lp-2  ] DEBUG 2025/05/07 22:42:45 pv charge current: 0A = 0A + -6.83A (1571W @ 1p)
[site  ] DEBUG 2025/05/07 22:42:45 solar forecast: accumulated 97.240kWh, produced 182.542kWh, scale 1.877
[site  ] DEBUG 2025/05/07 22:42:46 ----
[lp-1  ] DEBUG 2025/05/07 22:42:46 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:47 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:47 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/07 22:42:47 power: 0W
[circuit-main] DEBUG 2025/05/07 22:42:47 current: 16A
[site  ] DEBUG 2025/05/07 22:42:47 pv 1 power: 0W
[site  ] DEBUG 2025/05/07 22:42:47 grid power: 1583W
[site  ] DEBUG 2025/05/07 22:42:47 site power: 1583W
[lp-1  ] DEBUG 2025/05/07 22:42:47 charger status: B
[lp-1  ] DEBUG 2025/05/07 22:42:47 next soc poll remaining time: 14m11s
[lp-1  ] DEBUG 2025/05/07 22:42:47 plan: charge 1h1m31s between 2025-05-07 22:42:47 +0200 CEST until 2025-05-07 23:12:00 +0200 CEST (overruns by 32m18s, power: 3680W, avg cost: 0.284)
[circuit-main] DEBUG 2025/05/07 22:42:47 validate current: 16A + (0A -> 16A) > 17A capped at 1A
[lp-1  ] DEBUG 2025/05/07 22:42:47 charger disable
[lp-1  ] DEBUG 2025/05/07 22:42:47 wake-up timer: stop
[site  ] DEBUG 2025/05/07 22:42:47 solar forecast: accumulated 97.240kWh, produced 182.542kWh, scale 1.877
[site  ] DEBUG 2025/05/07 22:42:48 ----
[lp-1  ] DEBUG 2025/05/07 22:42:48 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:49 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:42:49 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/07 22:42:49 power: 0W
[... 187 lines cut ...]
[circuit-main] DEBUG 2025/05/07 22:43:15 current: 0A
[site  ] DEBUG 2025/05/07 22:43:15 pv 1 power: 0W
[site  ] DEBUG 2025/05/07 22:43:15 grid power: 1676W
[site  ] DEBUG 2025/05/07 22:43:15 site power: 1676W
[lp-1  ] DEBUG 2025/05/07 22:43:15 charger status: B
[lp-1  ] DEBUG 2025/05/07 22:43:15 next soc poll remaining time: 13m43s
[lp-1  ] DEBUG 2025/05/07 22:43:15 plan: charge 1h1m31s between 2025-05-07 22:43:15 +0200 CEST until 2025-05-07 23:12:00 +0200 CEST (overruns by 32m46s, power: 3680W, avg cost: 0.284)
[lp-1  ] DEBUG 2025/05/07 22:43:15 set charge current limit: 16A
[lp-1  ] DEBUG 2025/05/07 22:43:15 charger enable
[lp-1  ] DEBUG 2025/05/07 22:43:15 wake-up timer: start
[site  ] DEBUG 2025/05/07 22:43:15 solar forecast: accumulated 97.240kWh, produced 182.542kWh, scale 1.877
[site  ] DEBUG 2025/05/07 22:43:16 ----
[lp-1  ] DEBUG 2025/05/07 22:43:16 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:43:17 charge power: 0W
[lp-2  ] DEBUG 2025/05/07 22:43:17 charge currents: [0 0 0]A
[circuit-main] DEBUG 2025/05/07 22:43:17 power: 0W
[... cut ...] 05/07 22:43:17 current: 16A
[site  ] DEBUG 2025/05/07 22:43:17 pv 1 power: 0W
[site  ] DEBUG 2025/05/07 22:43:17 grid power: 1674W
[site  ] DEBUG 2025/05/07 22:43:17 site power: 1674W
[lp-2  ] DEBUG 2025/05/07 22:43:17 charge total import: 18589.852kWh
[lp-2  ] DEBUG 2025/05/07 22:43:17 charger status: B
[lp-2  ] DEBUG 2025/05/07 22:43:17 pv charge current: 0A = 0A + -7.28A (1674W @ 1p)
[site  ] DEBUG 2025/05/07 22:43:17 solar forecast: accumulated 97.240kWh, produced 182.542kWh, scale 1.877
```

mfuchs1984, 2025-05-18:
Hier spuckt auch das Lastmanagement rein, ich denke weil dein charger kein meter hat. Stell erstmal auf 30 Sekunden um und mach dann nochmal ein Log.