evcc-io/evcc #21312: Laden bricht nach 90 oder 120 sec. ab

rmuelle3, 2025-05-19:
### Describe the bug

Nach Anschliessen eines Fahrzeuges beginnt der Ladevorgang wie erwartet (im Log um 13:06:45). Nach 90 oder 120 sec wird eine Warnmeldung ausgegeben "charger logic error: disabled but charging" (siehe Log 13:08:45). 30 sec. später bricht der Ladevorgang ab (13:09:15). Das System versucht weiterhin zu laden (siehe Screenshot: "Ladebereit. Warte auf Fahrzeug ..."), kann aber keine Verbindung zum Fahrzeug herstellen.

Ein neuer Ladevorgang wird erst gestartet, wenn das Ladekabel ausgesteckt und wieder eingesteckt wird. Es erfolgt erneut ein Abbruch nach 90 oder 120 sec. Das Verhalten passiert bei jedem Lademodus (PV, PV+Min,, Schnell).

Der Fehler tritt seit Release 0.203.4 auf. Es gab zu diesem Zeitpunkt auch ein Betriebssystem-Update vom Fahrzeug.

[evcc-20250519-130941-debug.log](https://github.com/user-attachments/files/20297222/evcc-20250519-130941-debug.log)

[image]

### Steps to reproduce

1. Ladekabel von Fahrzeug anschliessen
2. Start Charging
3. Nach 90 oder 120 sec: Warnmeldung "charger logic error: disabled but charging"
4. Nach weiteren 30 sec: Abbruch Ladevorgang
5. Ladevorgang kann nicht hergestellt werden (warten auf Fahrzeug)
...

### Configuration details

```yaml
Konfiguration:
EVCC (Release 0.203.6) auf Linux-Server (debian 12 bookworm, Intel N150 Prozessor, 16 GB Speicher)
SonnenCharger (Modell G-HB1S3CZ10-A02, HW-Version 1.3, SW-Version 4.0.2@sonn)
BMW i4 (SW-Version 03/2025.47)

evcc.yaml:
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant: xxx

interval: 30s # control cycle interval

sponsortoken: xxx

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: true

meters:
- type: template
  template: sonnenbatterie 
  usage: grid  
  host: 192.168.178.124  
  port: 8080  
  name: grid1
- type: template
  template: sonnenbatterie 
  usage: pv  
  host: 192.168.178.124  
  port: 8080  
  name: pv2
- type: template
  template: sonnenbatterie 
  usage: battery  
  host: 192.168.178.124  
  port: 8080  
  name: battery3
[... 225 lines cut ...]
[lp-1  ] DEBUG 2025/05/19 13:09:15 charger status: B
[lp-1  ] INFO 2025/05/19 13:09:15 stop charging <-
[lp-1  ] DEBUG 2025/05/19 13:09:16 vehicle status: B
[lp-1  ] DEBUG 2025/05/19 13:09:16 vehicle soc: 85%
[lp-1  ] DEBUG 2025/05/19 13:09:16 vehicle soc limit: 100%
[lp-1  ] DEBUG 2025/05/19 13:09:16 vehicle range: 329km
[lp-1  ] DEBUG 2025/05/19 13:09:16 pv charge current: 23.8A = 0A + 23.8A (-16456W @ 3p)
[site  ] DEBUG 2025/05/19 13:09:16 solar forecast: accumulated 57.344kWh, produced 73.436kWh, scale 1.281
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

mfuchs1984, 2025-05-19:
Ich denke, ein längeres Log ist notwendig. Hier sieht man nur, wie die Wallbox von selbst startet und evcc sie wieder einfängt. Darauffolgend würde ich wieder eine Startversuch erwarten, den enthält dein kurzes Log aber leider nicht.
Was zeigt die BMW App an?

StefanSchoof, 2025-05-19:
Ähnliches Problem unter https://github.com/evcc-io/evcc/discussions/21304

mfuchs1984, 2025-05-19:
> Der Fehler tritt seit Release 0.203.4 auf. 

Ist das denn reproduzierbar? Wenn du auf 0.203.3 gehst, gehst es wieder?

rmuelle3, 2025-05-19:
> Ich denke, ein längeres Log ist notwendig. Hier sieht man nur, wie die Wallbox von selbst startet und evcc sie wieder einfängt. Darauffolgend würde ich wieder eine Startversuch erwarten, den enthält dein kurzes Log aber leider nicht. Was zeigt die BMW App an?

Die BMW App zeigt an, dass das Laden pausiert ist. Das anhängende Logfile startet bereits um 10:41 und damit lange vor dem Anschliessen des Ladekabels um 13:06:45.

mfuchs1984, 2025-05-19:
Der Anfang ist nicht da Problem, sondern es endet zu früh.

rmuelle3, 2025-05-19:
> > Der Fehler tritt seit Release 0.203.4 auf.
> 
> Ist das denn reproduzierbar? Wenn du auf 0.203.3 gehst, geht es wieder?

Bei Release 0.203.3 konnte ich noch ohne Einschränkung laden. Ich habe versucht das Release zurückzusetzen; der Fehler trat unverändert auf. Die Umgebung lässt sich leider nicht vollständig zurücksetzen, da BMW ein neues Release ausgeliefert hat und auch auf der Linux-Seite einige Updates automatisch eingespielt wurden.

Die Warnmeldung "charger logic error: disabled but charging" habe ich schon länger beobachtet (seit mehreren Wochen) - es gab aber nie einen Abbruch und der Ladevorgang lief trotzdem ohne Probleme weiter,

rmuelle3, 2025-05-19:
> Aktuelle charger Firmware drauf? https://etrel.atlassian.net/wiki/spaces/Home/pages/2836168744/Etrel+INCH+Firmware

Das kann ich leider nicht feststellen. Die Fa. Sonnen spielt auf dem Etrel-Charger eine eigene Firmware auf und riegelt die Wallbox komplett für den Endbenutzer ab, Es ist kein Login möglich.

StefanSchoof, 2025-05-19:
> Die Fa. Sonnen

Achso Sonnen, dann schau mal unter #18499

rmuelle3, 2025-05-19:
> > Der frühzeitige Abbruch des Ladevorgangs ist ein neues Phänomen.
> 
> Gibts ein altes und ein neues Logfile im Vergleich? Was ist anders?
> 
Ein altes Logfile habe ich nicht; die alten Log-Datensätze sind auch nicht mehr im System verfügbar

> > Der Fehler tritt seit Release 0.203.4 auf. Es gab zu diesem Zeitpunkt auch ein Betriebssystem-Update vom Fahrzeug.
> 
> Daher: bitte analysieren ob das oder die evcc Version ursächliich ist.
Ich habe EVCC versuchsweise auf Release 0.203.4 (von 0.203.5) zurückgesetzt. Der Ladeabbruch war trotzdem da. Während des Releases 0.203.4 gab es einen Update der BMW-Software und mehrere Linux-Updates. Eventuell gab es auch ein Update von Sonnen (was ich leider nicht überprüfen kann), da es seit der Auslieferung der neuen Sonnen-Firmware (Dezember 2024) etliche Updates mit Bugfixes von der Sonnen-App gab.

andig, 2025-05-19:
> Ich habe EVCC versuchsweise auf Release 0.203.4 (von 0.203.5) zurückgesetzt. Der Ladeabbruch war trotzdem da.

Dann liegts also nicht an der evcc Version. Da können wir leider nicht helfen :(

closed 2025-05-19 by andig

andig, 2025-05-19:
Ich sehe keine Warnmeldung?

rmuelle3, 2025-05-19:
> Ich sehe keine Warnmeldung?

13:08:45 im Log

mfuchs1984, 2025-05-19:
Wie gesagt solltest du ein logfile erstellen, das zeigt, was danach passiert. Die Meldung ist erstmal erwartungsgemäß, da die Wallbox ein Eigenleben führt.

mfuchs1984, 2025-05-19:
Und testweise ggf. mal mit evcc 0.203.1 probieren.

rmuelle3, 2025-05-19:
> Wie gesagt solltest du ein logfile erstellen, das zeigt, was danach passiert. Die Meldung ist erstmal erwartungsgemäß, da die Wallbox ein Eigenleben führt.

Ich werde beim nächsten Ladevorgang das Logfile etwas länger laufen lassen und das neue Logfile hier einstellen. Das wird vermutlich übermorgen sein.

Könnt ihr das Issue wieder öffnen ? 
Danke.

andig, 2025-05-19:
Ich würde mich eher fragen, was hier Ursache und Wirkung ist:

```
[lp-1  ] DEBUG 2025/05/19 13:08:45 charge power: 0W
[lp-1  ] DEBUG 2025/05/19 13:08:45 charge currents: [0 0 0]A
[lp-1  ] WARN 2025/05/19 13:08:45 charger logic error: disabled but charging
```

Warum hört das Ding einfach auf? Wie schafft es die Box, laut Status

	// 4 Charging

und gleichzeitig Vorgabestrom 0 zu reporten?

Ich würde mal vorschlagen, evcc abzuschalten und alle Funktionen der WB mittels

    evcc charger

zu testen. 

Ein Logikfehler lässt sich nicht ausschließen, offensichtlich ist er aber nicht zu erkennen...

jomuench, 2025-05-21:
> Ähnliches Problem unter [#21304](https://github.com/evcc-io/evcc/discussions/21304)

mein charger (ebenfalls Sonnencharger, Hersteller ETREL) meldet:
`[lp-1 ] DEBUG 2025/05/21 08:56:48 charge power: 3136W
[lp-1 ] DEBUG 2025/05/21 08:56:48 charge currents: [4.98 5.18 4.54]A `

während das Fahrzeug sagt:

[image]

und irgendwelche ominösen Stromflüsse werden angezeigt:

[image]

Wenn ich den Charger per Sicherung kurz stromlos mache, fängt sich das ganze System und lädt und zeigt auch reale Werte an.

mfuchs1984, 2025-05-21:
Sofern das mal funktioniert hat, vermute ich, die wallbox hat ein Update bekommen, das die Modbus Schnittstelle verändert hat.

jomuench, 2025-05-21:
ja, das hat seit mehr als einem Jahr funktioniert.
Wäre es möglich, evcc an die geänderte Modbus Schnittstelle anzupassen?

jomuench, 2025-05-21:
wo mein Verständnis noch hakt: warum funktioniert das System nach einem reset der Wallbox.

mfuchs1984, 2025-05-21:
Der Neustart korrigiert ggf. eine Inkonsistenz im Status.

Ein modbus trace log, einmal vor dem Neustart mit dem Fehler und einmal danach ohne Fehler könnte ggf. helfen, einen Workaround zu finden.

jomuench, 2025-05-21:
puh. Wie komme ich an das modbus trace log dran? Die Wallbox ist closed, gibt es ein API oder reicht ein tcpdump evcc <> Wallbox?

mfuchs1984, 2025-05-21:
Evcc kann diese erstellen.

```
log: debug
levels:
  cache: error
  etrel: trace
```

rmuelle3, 2025-05-21:
Ich habe das von jomuench beschriebene Verhalten an meiner Wallbox getestet und kann dies bestätigen. Nach kurzzeitigem Abschalten der Wallbox im Sicherungskasten wird das Laden korrekt fortgesetzt und nicht mehr unterbrochen.

Hier ist der Log (EVCC-Release 0.203.6):
Ereignisse: 
10:13:40 Auto angeschlossen
10:15:41 Stop Charging
10:15:42 Warnmeldung: Charger out of sync (diesmal eine andere Meldung ...)
10:18:41 Errormeldung: Wake-Up Vehicle: unexpected status --> nach Aus-und Einschalten der Sicherung
10:19:40 Warnmeldung: Charger logic error

[lp-1 ] DEBUG 2025/05/21 10:13:10 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 10:13:10 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 10:13:11 grid power: -8723W
[site ] DEBUG 2025/05/21 10:13:11 pv 1 power: 13358W
[site ] DEBUG 2025/05/21 10:13:11 battery 1 power: 1W
[site ] DEBUG 2025/05/21 10:13:11 battery 1 soc: 82%
[site ] DEBUG 2025/05/21 10:13:11 site power: -8622W
[lp-1 ] DEBUG 2025/05/21 10:13:11 charger status: A
[site ] DEBUG 2025/05/21 10:13:11 solar forecast: accumulated 51.816kWh, produced 131.140kWh, scale 2.531
[site ] DEBUG 2025/05/21 10:13:40 ----
[lp-1 ] DEBUG 2025/05/21 10:13:40 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 10:13:40 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 10:13:40 grid power: -10404W
[site ] DEBUG 2025/05/21 10:13:40 battery 1 power: 0W
[site ] DEBUG 2025/05/21 10:13:40 pv 1 power: 14403W
[site ] DEBUG 2025/05/21 10:13:40 battery 1 soc: 82%
[site ] DEBUG 2025/05/21 10:13:40 site power: -10304W
[lp-1 ] DEBUG 2025/05/21 10:13:40 charger status: B
[lp-1 ] INFO 2025/05/21 10:13:40 car connected
[lp-1 ] DEBUG 2025/05/21 10:13:40 vehicle api refresh
[lp-1 ] DEBUG 2025/05/21 10:13:40 pv timer elapse
[lp-1 ] DEBUG 2025/05/21 10:13:40 pv timer inactive
[main ] DEBUG 2025/05/21 10:13:42 vehicle status: B (BMW i4)
[lp-1 ] INFO 2025/05/21 10:13:42 vehicle updated: unknown -> BMW i4
[lp-1 ] DEBUG 2025/05/21 10:13:42 vehicle status: B
[lp-1 ] DEBUG 2025/05/21 10:13:42 vehicle soc: 62%
[lp-1 ] DEBUG 2025/05/21 10:13:42 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 10:13:42 vehicle range: 184km
[lp-1 ] DEBUG 2025/05/21 10:13:42 pv charge current: 14.9A = 0A + 14.9A (-10304W @ 3p)
[lp-1 ] DEBUG 2025/05/21 10:13:42 site power -10304W <= 0W enable threshold
[lp-1 ] DEBUG 2025/05/21 10:13:42 pv enable in 0s
[lp-1 ] DEBUG 2025/05/21 10:13:42 pv enable timer elapsed
[lp-1 ] DEBUG 2025/05/21 10:13:42 pv timer reset
[lp-1 ] DEBUG 2025/05/21 10:13:42 pv timer inactive
[lp-1 ] DEBUG 2025/05/21 10:13:42 set charge current limit: 6A
[lp-1 ] DEBUG 2025/05/21 10:13:42 charger enable
[lp-1 ] DEBUG 2025/05/21 10:13:42 wake-up timer: start
[site ] DEBUG 2025/05/21 10:13:42 solar forecast: accumulated 51.845kWh, produced 131.260kWh, scale 2.532
[lp-1 ] DEBUG 2025/05/21 10:13:45 set session soc limit: 80
[site ] DEBUG 2025/05/21 10:13:45 ----
[lp-1 ] DEBUG 2025/05/21 10:13:45 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 10:13:45 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 10:13:46 battery 1 power: 3W
[site ] DEBUG 2025/05/21 10:13:46 grid power: -10164W
[site ] DEBUG 2025/05/21 10:13:46 pv 1 power: 14532W
[site ] DEBUG 2025/05/21 10:13:46 battery 1 soc: 82%
[site ] DEBUG 2025/05/21 10:13:46 site power: -10061W
[lp-1 ] DEBUG 2025/05/21 10:13:46 vehicle odometer: 10901km
[lp-1 ] DEBUG 2025/05/21 10:13:46 charger status: B
[lp-1 ] DEBUG 2025/05/21 10:13:46 pv charge current: 14.6A = 0A + 14.6A (-10061W @ 3p)
[lp-1 ] DEBUG 2025/05/21 10:13:46 set charge current limit: 14.6A
[site ] DEBUG 2025/05/21 10:13:46 solar forecast: accumulated 51.849kWh, produced 131.281kWh, scale 2.532
[site ] DEBUG 2025/05/21 10:14:10 ----
[lp-1 ] DEBUG 2025/05/21 10:14:10 charge power: 8448W
[lp-1 ] DEBUG 2025/05/21 10:14:10 charge currents: [13 13 12.9]A
[site ] DEBUG 2025/05/21 10:14:10 grid power: -2015W
[site ] DEBUG 2025/05/21 10:14:10 battery 1 power: -1W
[site ] DEBUG 2025/05/21 10:14:10 pv 1 power: 14689W
[site ] DEBUG 2025/05/21 10:14:10 battery 1 soc: 82%
[site ] DEBUG 2025/05/21 10:14:10 site power: -1916W
[... 231 lines cut ...]
[site ] DEBUG 2025/05/21 10:21:40 site power: -1729W
[lp-1 ] DEBUG 2025/05/21 10:21:40 detected active phases: 3p
[lp-1 ] DEBUG 2025/05/21 10:21:40 charger status: C
[lp-1 ] DEBUG 2025/05/21 10:21:40 soc estimated: 62.52% (vehicle: 62.00%)
[lp-1 ] DEBUG 2025/05/21 10:21:40 vehicle soc: 63%
[lp-1 ] DEBUG 2025/05/21 10:21:40 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 10:21:40 vehicle range: 185km
[lp-1 ] DEBUG 2025/05/21 10:21:40 pv charge current: 18.5A = 16A + 2.51A (-1729W @ 3p)
[site ] DEBUG 2025/05/21 10:21:40 solar forecast: accumulated 52.294kWh, produced 133.203kWh, scale 2.547
[site ] DEBUG 2025/05/21 10:22:10 ----
[lp-1 ] DEBUG 2025/05/21 10:22:10 charge power: 9792W
[lp-1 ] DEBUG 2025/05/21 10:22:10 charge currents: [14.7 14.9 14.6]A
[site ] DEBUG 2025/05/21 10:22:10 grid power: -1824W
[site ] DEBUG 2025/05/21 10:22:11 battery 1 power: 0W
[site ] DEBUG 2025/05/21 10:22:11 pv 1 power: 12205W
[site ] DEBUG 2025/05/21 10:22:11 battery 1 soc: 82%
[site ] DEBUG 2025/05/21 10:22:11 site power: -1724W
[lp-1 ] DEBUG 2025/05/21 10:22:11 detected active phases: 3p
[lp-1 ] DEBUG 2025/05/21 10:22:11 charger status: C
[lp-1 ] DEBUG 2025/05/21 10:22:11 soc estimated: 62.61% (vehicle: 62.00%)
[lp-1 ] DEBUG 2025/05/21 10:22:11 vehicle soc: 63%
[lp-1 ] DEBUG 2025/05/21 10:22:11 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 10:22:11 vehicle range: 185km
[lp-1 ] DEBUG 2025/05/21 10:22:11 pv charge current: 18.5A = 16A + 2.5A (-1724W @ 3p)
[site ] DEBUG 2025/05/21 10:22:11 solar forecast: accumulated 52.322kWh, produced 133.305kWh, scale 2.548

rmuelle3, 2025-05-21:
Noch ein Versuch (EVCC Release 0.203.6). Diesmal das Log mit modbus-Trace

[site ] DEBUG 2025/05/21 10:57:49 ----
[etrel ] TRACE 2025/05/21 10:57:49 modbus: send 00 1c 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2025/05/21 10:57:49 modbus: recv 00 1c 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2025/05/21 10:57:49 charge power: 0W
[etrel ] TRACE 2025/05/21 10:57:49 modbus: send 00 1d 00 00 00 06 01 04 00 0e 00 06
[etrel ] TRACE 2025/05/21 10:57:49 modbus: recv 00 1d 00 00 00 0f 01 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[lp-1 ] DEBUG 2025/05/21 10:57:49 charge currents: [0 0 0]A
[grid1 ] TRACE 2025/05/21 10:57:49 GET http://192.168.178.124:8080/api/v1/status
[pv2 ] TRACE 2025/05/21 10:57:49 GET http://192.168.178.124:8080/api/v1/status
[battery3] TRACE 2025/05/21 10:57:49 GET http://192.168.178.124:8080/api/v1/status
[grid1 ] TRACE 2025/05/21 10:57:49 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":328,"Consumption_W":315,"Fac":49.9739990234375,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":10348,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":3,"Production_W":10662,"RSOC":83,"RemainingCapacity_Wh":69593,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 10:57:48","USOC":82,"Uac":233,"Ubat":213,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 10:57:49 grid power: -10348W
[pv2 ] TRACE 2025/05/21 10:57:49 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":328,"Consumption_W":315,"Fac":49.9739990234375,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":10348,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":3,"Production_W":10662,"RSOC":83,"RemainingCapacity_Wh":69593,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 10:57:48","USOC":82,"Uac":233,"Ubat":213,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 10:57:49 pv 1 power: 10662W
[battery3] TRACE 2025/05/21 10:57:49 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":328,"Consumption_W":315,"Fac":49.9739990234375,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":10348,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":3,"Production_W":10662,"RSOC":83,"RemainingCapacity_Wh":69593,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 10:57:48","USOC":82,"Uac":233,"Ubat":213,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 10:57:49 battery 1 power: 3W
[battery3] TRACE 2025/05/21 10:57:49 GET http://192.168.178.124:8080/api/v1/status
[battery3] TRACE 2025/05/21 10:57:49 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":331,"Consumption_W":542,"Fac":49.97600173950195,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":10097,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":2,"Production_W":10640,"RSOC":83,"RemainingCapacity_Wh":69593,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 10:57:49","USOC":82,"Uac":233,"Ubat":213,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 10:57:49 battery 1 soc: 82%
[site ] DEBUG 2025/05/21 10:57:49 site power: -10245W
[etrel ] TRACE 2025/05/21 10:57:49 modbus: send 00 1e 00 00 00 06 01 04 00 00 00 01
[etrel ] TRACE 2025/05/21 10:57:49 modbus: recv 00 1e 00 00 00 05 01 04 02 00 03
[lp-1 ] DEBUG 2025/05/21 10:57:49 charger status: B
[lp-1 ] INFO 2025/05/21 10:57:49 car connected
[lp-1 ] DEBUG 2025/05/21 10:57:49 vehicle api refresh
[lp-1 ] DEBUG 2025/05/21 10:57:49 pv timer elapse
[lp-1 ] DEBUG 2025/05/21 10:57:49 pv timer inactive
[bmw ] TRACE 2025/05/21 10:57:49 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1747817869851
[bmw ] TRACE 2025/05/21 10:57:50 {"state":{"isLeftSteering":true,"lastFetched":"2025-05-21T08:57:49.944Z","lastUpdatedAt":"2025-05-21T08:55:16Z","isLscSupported":true,"range":194,"doorsState":{"combinedSecurityState":"UNLOCKED","leftFront":"CLOSED","leftRear":"CLOSED","rightFront":"CLOSED","rightRear":"CLOSED","combinedState":"CLOSED","hood":"CLOSED","trunk":"CLOSED"},"windowsState":{"leftFront":"CLOSED","leftRear":"CLOSED","rightFront":"CLOSED","rightRear":"CLOSED","combinedState":"CLOSED"},"roofState":{"roofState":"CLOSED","roofStateType":"SUN_ROOF"},"tireState":{"frontLeft":{"details":{"dimension":"225/55 R17 101Y XL","treadDesign":"Eco Contact 6","manufacturer":"Continental","manufacturingWeek":3323,"isOptimizedForOemBmw":true,"partNumber":"5A40407","speedClassification":{"speedRating":300,"atLeast":false},"mountingDate":"2025-04-26T00:00:00.000Z","season":2,"identificationInProgress":false},"status":{"currentPressure":250,"targetPressure":250}},"frontRight":{"details":{"dimension":"225/55 R17 101Y XL","treadDesign":"Eco Contact 6","manufacturer":"Continental","manufacturingWeek":3323,"isOptimizedForOemBmw":true,"partNumber":"5A4 [... cut ...]
[etrel ] TRACE 2025/05/21 11:02:19 modbus: recv 00 52 00 00 00 07 01 04 04 40 c0 00 00
[lp-1 ] DEBUG 2025/05/21 11:02:19 pv charge current: 14.3A = 0A + 14.3A (-9882W @ 3p)
[etrel ] TRACE 2025/05/21 11:02:19 modbus: send 00 53 00 00 00 0b 01 10 00 08 00 02 04 41 65 25 d8
[etrel ] TRACE 2025/05/21 11:02:19 modbus: recv 00 53 00 00 00 06 01 10 00 08 00 02
[lp-1 ] DEBUG 2025/05/21 11:02:19 set charge current limit: 14.3A
[site ] DEBUG 2025/05/21 11:02:19 solar forecast: accumulated 0.423kWh, produced 1.218kWh, scale 2.877

rmuelle3, 2025-05-21:
Zur Überprüfung: EVCC Release 0.203.1 zeigt das gleiche Verhalten

Log (ohne modbus-Trace):
[site ] DEBUG 2025/05/21 11:11:47 ----
[lp-1 ] DEBUG 2025/05/21 11:11:47 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 11:11:47 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 11:11:47 pv 1 power: 9617W
[site ] DEBUG 2025/05/21 11:11:47 battery 1 power: -9835W
[site ] DEBUG 2025/05/21 11:11:47 grid power: 661W
[site ] DEBUG 2025/05/21 11:11:47 battery 1 soc: 86%
[site ] DEBUG 2025/05/21 11:11:47 site power: -9074W
[lp-1 ] DEBUG 2025/05/21 11:11:47 charger status: B
[lp-1 ] INFO 2025/05/21 11:11:47 car connected
[lp-1 ] DEBUG 2025/05/21 11:11:47 vehicle api refresh
[lp-1 ] DEBUG 2025/05/21 11:11:47 pv timer elapse
[lp-1 ] DEBUG 2025/05/21 11:11:47 pv timer inactive
[main ] DEBUG 2025/05/21 11:11:49 vehicle status: B (BMW i4)
[lp-1 ] INFO 2025/05/21 11:11:49 vehicle updated: unknown -> BMW i4
[lp-1 ] DEBUG 2025/05/21 11:11:49 vehicle status: B
[lp-1 ] DEBUG 2025/05/21 11:11:49 vehicle soc: 65%
[lp-1 ] DEBUG 2025/05/21 11:11:49 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 11:11:49 vehicle range: 195km
[lp-1 ] DEBUG 2025/05/21 11:11:49 pv charge current: 13.2A = 0A + 13.2A (-9074W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:11:49 site power -9074W <= 0W enable threshold
[lp-1 ] DEBUG 2025/05/21 11:11:49 pv enable in 0s
[lp-1 ] DEBUG 2025/05/21 11:11:49 pv enable timer elapsed
[lp-1 ] DEBUG 2025/05/21 11:11:49 set charge current limit: 6A
[lp-1 ] DEBUG 2025/05/21 11:11:49 charger enable
[lp-1 ] DEBUG 2025/05/21 11:11:49 wake-up timer: start
[lp-1 ] DEBUG 2025/05/21 11:11:51 set session soc limit: 80
[site ] DEBUG 2025/05/21 11:11:51 ----
[lp-1 ] DEBUG 2025/05/21 11:11:51 charge power: 256W
[lp-1 ] DEBUG 2025/05/21 11:11:51 charge currents: [0.672 0.656 0.736]A
[site ] DEBUG 2025/05/21 11:11:51 battery 1 power: -9833W
[site ] DEBUG 2025/05/21 11:11:51 pv 1 power: 9588W
[site ] DEBUG 2025/05/21 11:11:51 grid power: 1231W
[site ] DEBUG 2025/05/21 11:11:51 battery 1 soc: 86%
[site ] DEBUG 2025/05/21 11:11:51 site power: -8502W
[lp-1 ] DEBUG 2025/05/21 11:11:51 vehicle odometer: 10901km
[lp-1 ] DEBUG 2025/05/21 11:11:51 charger status: C
[lp-1 ] INFO 2025/05/21 11:11:51 start charging ->
[lp-1 ] DEBUG 2025/05/21 11:11:51 wake-up timer: stop
[lp-1 ] DEBUG 2025/05/21 11:11:51 soc estimated: 65.00% (vehicle: 65.00%)
[lp-1 ] DEBUG 2025/05/21 11:11:51 vehicle soc: 65%
[lp-1 ] DEBUG 2025/05/21 11:11:51 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 11:11:51 vehicle range: 195km
[lp-1 ] DEBUG 2025/05/21 11:11:51 pv charge current: 15.1A = 2.74A + 12.3A (-8502W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:11:51 pv timer reset
[lp-1 ] DEBUG 2025/05/21 11:11:51 pv timer inactive
[lp-1 ] DEBUG 2025/05/21 11:11:51 set charge current limit: 15.1A
[site ] DEBUG 2025/05/21 11:12:17 ----
[lp-1 ] DEBUG 2025/05/21 11:12:17 charge power: 9920W
[lp-1 ] DEBUG 2025/05/21 11:12:17 charge currents: [14.9 14.9 14.8]A
[site ] DEBUG 2025/05/21 11:12:17 battery 1 power: 1046W
[site ] DEBUG 2025/05/21 11:12:17 pv 1 power: 9512W
[site ] DEBUG 2025/05/21 11:12:18 grid power: -111W
[site ] DEBUG 2025/05/21 11:12:18 battery 1 soc: 86%
[site ] DEBUG 2025/05/21 11:12:18 site power: 1035W
[lp-1 ] DEBUG 2025/05/21 11:12:18 detected active phases: 3p
[lp-1 ] DEBUG 2025/05/21 11:12:18 charger status: C
[lp-1 ] DEBUG 2025/05/21 11:12:18 soc estimated: 65.08% (vehicle: 65.00%)
[lp-1 ] DEBUG 2025/05/21 11:12:18 vehicle soc: 65%
[lp-1 ] DEBUG 2025/05/21 11:12:18 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 11:12:18 vehicle range: 195km
[lp-1 ] DEBUG 2025/05/21 11:12:18 pv charge current: 13.6A = 15.1A + -1.5A (1035W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:12:18 set charge current limit: 13.6A
[site ] DEBUG 2025/05/21 11:12:47 ----
[lp-1 ] DEBUG 2025/05/21 11:12:47 charge power: 7872W
[lp-1 ] DEBUG 2025/05/21 11:12:47 charge currents: [11.9 11.9 11.9]A
[site ] DEBUG 2025/05/21 11:12:47 battery 1 power: -912W
[... 238 lines cut ...]
[lp-1 ] DEBUG 2025/05/21 11:21:48 charger status: B
[lp-1 ] DEBUG 2025/05/21 11:21:48 pv charge current: 13.7A = 0A + 13.7A (-9457W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:21:48 set charge current limit: 13.7A
[site ] DEBUG 2025/05/21 11:22:17 ----
[lp-1 ] DEBUG 2025/05/21 11:22:17 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 11:22:17 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 11:22:17 grid power: -9618W
[site ] DEBUG 2025/05/21 11:22:17 pv 1 power: 9809W
[site ] DEBUG 2025/05/21 11:22:17 battery 1 power: 2W
[site ] DEBUG 2025/05/21 11:22:18 battery 1 soc: 88%
[site ] DEBUG 2025/05/21 11:22:18 site power: -9516W
[lp-1 ] DEBUG 2025/05/21 11:22:18 charger status: B
[lp-1 ] DEBUG 2025/05/21 11:22:18 pv charge current: 13.8A = 0A + 13.8A (-9516W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:22:18 set charge current limit: 13.8A
[site ] DEBUG 2025/05/21 11:22:47 ----
[lp-1 ] DEBUG 2025/05/21 11:22:47 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 11:22:47 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 11:22:47 grid power: -9544W
[site ] DEBUG 2025/05/21 11:22:47 battery 1 power: 0W
[site ] DEBUG 2025/05/21 11:22:47 pv 1 power: 9741W
[site ] DEBUG 2025/05/21 11:22:47 battery 1 soc: 88%
[site ] DEBUG 2025/05/21 11:22:47 site power: -9444W
[lp-1 ] DEBUG 2025/05/21 11:22:47 charger status: B
[lp-1 ] DEBUG 2025/05/21 11:22:47 pv charge current: 13.7A = 0A + 13.7A (-9444W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:22:47 set charge current limit: 13.7A

rmuelle3, 2025-05-21:
Jetzt das Ganze (Release 0.203.1) mit modbus-Trace und kurzzeitiger Stromunterbrechung der Wallbox. Das Laden wird im Anschluss fortgesetzt.

[site ] DEBUG 2025/05/21 11:33:17 ----
[etrel ] TRACE 2025/05/21 11:33:17 modbus: send 01 01 00 00 00 06 01 04 00 1a 00 02
[etrel ] TRACE 2025/05/21 11:33:17 modbus: recv 01 01 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2025/05/21 11:33:17 charge power: 0W
[etrel ] TRACE 2025/05/21 11:33:17 modbus: send 01 02 00 00 00 06 01 04 00 0e 00 06
[etrel ] TRACE 2025/05/21 11:33:17 modbus: recv 01 02 00 00 00 0f 01 04 0c 00 00 00 00 00 00 00 00 00 00 00 00
[lp-1 ] DEBUG 2025/05/21 11:33:17 charge currents: [0 0 0]A
[pv2 ] TRACE 2025/05/21 11:33:17 GET http://192.168.178.124:8080/api/v1/status
[battery3] TRACE 2025/05/21 11:33:17 GET http://192.168.178.124:8080/api/v1/status
[grid1 ] TRACE 2025/05/21 11:33:17 GET http://192.168.178.124:8080/api/v1/status
[pv2 ] TRACE 2025/05/21 11:33:17 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":240,"Consumption_W":258,"Fac":49.952999114990234,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":5643,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":0,"Production_W":5900,"RSOC":89,"RemainingCapacity_Wh":74898,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 11:33:17","USOC":88,"Uac":234,"Ubat":214,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 11:33:17 pv 1 power: 5900W
[battery3] TRACE 2025/05/21 11:33:17 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":240,"Consumption_W":258,"Fac":49.952999114990234,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":5643,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":0,"Production_W":5900,"RSOC":89,"RemainingCapacity_Wh":74898,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 11:33:17","USOC":88,"Uac":234,"Ubat":214,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 11:33:17 battery 1 power: 0W
[battery3] TRACE 2025/05/21 11:33:17 GET http://192.168.178.124:8080/api/v1/status
[grid1 ] TRACE 2025/05/21 11:33:17 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":240,"Consumption_W":258,"Fac":49.952999114990234,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":5643,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":0,"Production_W":5900,"RSOC":89,"RemainingCapacity_Wh":74898,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 11:33:17","USOC":88,"Uac":234,"Ubat":214,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 11:33:17 grid power: -5643W
[battery3] TRACE 2025/05/21 11:33:17 {"BackupBuffer":"10","BatteryCharging":false,"BatteryDischarging":false,"Consumption_Avg":240,"Consumption_W":258,"Fac":49.952999114990234,"FlowConsumptionBattery":false,"FlowConsumptionGrid":false,"FlowConsumptionProduction":true,"FlowGridBattery":false,"FlowProductionBattery":false,"FlowProductionGrid":true,"GridFeedIn_W":5643,"IsSystemInstalled":1,"OperatingMode":"1","Pac_total_W":0,"Production_W":5900,"RSOC":89,"RemainingCapacity_Wh":74898,"SystemStatus":"OnGrid","Timestamp":"2025-05-21 11:33:17","USOC":88,"Uac":234,"Ubat":214,"dischargeNotAllowed":false,"generator_autostart":false,"NVM_REINIT_STATUS":0}
[site ] DEBUG 2025/05/21 11:33:17 battery 1 soc: 88%
[site ] DEBUG 2025/05/21 11:33:17 site power: -5543W
[etrel ] TRACE 2025/05/21 11:33:17 modbus: send 01 03 00 00 00 06 01 04 00 00 00 01
[etrel ] TRACE 2025/05/21 11:33:17 modbus: recv 01 03 00 00 00 05 01 04 02 00 06
[lp-1 ] DEBUG 2025/05/21 11:33:17 charger status: B
[lp-1 ] INFO 2025/05/21 11:33:17 car connected
[lp-1 ] DEBUG 2025/05/21 11:33:17 vehicle api refresh
[lp-1 ] DEBUG 2025/05/21 11:33:17 pv timer elapse
[lp-1 ] DEBUG 2025/05/21 11:33:17 pv timer inactive
[bmw ] TRACE 2025/05/21 11:33:17 GET https://cocoapi.bmwgroup.com/eadrax-vcs/v4/vehicles/state?apptimezone=120&appDateTime=1747819997996
[bmw ] TRACE 2025/05/21 11:33:18 { "statusCode": 403, "message": "Out of call volume quota. Quota will be replenished in 00:26:42." }
[main ] ERROR 2025/05/21 11:33:18 vehicle status: 403: Out of call volume quota. Quota will be replenished in 00:26:42. (unexpected status: 403 (Forbidden))
[etrel ] TRACE 2025/05/21 11:33:18 modbus: send 01 04 00 00 00 06 01 04 00 04 00 02
[etrel ] TRACE 2025/05/21 11:33:18 modbus: recv 01 04 00 00 00 07 01 04 04 00 00 00 00
[lp-1 ] DEBUG 2025/05/21 11:33:18 pv charge current: 8.03A = 0A + 8.03A (-5543W @ 3p)
[lp-1 ] DEBUG 2025/05/21 11:33:18 site power -5543W <= 0W enable threshold
[lp-1 ] DEBUG 2025/05/21 11:33:18 pv enable in 0s
[lp-1 ] DEBUG 2025/05/21 11:33:18 pv enable timer elapsed
[etrel ] TRACE 2025/05/21 11:33:18 modbus: send 01 05 00 00 00 0b 01 10 00 08 00 02 04 40 c0 00 00
[etrel ] TRACE 2025/05/21 11:33:18 modbus: recv 01 05 00 00 00 06 01 10 00 08 00 02
[lp-1 ] DEBUG 2025/05/21 11:33:18 set charge current limit: 6A
[etrel ] TRACE 2025/05/21 11:33:18 modbus: send 01 06 00 00 00 0b 01 10 00 08 00 02 04 40 c0 00 00
[... cut ...]
[lp-1 ] DEBUG 2025/05/21 11:42:48 pv charge current: 9.17A = 9.25A + -0.0783A (54W @ 3p)
[etrel ] TRACE 2025/05/21 11:42:48 modbus: send 01 8c 00 00 00 0b 01 10 00 08 00 02 04 41 12 ca 40
[etrel ] TRACE 2025/05/21 11:42:48 modbus: recv 01 8c 00 00 00 06 01 10 00 08 00 02
[lp-1 ] DEBUG 2025/05/21 11:42:48 set charge current limit: 9.17A
[httpd ] TRACE 2025/05/21 11:42:51 GET /api/state 200
[httpd ] TRACE 2025/05/21 11:42:56 GET /api/state 200
[httpd ] TRACE 2025/05/21 11:43:01 GET /api/state 200

rmuelle3, 2025-05-22:
Ich habe mit der Sonnen App etwas herumgespielt und einen Workaround gefunden, der das Problem anscheinend löst.
Sonnen hat im Dezember die alte SonnenCharger-App abgeschaltet und die SonnenCharger-Funktionalität in die Sonnen-App integriert. Ich habe mir im Dezember die Sonnen-App im Power-Mode eingerichtet und den Batterie-Betriebsmodus auf "2" (Eigenverbrauch) eingestellt um weiterhin EVCC nutzen zu können. Danach habe ich mich um die Charger-Funktionalität in der Sonnen-App nicht mehr gekümmert, da ich nur über EVCC mein Auto geladen habe.

Seitdem hat Sonnen den Power-Mode funktional verändert und in der App einen Schieberegler für den maximalen Ladestrom eingeführt. Dieser Schieberegler war in meiner App initial, womit vermutlich der max. Ladestrom nicht definiert war. Das könnte der Grund sein, warum seit 1-2 Monaten immer wieder die Warnmeldung "charger logic error: disabled but charging" ausgegeben wurden. Vermutlich gab es vor ca. 2 Wochen noch einmal ein Update von Sonnen, was zu dem Lade-Abbruch nach 90 bzw. 120 sec. geführt hat. 

Mit den folgenden Schritten konnte ich den SonnenCharger wieder in einen stabilen Zustand versetzen und EVCC ohne Warnmeldung, Ladeabbruch oder Charger-Neustart mit Aus-/Einschalten am Sicherungskasten erfolgreich nutzen:

Schritt 1: EVCC abschalten

Schritt 2: Charger-Konfiguration in der App auswählen

[image]

Schritt 3: Batterie-Betriebsmodus 11 setzen

[image]
Der Modus wird gesetzt durch Auswahl eines Fahrzeuges. Der Vorgang muss solange wiederholt werden, bis die App die Fahrzeugauswahl bestätigt hat und das rote Feld blau wird. Danach ist der Batterie-Betriebsmodus 11 gesetzt.
Wenn das rote Feld blau geworden ist kann auf den Button Power Mode geklickt werden.

Schritt 4: Ladestrom im Power-Mode setzen

[image]
Durch Verschieben des Schiebereglers kann der Ladestrom gesetzt werden.

Schritt 5: Auto anschliessen und im Power-Mode laden; dabei Ladestrom mit Schieberegler einstellen bis der Charger korrekt funktioniert
Bei mir hat dies erst im zweiten Anlauf geklappt. Ich musste das Ladekabel einmal im Auto aus- und wieder einstecken bevor der Ladevorgang korrekt durchgeführt wurde. Zum Abschluss den Schieberegler auf max. Ladestrom (16A / 11,04 kW) einstellen.

Schritt 6: Ladevorgang durch Ausstecken des Ladekabel beenden

Schritt 7: Batteriemodus auf "2" (Eigenverbrauch) einstellen

[image]
Die Einstellung geschieht in der Batteriekonfiguration durch Aufruf der IP-Adresse im Browser. Der Betriebsmodus "11" wird nicht angezeigt (leeres Feld). Der Betriebsmodus "2" (Eigenverbrauch) wird Klicken auf den Auswahlpfeil ausgewählt. Das Feld in der Sonnen-App aus Schritt 3 wird wieder rot.
Wenn der Batteriemodus "11" eingestellt ist (blaues Feld) konkurrieren  EVCC und die Sonnenbatterie bzw. -Cloud um die Steuerung des Chargers.

Schritt 8: EVCC wieder anschalten
Das Fahrzeug kann jetzt wieder ohne Probleme mit EVCC über den SonnenCharger geladen werden.

jomuench, 2025-05-23:
@rmuelle3 Respekt, und danke für das Teilen!
Ich hänge bei Schritt 3. Ist der Batteriemodus 11 schon eingestellt, wenn die App Folgendes anzeigt?

[image]

danke!

jomuench, 2025-06-06:
Kurzes Update -
mein Sonnencharger hat jetzt wieder sein Eigenleben angefangen. Oszillierende Ladevorgänge... sprich: Sonnencharger startet Laden, evcc stellt es wieder ab.
In den Einstellungen der Sonnenbatterie ist _Operating Mode_ = _Automatic - Optimization_ eingestellt. Das war vorher definitiv anders, nämlich wie in der [Lösung](https://github.com/evcc-io/evcc/issues/21312#issuecomment-2901651528) erklärt, _Automatic - Self-Consumption_.
Ich weiß leider nicht, was das ausgelöst hat. Eventuell ein Softwareupdate der Sonnenbatterie?

andig, 2025-06-18:
Mal anderes Auto probieren?

2bbionic, 2025-06-18:
> Mal anderes Auto probieren?

Wenn ich eines hätte :)
Bei mir in der Strasse bin ich aktuell der einzige mit 'nem E-Fzg.
Hier noch ein Modbus-Trace; das ist allerdings im idle gerade, da kein Auto angeschlossen. Den EOF-Fehler bekomme ich auch noch nicht immer, kann leider nicht genau sagen, seit welcher evcc-Version; ich denke aber mal seit diesem Jahr
Modbus-Traces werde ich morgen nachreichen, wenn wieder geladen wird

mfuchs1984, 2025-06-19:
@2bbionic soweit ich sehen kann ist das eine Bender Controller basierte wallbox. Dort gibt's im Webinterface der Wallbox ziemlich detaillierte logs.