evcc-io/evcc #21326: Sungrow HT battery > evcc charging priority not considered

mikeharris2017, 2025-05-20:
### Describe the bug

I have at Sungrow SH25T and a Sungrow SBR battery. Instead of a car I have a MyPV 12kW water heater installed, which should be used before the battery starts charging. 

[image]
Once the sun comes out the heater and the battery start charging at the same time: 
[image]

Based on the settings I assumed that the battery charging power would only be the remaining power. 

PS: Installation is done with https://raw.githubusercontent.com/community-scripts/ProxmoxVE/main/ct/evcc.sh in a proxmox container. Therefore I assume it is linux. If this is not correct, I will do a setup in a manually setup linux vm. 

### Steps to reproduce

I'm not sure which info is needed exactly
1. Setup evcc.yaml and battery config as shown above
2. waiting for the sun coming out in the morning

[evcc-20250520-083724-trace.log](https://github.com/user-attachments/files/20326953/evcc-20250520-083724-trace.log)

[evcc-20250520-083718-debug.log](https://github.com/user-attachments/files/20326954/evcc-20250520-083718-debug.log)
3. Nightly installed at 8:58 today --> No change
[image]

[evcc-20250520-092026-trace.log](https://github.com/user-attachments/files/20327483/evcc-20250520-092026-trace.log)

[evcc-20250520-092014-debug.log](https://github.com/user-attachments/files/20327486/evcc-20250520-092014-debug.log)

### Configuration details

```yaml
meters:
- type: template
  template: siemens-pac2200
  id: 1  
  host: 192.168.43.121  
  port: 5021  
  usage: grid  
  modbus: tcpip  
  name: grid1
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.43.123  
  port: 5023  
  usage: pv  
  modbus: tcpip  
  name: pv
- type: template
  template: sungrow-hybrid 
  id: 1  
  host: 192.168.43.123  
  port: 5023  
  usage: battery  
  modbus: tcpip  
  name: battery
  capacity: 12 

site:
  title: EVCC Test Setup
  meters:
    grid: grid1
    pv:
    - pv
    battery:
    - battery

loadpoints:
  - title: AC Thor 9s Heizstab
    charger: AC_Thor_9s_heizstab
#    meter: AC_Thor_9s_Meter # used for temperature
    # vehicle: heating1 #
    mode: pv  # standard mode of loadpoint
    priority: 1   
    enable:
      threshold: -300 # einschalten, wenn 30 sec. lang mind. 300W Überschuss vorhanden ist
      delay: 60s
    disable:
      threshold: -150 # Ausschalten, wenn 10 sec. lang der Überschuss von 50W unterschritten wird.
      delay: 120s

chargers:
  - name: AC_Thor_9s_heizstab
    type: template
    template: ac-thor
    modbus: tcpip
    id: 1
    host: 192.168.43.124
    port: 5024
    tempsource: 1
```

### Log details

```text
pasted above as they are long files
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v.203.6

VolkerK62, 2025-05-20:
The priority setting doesn't prevent battery charging. 
It simply ensures that all excess power is made available to the heater.

mikeharris2017, 2025-05-20:
I realized that behaviour, as it has no effect. 

I need a setting that allows me to put an defined amount power to the heater first and charge the battery once the temperature of 62 degree is reached. Same as a car would be charged first go 62% and then the battery. 
Currently we charge the battery and the furnace burns 25kwh gas in the atmosphere as the water remains too cold. Around noon we feed the grid with 30kwh energy. With evcc we already save some gas.

VolkerK62, 2025-05-20:
>I realized that behaviour, as it has no effect.

the question is, why does your heater do not use the offered power?
```
[lp-1  ] DEBUG 2025/05/20 09:08:14 charge power: 2950W
[lp-1  ] DEBUG 2025/05/20 09:08:14 pv charge current: 46.1A = 12.8A + 33.3A (-7649W @ 1p)
[lp-1  ] DEBUG 2025/05/20 09:08:14 set charge current limit: 46.1A
[site  ] DEBUG 2025/05/20 09:08:44 ----
[lp-1  ] DEBUG 2025/05/20 09:08:44 charge power: 2651W
```

and I noticed, that your batterypower is oscillating
```
[site  ] DEBUG 2025/05/20 09:08:14 battery 1 power: -7629W
[site  ] DEBUG 2025/05/20 09:08:44 battery 1 power: -1851W
[site  ] DEBUG 2025/05/20 09:09:14 battery 1 power: -7631W
```