evcc-io/evcc #21327: evcc does not start after evcc 200.5 caused by cannot create RCT meter (actual evcc version 203.5)

frog60, 2025-05-20:
### Describe the bug

203.6:

Logfile after restart:

[main  ] FATAL 2025/05/20 09:55:11 meter [my_rct_pv] cannot create meter 'my_rct_pv': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/20 09:55:11 will attempt restart in: 15m0s

### Steps to reproduce

1. Starting with evcc 200.5
2. Install latest version evcc 203.6
3. evcc log file shows: 
[main  ] FATAL 2025/05/20 09:55:11 meter [my_rct_pv] cannot create meter 'my_rct_pv': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/20 09:55:11 will attempt restart in: 15m0s
4. system does not successfully restart
...

### Configuration details

```yaml
evcc.yaml:
...
meters:
  - name: my_rct_grid
    type: template
    template: rct-power
    usage: grid
    host: 192.168.178.58
    minsoc: 7 # Lower limit when discharging the battery (normal operation) (optional)
    maxsoc: 97 # Upper limit when charging the battery from the grid (optional)
  - name: my_rct_pv
    type: template
    template: rct-power
    usage: pv
    host: 192.168.178.58
    minsoc: 7 # Lower limit when discharging the battery (normal operation) (optional)
    maxsoc: 97 # Upper limit when charging the battery from the grid (optional)
  - name: my_rct_battery
    type: template
    template: rct-power
    usage: battery
    host: 192.168.178.58
    capacity: 10.37
    minsoc: 7 # Lower limit when discharging the battery (normal operation) (optional)
    maxsoc: 97 # Upper limit when charging the battery from the grid (optional)
    # maxchargepower: # Maximum charge power for forced charging of the battery in watts. (optional)
...
```

### Log details

```text
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/20 09:53:41 evcc 0.203.6
[main  ] INFO 2025/05/20 09:53:41 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/20 09:53:41 connecting evcc-473776331 at tcp://localhost:1883
[main  ] INFO 2025/05/20 09:53:41 listening at :7070
[main  ] FATAL 2025/05/20 09:55:11 meter [my_rct_pv] cannot create meter 'my_rct_pv': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/20 09:55:11 will attempt restart in: 15m0s
[main  ] ERROR 2025/05/20 09:55:12 version check failed: GET https://api.github.com/repos/evcc-io/evcc/releases/latest: 403 API rate limit exceeded for 37.201.241.215. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.) [rate reset in 44m49s] (installed: 0.203.6)
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/20 09:56:49 evcc 0.203.6
[main  ] INFO 2025/05/20 09:56:49 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/20 09:56:49 connecting evcc-1922369903 at tcp://localhost:1883
[main  ] INFO 2025/05/20 09:56:49 listening at :7070
[main  ] FATAL 2025/05/20 09:58:19 meter [my_rct_battery] cannot create meter 'my_rct_battery': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/20 09:58:19 will attempt restart in: 15m0s
[main  ] ERROR 2025/05/20 09:58:19 version check failed: GET https://api.github.com/repos/evcc-io/evcc/releases/latest: 403 API rate limit exceeded for 37.201.241.215. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.) [rate reset in 41m41s] (installed: 0.203.6)
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/20 09:59:55 evcc 0.203.6
[main  ] INFO 2025/05/20 09:59:55 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/20 09:59:55 connecting evcc-848459797 at tcp://localhost:1883
[main  ] INFO 2025/05/20 09:59:55 listening at :7070
[main  ] FATAL 2025/05/20 10:01:25 meter [my_rct_grid] cannot create meter 'my_rct_grid': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/20 10:01:25 will attempt restart in: 15m0s
[main  ] ERROR 2025/05/20 10:01:25 version check failed: GET https://api.github.com/repos/evcc-io/evcc/releases/latest: 403 API rate limit exceeded for 37.201.241.215. (But here's the good news: Authenticated requests get a higher rate limit. Check out the documentation for more details.) [rate reset in 38m36s] (installed: 0.203.6)
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/20 10:03:01 evcc 0.203.6
[main  ] INFO 2025/05/20 10:03:01 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/20 10:03:01 connecting evcc-1246584242 at tcp://localhost:1883
[main  ] INFO 2025/05/20 10:03:01 listening at :7070
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

203.6

frog60, 2025-05-20:
Maybe the timeout behavior changed after evcc 0.200.5 - since evcc 0.200.5 I was not able to restart evcc successfully - it loops always, i.e. it tries to restart again and again. Thx in advance.

In the following Logfile you can see several restarts until evcc runs stable after 12:51:50.

Logfile evcc 0200.5:

**Using config file: /config/evcc.yaml**
**starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'**
[main  ] INFO 2025/05/20 12:39:23 evcc 0.200.5
[main  ] INFO 2025/05/20 12:39:23 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/20 12:39:23 connecting evcc-447127107 at tcp://localhost:1883
[main  ] INFO 2025/05/20 12:39:23 listening at :7070
[lp-1  ] DEBUG 2025/05/20 12:39:25 charge total import: 5499.931kWh
[site  ] INFO 2025/05/20 12:39:26 site config:
[site  ] INFO 2025/05/20 12:39:26   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/05/20 12:39:26     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/20 12:39:26     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/20 12:39:26     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/20 12:39:26     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/20 12:39:26     pv 4:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/20 12:39:26     pv 5:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/20 12:39:26     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/05/20 12:39:26   vehicles:
[site  ] INFO 2025/05/20 12:39:26     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/20 12:39:26     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/20 12:39:26     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/20 12:39:26     vehicle 4: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/05/20 12:39:26   tariffs:
[site  ] INFO 2025/05/20 12:39:26     grid:      priceforecast
[site  ] INFO 2025/05/20 12:39:26     feed-in:   pricestatic
[site  ] INFO 2025/05/20 12:39:26     co2:       ✗
[site  ] INFO 2025/05/20 12:39:26     solar:     solar
[lp-1  ] INFO 2025/05/20 12:39:26 loadpoint 1:
[lp-1  ] INFO 2025/05/20 12:39:26   mode:        pv
[lp-1  ] INFO 2025/05/20 12:39:26   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/05/20 12:39:26   meters:      charge ✓
[lp-1  ] INFO 2025/05/20 12:39:26     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2025/05/20 12:39:26 set priority soc: 80
[lp-1  ] DEBUG 2025/05/20 12:39:26 phase timer inactive
[lp-1  ] DEBUG 2025/05/20 12:39:26 pv timer inactive
[lp-1  ] INFO 2025/05/20 12:39:26 vehicle updated: unknown -> FIAT 500e
[site  ] DEBUG 2025/05/20 12:39:27 ----
[lp-1  ] DEBUG 2025/05/20 12:39:27 charge power: 0W
[lp-1  ] DEBUG 2025/05/20 12:39:27 charge currents: [0 0 0]A
[main  ] INFO 2025/05/20 12:39:27 new version available: 0.203.6
[site  ] DEBUG 2025/05/20 12:39:27 grid power: -5735W
[site  ] DEBUG 2025/05/20 12:39:27 battery 1 power: 4W
[site  ] DEBUG 2025/05/20 12:39:27 battery 1 soc: 97%
[site  ] DEBUG 2025/05/20 12:39:27 pv 1 power: 3348W
[site  ] DEBUG 2025/05/20 12:39:30 pv 3 power: 281W
[site  ] DEBUG 2025/05/20 12:39:31 pv 2 power: 1644W
**Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'**
[main  ] INFO 2025/05/20 12:42:29 evcc 0.200.5
[main  ] INFO 2025/05/20 12:42:29 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/20 12:42:29 connecting evcc-648617054 at tcp://localhost:1883
[main  ] INFO 2025/05/20 12:42:29 listening at :7070
[lp-1  ] DEBUG 2025/05/20 12:42:31 charge total import: 5499.931kWh
[site  ] INFO 2025/05/20 12:42:32 site config:
[site  ] INFO 2025/05/20 12:42:32   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/05/20 12:42:32     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/20 12:42:32     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/20 12:42:32     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/20 12:42:32     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/20 12:42:32     pv 4:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/20 12:42:32     pv 5:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/20 12:42:32     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/05/20 12:42:32   vehicles:
[site  ] INFO 2025/05/20 12:42:32     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/20 12:42:32     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[... 722 lines cut ...]
[site  ] DEBUG 2025/05/20 13:12:20 pv 1 power: 3349W
[site  ] DEBUG 2025/05/20 13:12:21 pv 2 power: 1646W
[site  ] DEBUG 2025/05/20 13:12:22 pv 3 power: 286W
[site  ] DEBUG 2025/05/20 13:12:24 pv power: 6140W
[site  ] DEBUG 2025/05/20 13:12:24 site power: -5323W
[lp-1  ] DEBUG 2025/05/20 13:12:24 charge total import: 5499.931kWh
[lp-1  ] DEBUG 2025/05/20 13:12:24 charger status: A
[site  ] DEBUG 2025/05/20 13:12:50 ----
[lp-1  ] DEBUG 2025/05/20 13:12:50 charge power: 0W
[lp-1  ] DEBUG 2025/05/20 13:12:50 charge currents: [0 0 0]A
[site  ] DEBUG 2025/05/20 13:12:50 pv 5 power: 570W
[site  ] DEBUG 2025/05/20 13:12:50 pv 4 power: 289W
[site  ] DEBUG 2025/05/20 13:12:50 grid power: -5717W
[site  ] DEBUG 2025/05/20 13:12:50 battery 1 power: 4W
[site  ] DEBUG 2025/05/20 13:12:50 battery 1 soc: 97%
[site  ] DEBUG 2025/05/20 13:12:51 pv 1 power: 3355W
[site  ] DEBUG 2025/05/20 13:12:51 pv 2 power: 1644W
[site  ] DEBUG 2025/05/20 13:12:52 pv 3 power: 286W
[site  ] DEBUG 2025/05/20 13:12:54 pv power: 6144W
[site  ] DEBUG 2025/05/20 13:12:54 site power: -5314W
[lp-1  ] DEBUG 2025/05/20 13:12:54 charge total import: 5499.931kWh
[lp-1  ] DEBUG 2025/05/20 13:12:54 charger status: A
[site  ] DEBUG 2025/05/20 13:13:20 ----
[lp-1  ] DEBUG 2025/05/20 13:13:20 charge power: 0W
....

andig, 2025-05-20:
> In the following Logfile you can see several restarts until evcc runs stable after 12:51:50.

So .5 does not run stable either? The log does not show why it restarts?

frog60, 2025-05-21:
After setting log/levels for "log" and "meters" the system 203.6 needs again more than one attemps to start again.

log: **debug** # 2025-05-21 changed trace -> debug
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  **meters: debug** # 2025-05-21 changed trace -> debug

...
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/21 13:15:05 evcc 0.203.6
[main  ] INFO 2025/05/21 13:15:05 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/21 13:15:05 connecting evcc-491466158 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/21 13:15:05 tcp://localhost:1883 connected
[main  ] INFO 2025/05/21 13:15:05 listening at :7070
[main  ] FATAL 2025/05/21 13:16:36 meter [my_rct_battery] cannot create meter 'my_rct_battery': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/21 13:16:36 will attempt restart in: 15m0s
...

andig, 2025-05-23:
Try

    cache: 1m

frog60, 2025-05-23:
cache: 1m or 90s and more added for:

- grid
- pv
- battery

In nearly every cases -> No success:

[main  ] INFO 2025/05/23 **16:00:54** using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/23 16:00:54 connecting evcc-2026738152 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/23 16:00:54 tcp://localhost:1883 connected
[main  ] INFO 2025/05/23 16:00:54 listening at :7070
[main  ] FATAL 2025/05/23 **16:02:24** meter [my_rct_battery] cannot create meter 'my_rct_battery': cannot create meter type 'template': cannot create meter type 'rct': timeout
[main  ] FATAL 2025/05/23 16:02:24 will attempt restart in: 15m0s

Sometimes it happens that it starts after 4 seconds:

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/23 **16:03:09** evcc 0.203.6
[main  ] INFO 2025/05/23 16:03:09 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/23 16:03:09 connecting evcc-498363957 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/23 16:03:09 tcp://localhost:1883 connected
[main  ] INFO 2025/05/23 16:03:09 listening at :7070
[lp-1  ] DEBUG 2025/05/23 16:03:12 charge total import: 5499.931kWh
[site  ] INFO 2025/05/23 16:03:13 site config:
[site  ] INFO 2025/05/23 **16:03:13**   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/05/23 16:03:13     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:03:13     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:03:13     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:03:13     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:03:13     pv 4:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:03:13     pv 5:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:03:13     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity

andig, 2025-05-23:
I repeat: we need a log file that shows RCT trace log. Formatted in a readable way.

frog60, 2025-05-23:
How can I produce a RCT trace log -> levels: -> **rct: trace**?

log: debug # 2025-05-21 changed
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error
  rct: **debug** # 2025-05-21 added

**After restarting homeassistent** the evcc addon starts successfully. Could this be a hint?

[main  ] **FATAL 2025/05/23 16:16:14 will attempt restart in: 15m0s**
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/23 **16:17:50** evcc 0.203.6
[main  ] INFO 2025/05/23 16:17:50 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/23 16:17:50 connecting evcc-1527174830 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/23 16:17:50 tcp://localhost:1883 connected
[main  ] INFO 2025/05/23 16:17:50 listening at :7070
[lp-1  ] DEBUG 2025/05/23 16:18:05 charge total import: 5499.931kWh
[site  ] INFO 2025/05/23 16:18:09 site config:
[site  ] INFO 2025/05/23 **16:18:09**   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/05/23 16:18:09     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:18:09     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:18:09     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:18:09     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:18:09     pv 4:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:18:09     pv 5:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:18:09     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓

frog60, 2025-05-23:
Her is the rct trace log:

Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/23 16:20:57 evcc 0.203.6
[main  ] INFO 2025/05/23 16:20:57 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/23 16:20:57 connecting evcc-64630476 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/23 16:20:57 tcp://localhost:1883 connected
[main  ] INFO 2025/05/23 16:20:57 listening at :7070
[rct   ] TRACE 2025/05/23 16:21:40 recv: (05) Response (EBC62737) #INVALID [50 53 20 36 2e 30 20 4e 54 4c 4d 00 00 00 00 00 00 00 00 00 ... (64)]
[rct   ] TRACE 2025/05/23 16:21:40 recv: (05) Response (392D1BEE) #INVALID [00]
[rct   ] TRACE 2025/05/23 16:21:40 recv: (05) Response (DDD1C2D0) #INVALID [32 2e 33 2e 35 36 38 35 00 00 00 00 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:40 recv: (05) Response (FF2A258B) #INVALID [30 2e 30 2e 30 2e 30 00 00 00 00 00 00 00 00 00 00 00 00 00 ... (32)]
[rct   ] TRACE 2025/05/23 16:21:40 recv: (05) Response (D83DC6AC) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:40 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
**[main  ] FATAL 2025/05/23 16:21:40 meter [my_rct_grid] cannot create meter 'my_rct_grid': cannot create meter type 'template': cannot create meter type 'rct': timeout**
[main  ] FATAL 2025/05/23 16:21:40 will attempt restart in: 15m0s
[rct   ] TRACE 2025/05/23 16:21:43 recv: (05) Response (392D1BEE) #INVALID [00]
[rct   ] TRACE 2025/05/23 16:21:43 recv: (05) Response (DDD1C2D0) #INVALID [32 2e 33 2e 35 36 38 35 00 00 00 00 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:43 recv: (05) Response (FF2A258B) #INVALID [30 2e 30 2e 30 2e 30 00 00 00 00 00 00 00 00 00 00 00 00 00 ... (32)]
[rct   ] TRACE 2025/05/23 16:21:43 recv: (05) Response (D83DC6AC) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:43 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[rct   ] TRACE 2025/05/23 16:21:43 recv: (05) Response (7924ABD9) #INVALID [30 30 36 35 41 34 36 33 31 34 38 37 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:46 recv: (05) Response (FF2A258B) #INVALID [30 2e 30 2e 30 2e 30 00 00 00 00 00 00 00 00 00 00 00 00 00 ... (32)]
[rct   ] TRACE 2025/05/23 16:21:46 recv: (05) Response (D83DC6AC) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:46 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[rct   ] TRACE 2025/05/23 16:21:49 recv: (05) Response (7924ABD9) #INVALID [30 30 36 35 41 34 36 33 31 34 38 37 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:49 recv: (05) Response (7924ABD9) #INVALID [30 30 36 35 41 34 36 33 31 34 38 37 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:21:49 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
....

andig, 2025-05-23:
Könntest du das bitte formatieren? Einfach `< >` benutzen. Danke.

frog60, 2025-05-23:
Was meinst du mit <> ? Wie soll ich hier das Log formatieren?

Hier noch ein Trace nach einem erfolgreichen Start von evcc:

starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/23 **16:36:44** evcc 0.203.6
[main  ] INFO 2025/05/23 16:36:44 using config file: /config/evcc.yaml
[mqtt  ] INFO 2025/05/23 16:36:44 connecting evcc-1304134143 at tcp://localhost:1883
[mqtt  ] DEBUG 2025/05/23 16:36:44 tcp://localhost:1883 connected
[main  ] INFO 2025/05/23 16:36:44 listening at :7070
[rct   ] TRACE 2025/05/23 **16:36:46** recv: (05) Response (DDD1C2D0) #INVALID [32 2e 33 2e 35 36 38 35 00 00 00 00 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:46 recv: (05) Response (FF2A258B) #INVALID [30 2e 30 2e 30 2e 30 00 00 00 00 00 00 00 00 00 00 00 00 00 ... (32)]
[rct   ] TRACE 2025/05/23 16:36:46 recv: (05) Response (D83DC6AC) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:46 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[rct   ] TRACE 2025/05/23 16:36:46 recv: (05) Response (7924ABD9) #INVALID [30 30 36 35 41 34 36 33 31 34 38 37 00 00 00 00]
[lp-1  ] DEBUG 2025/05/23 16:36:48 charge total import: 5499.931kWh
[site  ] INFO 2025/05/23 16:36:49 site config:
[site  ] INFO 2025/05/23 16:36:49   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/05/23 16:36:49     grid:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:36:49     pv 1:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:36:49     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:36:49     pv 3:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/05/23 16:36:49     pv 4:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:36:49     pv 5:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/05/23 16:36:49     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/05/23 16:36:49   vehicles:
[site  ] INFO 2025/05/23 16:36:49     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/23 16:36:49     vehicle 2: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/23 16:36:49     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/05/23 16:36:49     vehicle 4: range ✓ finish ✓ status ✓ climate ✓ wakeup ✓
[site  ] INFO 2025/05/23 16:36:49   tariffs:
[site  ] INFO 2025/05/23 16:36:49     grid:      priceforecast
[site  ] INFO 2025/05/23 16:36:49     feed-in:   pricestatic
[site  ] INFO 2025/05/23 16:36:49     co2:       ✗
[site  ] INFO 2025/05/23 16:36:49     solar:     ✓
[lp-1  ] INFO 2025/05/23 16:36:49 loadpoint 1:
[lp-1  ] INFO 2025/05/23 16:36:49   mode:        pv
[lp-1  ] INFO 2025/05/23 16:36:49   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1  ] INFO 2025/05/23 16:36:49   meters:      charge ✓
[lp-1  ] INFO 2025/05/23 16:36:49     charge:    power ✓ energy ✓ currents ✓
[site  ] DEBUG 2025/05/23 16:36:49 set priority soc: 80
[lp-1  ] DEBUG 2025/05/23 16:36:49 phase timer inactive
[lp-1  ] DEBUG 2025/05/23 16:36:49 pv timer inactive
[lp-1  ] INFO 2025/05/23 16:36:49 vehicle updated: unknown -> FIAT 500e
[site  ] DEBUG 2025/05/23 16:36:49 ----
[lp-1  ] DEBUG 2025/05/23 16:36:49 charge power: 0W
[lp-1  ] DEBUG 2025/05/23 16:36:49 charge currents: [0 0 0]A
[rct   ] TRACE 2025/05/23 16:36:49 send: (01) Read (400F015B) Battery power [W] []
[rct   ] TRACE 2025/05/23 16:36:49 send: (01) Read (91617C58) Total grid power [W] []
[rct   ] TRACE 2025/05/23 16:36:49 send: (01) Read (E96F1844) #INVALID []
[rct   ] TRACE 2025/05/23 16:36:49 send: (01) Read (B5317B78) Solar generator A power [W] []
[rct   ] TRACE 2025/05/23 16:36:49 send: (01) Read (AA9AA253) Solar generator B power [W] []
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (EBC62737) #INVALID [50 53 20 36 2e 30 20 4e 54 4c 4d 00 00 00 00 00 00 00 00 00 ... (64)]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (400F015B) Battery power [W] [bf 33 7d 82]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (91617C58) Total grid power [W] [c5 28 7a 5d]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (E96F1844) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (B5317B78) Solar generator A power [W] [44 95 12 d9]
[lp-1  ] DEBUG 2025/05/23 16:36:49 set phases: 0
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (AA9AA253) Solar generator B power [W] [44 a6 40 e8]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (DDD1C2D0) #INVALID [32 2e 33 2e 35 36 38 35 00 00 00 00 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (FF2A258B) #INVALID [30 2e 30 2e 30 2e 30 00 00 00 00 00 00 00 00 00 00 00 00 00 ... (32)]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (D83DC6AC) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:49 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[lp-1  ] DEBUG 2025/05/23 16:36:49 set phases: 1
[site  ] DEBUG 2025/05/23 16:36:50 pv 1 power: 2523W
[site  ] DEBUG 2025/05/23 16:36:50 battery 1 power: -1W
[rct   ] TRACE 2025/05/23 16:36:50 send: (01) Read (959930BF) Battery state of charge []
[site  ] DEBUG 2025/05/23 16:36:50 grid power: -2696W
[rct   ] TRACE 2025/05/23 16:36:50 send: (01) Read (A59C8428) Total energy grid [Wh] []
[rct   ] TRACE 2025/05/23 16:36:50 recv: (05) Response (7924ABD9) #INVALID [30 30 36 35 41 34 36 33 31 34 38 37 00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:50 recv: (05) Response (959930BF) Battery state of charge [3f 77 dc 71]
[... cut ...]
[rct   ] TRACE 2025/05/23 16:36:52 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[site  ] DEBUG 2025/05/23 16:36:54 pv 3 power: 82W
[site  ] DEBUG 2025/05/23 16:36:55 pv 2 power: 516W
[rct   ] TRACE 2025/05/23 16:36:55 recv: (05) Response (D83DC6AC) #INVALID [00 00 00 00]
[rct   ] TRACE 2025/05/23 16:36:58 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[rct   ] TRACE 2025/05/23 16:36:58 recv: (05) Response (9D785E8C) #INVALID [00 00 15 eb]
[site  ] DEBUG 2025/05/23 16:37:35 pv 4 power: 78W

andig, 2025-05-23:
[image]

closed 2025-05-25 by andig