evcc-io/evcc #21328: No switch from 1p to 3p despite sufficient PV surplus [v0.203.4]

flobo1990, 2025-05-20:
### Describe the bug

EVCC does not usually switch from 1p to 3p despite sufficient PV surplus. The option is set to automatic switch.
If I restart EVCC, the corresponding symbol immediately appears in the interface.

The logs show the switch from 1p/3p only after a restart:
[lp-1 ] INFO 2025/05/20 10:15:43 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] DEBUG 2025/05/20 10:15:43 phase timer inactive
[lp-1 ] DEBUG 2025/05/20 10:16:20 start phase scale3p timer
[lp-1 ] DEBUG 2025/05/20 10:16:20 phase scale3p in 1m0s
[lp-1 ] DEBUG 2025/05/20 10:16:50 detected active phases: 3p

Auto switch 1p/3p:
[image]

After restart of EVCC:
[image]

EVCC.yaml :

loadpoints:
  - title: Garage
    charger: go_e_wallbox_custom
    vehicle: ford_explorer
    mode: pv
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true
    enable:
      delay: 1m
      threshold: 0
    disable:
      delay: 10m
      threshold: 200
    phases: 0
    mincurrent: 6
    maxcurrent: 16
    priority: 0

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
network:
  port: 7070

log: debug

interval: 30s

site:
  title: Zuhause
  meters:
    grid: grid
    pv: my_pv
  residualPower: 0

loadpoints:
  - title: Garage
    charger: go_e_wallbox_custom
    vehicle: ford_explorer
    mode: pv
    soc:
      poll:
        mode: charging
        interval: 60m
      estimate: true
    enable:
      delay: 1m
      threshold: 0
    disable:
      delay: 10m
      threshold: 200
    phases: 0
    mincurrent: 6
    maxcurrent: 16
    priority: 0

chargers:
  - name: go_e_wallbox
    type: template
    template: go-e
    host: 

  - name: go_e_wallbox_custom
    type: custom

    ### read ###
    status:
      source: mqtt
      topic: charger/status

    enabled:
      source: mqtt
      topic: go-eCharger/276582/alw

    power:
      source: mqtt
      topic: go-eCharger/276582/nrg
      jq: .[11]

    energy:
      source: mqtt
      topic: go-eCharger/276582/eto
      scale: 0.001

    powers: # Phasenleistungen in W
      - source: mqtt # Phase 1 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[8]
      - source: mqtt # Phase 2 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[9]
      - source: mqtt # Phase 3 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[10]

    currents: # Stromstärke (A) pro Phase
      - source: mqtt # Phase 1 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[4]
      - source: mqtt # Phase 2 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[5]
      - source: mqtt # Phase 3 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[6]

    voltages: # Phasenspannungen in V
      - source: mqtt # Phase 1 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[0]
      - source: mqtt # Phase 2 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[1]
      - source: mqtt # Phase 3 Plugin Typ
        topic: go-eCharger/276582/nrg
        jq: .[2]

    ### write ###
    enable: # Ladung freigeben / sperren
      source: mqtt
      topic: charger_set/enable
      timeout: 30s

    maxcurrent:
      source: mqtt
      topic: charger_set/ama
      timeout: 30s

    tos: true 

    phases1p3p:
      source: mqtt
      topic: charger_set/psm
      timeout: 30s

meters:
  - name: grid
    type: custom
    power: #Power # default value, optionally override # Leistung (W)
      source: mqtt
      topic: home/grid_power
      timeout: 30s
      # scale: -1
    energy:
      source: mqtt
      topic: home/grid_energy
    currents: # Stromstärke (A) pro Phase
      - source: mqtt # Phase 1 Plugin Typ
        topic: home/grid_current_l1
      - source: mqtt # Phase 2 Plugin Typ
        topic: home/grid_current_l2
      - source: mqtt # Phase 3 Plugin Typ
        topic: home/grid_current_l3

  - name: my_pv
    type: template
    template: fronius-gen24
    usage: pv
    host: 
    port: 502 # Port (optional)

vehicles:
  - name: ford_explorer
    title: Ford Explorer
    type: custom
    capacity: 79.00
    soc:
        source: mqtt
        topic: explorer/soc
    range:
        source: mqtt
        topic: explorer/range
    status:
        source: mqtt
        topic: explorer/status
    odometer:
        source: mqtt
        topic: explorer/odometer

mqtt:
  broker: core-mosquitto:1883
  topic: evcc # root topic for publishing, set empty to disable
  user: 
  password:
```

### Log details

```text
issue:
[lp-1 ] DEBUG 2025/05/20 10:10:50 detected active phases: 1p

after restart of EVCC:
[lp-1 ] INFO 2025/05/20 10:15:43 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] DEBUG 2025/05/20 10:15:43 phase timer inactive
[lp-1 ] DEBUG 2025/05/20 10:16:20 start phase scale3p timer
[lp-1 ] DEBUG 2025/05/20 10:16:20 phase scale3p in 1m0s
[lp-1 ] DEBUG 2025/05/20 10:16:50 detected active phases: 3p
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

v0.203.6

Lako-Unbekannt, 2025-05-20:
In the Log you posted the countdown is beeing startet:
`[lp-1 ] DEBUG 2025/05/20 10:16:20 phase scale3p in 1m0s`
Can you post a longer log?

flobo1990, 2025-05-20:
After the charge has jumped back from 3p to 1p due to clouds, the whole thing now hangs in 1p again.
[image]

[image]

flobo1990, 2025-05-20:
Here is a longer log file:
[49686a9f_evcc_2025-05-20T10-13-33.099Z.log](https://github.com/user-attachments/files/20343477/49686a9f_evcc_2025-05-20T10-13-33.099Z.log)

Lako-Unbekannt, 2025-05-20:
Also soweit ich sehe, startet evcc den Timer:
`[lp-1  ] DEBUG 2025/05/20 10:16:20 start phase scale3p timer`
Wechselt zu 3 Phasen:
`[lp-1  ] DEBUG 2025/05/20 10:16:50 switched phases: 3p`
und lädt dann mit 3 Phasen:
`[lp-1  ] DEBUG 2025/05/20 10:17:20 pv charge current: 12.4A = 6A + 6.42A (-4431W @ 3p)`

Vileicht verstehe ich die Meldungen falsch, aber du bist dir sicher das es nicht gewechselt hat?

flobo1990, 2025-05-20:
Ja, in dem Teil des Logs schon. Aber das funktioniert eben erst nach einem Neustart von EVCC. Wenn ich keinen manuellen Neustart mache, verharrt EVCC bei 1p

jetronic18s, 2025-05-20:
> Ja, in dem Teil des Logs schon. Aber das funktioniert eben erst nach einem Neustart von EVCC. Wenn ich keinen manuellen Neustart mache, verharrt EVCC bei 1p. Ich weiß aber nicht, was sich für EVCC nach einem Neustart ändert.

Genau das gleiche habe ich mit v0.201.1 beobachtet, restart von evcc über das UI. Danach wird erst auf 3p umgeschaltet. 
Ich habe leider kein Logfile zur Hand.

VolkerK62, 2025-05-20:
Funktioniert es per CLI?
https://docs.evcc.io/docs/reference/cli/evcc_charger

Frage am Rande: Warum nicht einfach das go-e Template nutzen?

andig, 2025-05-20:
> Frage am Rande: Warum nicht einfach das go-e Template nutzen?

Reproduziere den Fehler bitte mit der vorhandenen Integration, anderenfalls müssen wir davon ausgehen, dass es nicht an evcc liegt.

closed 2025-05-20 by andig