evcc-io/evcc #21334: Add support for FoxESS Open API

hasselmm, 2025-05-20:
Inverters and batteries from FoxESS also support a cloud API for monitoring and configuration. See #16326 for initial discussion.

### Benefits

Supporting the Open API...

- would allow FoxESS owners to evaluate evcc without manipulating their photovoltaic installation
- would give access to basic evcc users who cannot, or do not want access the RS485 bus of their FoxESS inverters

### Limitations

This API isn't too reliable, mostly as it updates only once every five minutes. Still for basic use cases it should work. 

### Building Blocks

- API documentation: https://www.foxesscloud.com/public/i18n/en/OpenApiDocument.html
- GO implementation: https://github.com/jbub/foxesscloud

hasselmm, 2025-05-20:
So let's move the discussion from #16326 to here, @andig.

I am not a Go developer at all; C++ is my comfort zone.

Still, I own a FoxESS H3-10.0 with battery. Given that there is a working(?) Go implementation already doing the heavy lifting I could imaging to take a look at this.

But not being profound in Go I'd need some help: Is there already a good practice example in the code base, which I could use a starting point for hooking the library from @jbub?

andig, 2025-05-28:
https://github.com/evcc-io/evcc/issues/16326#issuecomment-2893829823 

No progress, closing

closed 2025-05-28 by andig