evcc-io/evcc #21342: Can't finish configuration with custom integrateddevice (http)

niklaswa, 2025-05-20:
**Describe the bug**
When I configure a grid meter, PV, tariff, and a custom HTTP charger (with source http, using Home Assistant API endpoints), the evcc UI does not display any data and only shows "let's start configurating". What am I missing or is this a bug?

**Steps to reproduce**
1. Set up evcc with the configuration below (including grid, PV, and a charger with HTTP endpoints to Home Assistant)
2. Start evcc and open the interface
3. Observe that the UI does not display data, only "let's start configurating"

**Configuration details**
```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: trace
levels:
  cache: error

# unique installation id
plant: xxxxx

interval: 30s # control cycle interval

meters:
  - type: template
    template: shelly-3em
    usage: grid
    host: 192.168.188.23
    name: grid1
  - type: template
    template: shelly-1pm
    usage: pv
    host: 192.168.188.106
    channel: 0
    name: pv2

chargers:
  - name: ha_switch
    type: custom
    icon: bike
    features:
      - integrateddevice
    status:
      source: const
      value: 'B'
    enabled:  # Read current state
      source: http
      headers:
        Content-Type: application/json
        Authorization: Bearer xxxxx
      uri: http://192.168.188.58:8123/api/states/switch.teckin_3_relay
      jq: '.state == "on"'
    enable:
      source: http
      uri: http://192.168.188.58:8123/api/services/switch/{{if .enable}}turn_on{{else}}turn_off{{end}}
      headers:
        Content-Type: application/json
        Authorization: Bearer xxxxx
      method: POST
      body: '{"entity_id": "switch.teckin_3_relay"}'
    maxcurrent:
      source: script
      cmd: echo 16
    power:
      source: http
      headers:
        Content-Type: application/json
        Authorization: Bearer xxxxx
      uri: http://192.168.188.58:8123/api/states/sensor.teckin_3_power
      jq: '.state | tonumber'

site:
  title: My home
  meters:
    grid: grid1
    pv:
      - pv2 
```

**Log details**
```
[ha_switch] TRACE 2025/05/20 17:26:28 GET http://192.168.188.58:8123/api/states/sensor.teckin_3_power
[ha_switch] TRACE 2025/05/20 17:26:28 {"entity_id":"sensor.teckin_3_power","state":"0.0","attributes":{"state_class":"measurement","unit_of_measurement":"W","device_class":"power","friendly_name":"Power-Switch-3 Teckin 3 Power"},"last_changed":"2025-05-20T14:39:05.737469+00:00","last_reported":"2025-05-20T14:39:05.737469+00:00","last_updated":"2025-05-20T14:39:05.737469+00:00","context":{"id":"01JVQ24HG9V05Y219A403R36CM","parent_id":null,"user_id":null}}
[ha_switch] TRACE 2025/05/20 17:26:28 GET http://192.168.188.58:8123/api/states/switch.teckin_3_relay
[ha_switch] TRACE 2025/05/20 17:26:28 {"entity_id":"switch.teckin_3_relay","state":"off","attributes":{"friendly_name":"E-Bike"},"last_changed":"2025-05-20T14:38:41.651720+00:00","last_reported":"2025-05-20T14:38:41.651720+00:00","last_updated":"2025-05-20T14:38:41.651720+00:00","context":{"id":"01JVQ23SYXZH3YMEA0BZFAM1VQ","parent_id":"01JVQ23G69TK7DDX3107359X16","user_id":null}}
```

**What type of operating system or environment does evcc run on?**
- [ ] Linux
- [ ] Windows
- [ ] Docker container
- [x] HomeAssistant Add-on
- [ ] macOS
- [ ] other

**External automation**
- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

**Nightly build**
- [ ] I have verified that the issue is reproducible with the latest nightly build

**Version**
Please fill in the output of `evcc -v`.
evcc version 0.203.6

VolkerK62, 2025-05-20:
You need to configure a loadpoint with this charger.

niklaswa, 2025-05-20:
Wow.... thank you, somehow I missed that. 😅 

By the way, a home assistant switch template would be awesome :)

closed 2025-05-20 by niklaswa

VolkerK62, 2025-05-20:
A notice to your charger:
if you use `type: switchsocket` instead of `type: custom` you do not need `status` and `maxcurrent`.
And the status switches automatically from B  to  C if power > 0