evcc-io/evcc #21348: Deaktivierung MQTT funktioniert nicht

VolkerK62, 2025-05-20:
Aus der Doku
```
# mqtt message broker
mqtt:
  broker: localhost:1883
  topic: evcc # root topic for publishing, set empty to disable publishing
  # clientid: foo
  # user:
  # password:
```

Wenn man `topic` leer lässt oder komplett auskommentiert, wird trotzdem noch nach `evcc` veröffentlicht.

Fehler in der Doku oder Fehler in der Implementierung?

jph76, 2025-11-04:
@andig Ich muss diesen Thread noch einmal anwärmen: Das Deaktivieren von MQTT funktioniert nicht, indem man die MQTT-Konfiguration aus der `evcc.yaml` löscht. Tatsächlich kann ich dort beliebigen Unfug konfigurieren. evcc ignoriert die Angaben.

Die Ursache scheint zu sein, dass in der `evcc.db` in der Tabelle `settings` der Key `mqtt` erhalten bleibt. Erst, nachdem ich diesen von Hand gelöscht habe, startet evcc ohne MQTT.

Ich kann dafür ein separates Ticket anlegen, müsste dann aber beim Kästchen "Fehler besteht im Nightly" flunkern, weil ich aktuell 0.209.6 verwende...

andig, 2025-11-04:
Na wenns im ui konfiguriert ist ists im ui konfiguriert?

jph76, 2025-11-05:
Ich nutze die UI-Konfiguration nicht.

andig, 2025-11-05:
> Die Ursache scheint zu sein, dass in der evcc.db in der Tabelle settings der Key mqtt erhalten bleibt

Na irgendwo muss es ja her kommen

jph76, 2025-11-06:
Wir werden es wohl nie erfahren. Egal, ein `delete from settings;` hat den Spuk bereinigt.