evcc-io/evcc #21355: evcc shows kwh instead of percentage in charging plan

ggolec, 2025-05-21:
### Describe the bug

Hi,
According to documentation [here](https://docs.evcc.io/en/docs/features/plans#create-charging-plan):

> To set a plan, click on Plan at the charging point. If evcc knows the charge level of your vehicle (online API) and the battery capacity, you can enter the desired charge level in % in addition to the departure time in the dialog.
I have Tesla Model Y and Tesla Wall Charger recognised correctly in evcc, their current capacity level in percentage but still when planning just I can select how many kwh want to charge instead of setting charge level in %. 

[image]
[image]

### Steps to reproduce

1. connect car to charger
2. evcc detects current batttery capacity
3. plan charge, in drop down list in charge goal should be selectable %%  instead of kWh. 

### Configuration details

```yaml
[main  ] INFO 2025/05/21 09:17:17 evcc 0.203.6
[main  ] INFO 2025/05/21 09:17:17 using config file: /etc/evcc.yaml
[db    ] INFO 2025/05/21 09:17:17 using sqlite database: /root/.evcc/evcc.db
[mqtt  ] INFO 2025/05/21 09:17:17 connecting evcc-746264270 at tcp://10.0.1.6:1883
[main  ] ERROR 2025/05/21 09:17:17 creating vehicle teslamodely failed: cannot create vehicle type 'template': cannot create vehicle type 'tesla': token expired

Konfiguration (/etc/evcc.yaml):

# open evcc at http://evcc.local:7070
# day config
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: info
levels:
  cache: error

# unique installation id
plant: *****
sponsortoken: *****

interval: 30s # control cycle interval

chargers:
- type: template
  template: twc3 
  host: tesla-wc3 
  name: wallbox1

loadpoints:
- title: Garage
  enable:
      # switch on when there is at least 2000 W surplus for 1 minute
      delay: 1m
      threshold: -2400
  disable:
      # switch off if more than 2000 W are drawn from the mains for 30 minutes
      delay: 30m
      threshold: 2000
  charger: wallbox1
  mode: pv

tariffs:
  currency: PLN # (default EUR)
  grid:
     type: fixed
     price: 0.71
     zones:
      - days: Mo-Fr
        hours: 7-13
        price: 0.9 # EUR/kWh
      - days: Mo-Fr
        hours: 13-19
        price: 0.71 # EUR/kWh
      - days: Mo-Fr
        hours: 19-22
        price: 1.09 # EUR/kWh
      - days: Sa,So
        price: 0.71 # EUR/kWh
#  feedin:
      #type: fixed
      #price: 0.3

meters:
- name: grid1
  type: custom
  power: # power (W)
    source: mqtt
    topic: /sensors/energy/inverter/ActivePower_PCC_Total
    scale: -10

- name: pv1
  type: custom
  power: # optional
    source: mqtt
    topic: /sensors/energy/inverter/ActivePower_Output_Total
    scale: 10
  energy: # optional
    source: mqtt
    topic: /sensors/energy/inverter/PV_Generation_Today
    scale: 0.01

vehicles:
 - name: teslamodely
   type: template
   template: tesla-command
   title: TeslaModelY
   clientId: ********
   accessToken: *****
   refreshToken: *****
   commandProxy: http://10.0.1.7:8080
   maxCurrent: 16

site:
  title: My home
  meters:
     grid: grid1
     pv: pv1
  residualPower: -600

mqtt:
  broker: 10.0.1.6:1883
  topic: evcc # root topic for publishing, set empty to disable

Version: `0.203.6`
```

### Log details

```text
none
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

naltatis, 2025-05-21:
Yes, what @RenatusRo says.

> If evcc knows the charge level of your vehicle (online API) **and the battery capacity**,

from the docs page you've linked above.

closed 2025-05-21 by naltatis

ggolec, 2025-05-22:
yes. it works after adding capacity to the config.
My incorrect understanding was: if evcc is able to detect charge level then
also should be able to show battery capacity in %% under charging plan.
anyway works fine now. thanks!