evcc-io/evcc #21359: After upgrade to 0.203.6 cannot startup

ambitzgi, 2025-05-21:
### Describe the bug

After the upgrade to 0.203.6 I get the following error message in the web interface:
"Error during startup. Check your configuration and restart.
charger: [wallbox5] cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'vestel': failsafe timeout: read tcp 192.168.178.61:55450->192.168.178.103:502: read: connection reset by peer"

Log messages: 
"[main ] INFO 2025/05/21 11:19:54 evcc 0.203.6
[main ] INFO 2025/05/21 11:19:54 using config file: /etc/evcc.yaml
[db ] INFO 2025/05/21 11:19:54 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/05/21 11:19:56 listening at :7070
[main ] FATAL 2025/05/21 11:19:57 charger [wallbox5] cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'vestel': failsafe timeout: read tcp 192.168.178.61:55450->192.168.178.103:502: read: connection reset by peer
[main ] FATAL 2025/05/21 11:19:57 will attempt restart in: 15m0s"

After a downgrade to 0.203.5, everything works again. 

### Steps to reproduce

1. Upgrade to 0.203.6 via "sudo apt install evcc=0.203.6"
2. Go to web interface

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

log: debug
levels:
  cache: error

# unique installation id
plant:

interval: 30s # control cycle interval

sponsortoken: 

# sponsors can set telemetry: true to enable anonymous data aggregation
# see https://github.com/evcc-io/evcc/discussions/4554
telemetry: false

meters:
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.78  
  port: 1502  
  usage: pv  
  modbus: tcpip  
  name: pv1
- type: template
  template: kostal-plenticore 
  id: 71  
  host: 192.168.178.78  
  port: 1502  
  usage: battery  
  modbus: tcpip  
  name: battery2
- type: template
  template: kostal-ksem-inverter 
  id: 71  
  host: 192.168.178.78  
  port: 1502  
  usage: grid  
  modbus: tcpip  
  name: grid3

chargers:
- type: template
  template: vestel 
  id: 255  
  host: 192.168.178.103  
  port: 502  
  modbus: tcpip  
  name: wallbox5

vehicles:
- type: template
  template: vw 
  user:   
  password:  
  vin:
  title:   
  capacity: 86  
  timeout: 10s  
  name: ev4

loadpoints:
- title: Garage
  charger: wallbox5
  mode: pv

site:
  title: Sonder 4A
  meters:
    grid: grid3
    pv:
    - pv1
    battery:
    - battery2

tariffs:
  currency: CHF # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.2597 # [currency]/kW
    zones:
      - days: Mo-Fr
        hours: 07-19
        price: 0.3440 # CHF/kWh
  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.138 # [currency]/kWh 0.115 + 0.023
  co2:
    type: template
    template: electricitymaps-free
    token:
    zone: CH # siehe https://api.electricitymap.org/v3/zones
  solar:
    - type: template
      template: forecast-solar
      lat: 
      lon: 
      dec: 24.8 # 0 = horizontal, 90 = vertikal
      kwp: 11.1
      az: 38 # -180 = Norden, -90 = Osten, 0 = Süden, 90 = Westen, 180 = Norden
```

### Log details

```text
[main ] INFO 2025/05/21 11:45:45 evcc 0.203.6 (c7898f06)
[main ] INFO 2025/05/21 11:45:45 using config file: /etc/evcc.yaml
[db ] INFO 2025/05/21 11:45:45 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/05/21 11:45:45 listening at :7070
[main ] FATAL 2025/05/21 11:45:46 charger [wallbox5] cannot create charger 'wallbox5': cannot create charger type 'template': cannot create charger type 'vestel': failsafe timeout: read tcp 192.168.178.61:45618->192.168.178.103:502: read: connection reset by peer
[main ] FATAL 2025/05/21 11:45:46 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.6 (c7898f06)

ambitzgi, 2025-05-21:
I have a Webasto Unite, using the vestel template.
HMI Softwarestand :  v2.99.11
Leistungsplatine Softwarestand :  1.0.109.0

StefanSchoof, 2025-05-21:
@thomkal 

> read: connection reset by peer

> cannot create charger type 'ocpp': timeout

That a totally different errors. So it is pretty unlikely that they are connected.

If a new version shows an error many times the problem has nothing to do with the update. So please verify if a downgrade fixes your problem or if it is only a coincidence. And if so you should create your a new bug

thomkal, 2025-05-21:
> [@thomkal](https://github.com/thomkal)
>
>
>
> That a totally different errors. So it is pretty unlikely that they are connected.
>
> If a new version shows an error many times the problem has nothing to do [...]

Hi Stefan
Will open a new one. 
Best regards Thomas

mfuchs1984, 2025-05-21:
See https://github.com/evcc-io/evcc/discussions/21172#discussioncomment-13180836

ambitzgi, 2025-05-21:
Thank you mfuchs1984. 
I am currently updating the Webasto-Firmware from 2.99.11 via several intermediate versions to the final Webasto-Version 3.156.0. Then I will test EVCC v0.203.6 again. Hopefully this will fix this issue.

mfuchs1984, 2025-05-21:
Let me know if it works with 3.156.0. currently we only know 1p3p and RFID works with 3.187.0 and does not work with 3.118.0.

In case it does not work, you could maybe use the firmware provided by Vestel or temporarily downgrade evcc. 

Next week, I might find time to do the RFID decoration by reading the version instead of attempting to read the RFID registers which apparently triggers the "connection reset by peer" error.

ambitzgi, 2025-05-21:
I just have updated to version 3.156.0. It seems to work properly. Unfortunately right now I don't have the time to test it extensively. However the error message has disappeared.

ambitzgi, 2025-05-21:
Everything looks good. I haven't found any problem. For it is ok to close this issue. 

Just one question: On the longer run I intend to upgrade to the latest Vestel firmware. 
But before I do this, should I test the 1p3p switching? How can I test it? Can I test it without charging a car? Just because we will receive our e-car in about 2 weeks.

mfuchs1984, 2025-05-21:
A log showing the startup of evc should be enough.it shows the charger features.

ambitzgi, 2025-05-21:
Here is a log of a start-up of evcc (log level = debug, if necessary I can send you one with level = trace)

main ] INFO 2025/05/21 23:06:28 evcc 0.203.6
[main ] INFO 2025/05/21 23:06:28 using config file: /etc/evcc.yaml
[db ] INFO 2025/05/21 23:06:28 using sqlite database: /var/lib/evcc/evcc.db
[main ] INFO 2025/05/21 23:06:28 listening at :7070
[main ] ERROR 2025/05/21 23:06:30 creating vehicle ev4 failed: cannot create vehicle type 'template': cannot create vehicle type 'vw': cannot find vehicle, got: []
[lp-1 ] DEBUG 2025/05/21 23:06:31 charge total import: 0.100kWh
[site ] INFO 2025/05/21 23:06:31 site config:
[site ] INFO 2025/05/21 23:06:31 meters: grid ✓ pv ✓ battery ✓
[site ] INFO 2025/05/21 23:06:31 grid: power ✓ energy ✗ currents ✗
[site ] INFO 2025/05/21 23:06:31 pv 1: power ✓ energy ✓ currents ✗
[site ] INFO 2025/05/21 23:06:31 battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
[site ] INFO 2025/05/21 23:06:31 vehicles:
[site ] INFO 2025/05/21 23:06:31 vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site ] INFO 2025/05/21 23:06:31 tariffs:
[site ] INFO 2025/05/21 23:06:31 grid: priceforecast
[site ] INFO 2025/05/21 23:06:31 feed-in: pricestatic
[site ] INFO 2025/05/21 23:06:31 co2: ✓
[site ] INFO 2025/05/21 23:06:31 solar: ✓
[lp-1 ] INFO 2025/05/21 23:06:31 loadpoint 1:
[lp-1 ] INFO 2025/05/21 23:06:31 mode: pv
[lp-1 ] INFO 2025/05/21 23:06:31 charger: power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
[lp-1 ] INFO 2025/05/21 23:06:31 meters: charge ✓
[lp-1 ] INFO 2025/05/21 23:06:31 charge: power ✓ energy ✓ currents ✓
[site ] DEBUG 2025/05/21 23:06:31 set buffer soc: 85
[site ] DEBUG 2025/05/21 23:06:31 set priority soc: 10
[lp-1 ] DEBUG 2025/05/21 23:06:31 phase timer inactive
[lp-1 ] DEBUG 2025/05/21 23:06:31 pv timer inactive
[site ] DEBUG 2025/05/21 23:06:31 ----
[lp-1 ] DEBUG 2025/05/21 23:06:31 charge power: 0W
[lp-1 ] DEBUG 2025/05/21 23:06:31 charge currents: [0 0 0]A
[site ] DEBUG 2025/05/21 23:06:31 grid power: 543W
[site ] DEBUG 2025/05/21 23:06:31 battery 1 power: 0W
[site ] DEBUG 2025/05/21 23:06:31 battery 1 soc: 0%
[site ] DEBUG 2025/05/21 23:06:31 pv 1 power: 0W
[site ] DEBUG 2025/05/21 23:06:32 site power: 643W
[lp-1 ] DEBUG 2025/05/21 23:06:32 charge total import: 0.100kWh
[lp-1 ] DEBUG 2025/05/21 23:06:32 charger status: A
[lp-1 ] INFO 2025/05/21 23:06:32 car disconnected
[lp-1 ] DEBUG 2025/05/21 23:06:32 set battery boost: false
[lp-1 ] DEBUG 2025/05/21 23:06:32 set session soc limit: 0
[lp-1 ] DEBUG 2025/05/21 23:06:32 set session energy limit: 0

naltatis, 2025-05-21:
Can we do a firmware version check to provide a better error message in this case?
\\cc @premultiply

PR #21519 (2025-05-30): Vestel: decorate RFID based on version
Follow-up to #21124

The current RFID decoration mechanism (attempt to read the registers) leads to a "connection reset by peer" error with versions not supporting the RFID registers. This PR decorates RFID based on version, the minimum version we know it works with is v3.156.0, see https://github.com/evcc-io/evcc/issues/21359#issuecomment-2898162220

Since I have no EVC04, I cannot test it.

Fixes #21359, #21313, #21172