evcc-io/evcc #21361: planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'entsoe':

tiemenbossijbe, 2025-05-21:
### Describe the bug

My ENTSOE integration with EVCC does no longer work and I am struggeling to understand why.

It has been working fine for some time now but since yesterday I am getting an error whereas my integration in Home Assistant itself works fine (hence ENTSOE is not broken this time)

planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'entsoe': Get \"https://web-api.tp.entsoe.eu/api?DocumentType=A44&In_Domain=10YNL----------L&Out_Domain=10YNL----------L&PeriodEnd=202505220600&PeriodStart=202505210600&securityToken=688a64ec-67dc-490e-ba2e-1ac41a76e993\": dial tcp 62.233.0.8:443: i/o timeout

My code as per information from EVCC:

tariffs:
  grid:
    type: template
    template: entsoe
    domain: BZN|NL
    securitytoken: XX
    charges: 0.1194
    tax: 0.21

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
tariffs:
  grid:
    type: template
    template: entsoe
    domain: BZN|NL
    securitytoken: XX
    charges: 0.1194
    tax: 0.21
```

### Log details

```text
site ] WARN 2025/05/21 13:28:33 planner: tariff not available: cannot create tariff type 'template': cannot create tariff type 'entsoe': Get "https://web-api.tp.entsoe.eu/api?DocumentType=A44&In_Domain=10YNL----------L&Out_Domain=10YNL----------L&PeriodEnd=202505220600&PeriodStart=202505210600&securityToken=688a64ec-67dc-490e-ba2e-1ac41a76e993": dial tcp 62.233.0.8:443: i/o timeout
[lp-1 ] DEBUG 2025/05/21 13:28:33 charge power: 11744W
[lp-1 ] DEBUG 2025/05/21 13:28:33 charge currents: [16.3 16.2 16.1]A
[site ] DEBUG 2025/05/21 13:28:33 grid power: 7936W
[site ] DEBUG 2025/05/21 13:28:33 pv 1 power: 4274W
[site ] DEBUG 2025/05/21 13:28:33 pv 2 power: 2037W
[site ] DEBUG 2025/05/21 13:28:33 pv power: 6311W
[site ] DEBUG 2025/05/21 13:28:33 grid currents: [1.58e+03 3.12e+03 3.22e+03]A
[site ] DEBUG 2025/05/21 13:28:33 site power: 7936W
[lp-1 ] DEBUG 2025/05/21 13:28:33 charge voltages: [238 244 243]V
[lp-1 ] DEBUG 2025/05/21 13:28:33 detected connected phases: 3p
[lp-1 ] DEBUG 2025/05/21 13:28:33 detected active phases: 3p
[lp-1 ] DEBUG 2025/05/21 13:28:33 charge total import: 5179.245kWh
[lp-1 ] DEBUG 2025/05/21 13:28:33 charger status: C
[lp-1 ] DEBUG 2025/05/21 13:28:33 soc estimated: 63.31% (vehicle: 61.00%)
[lp-1 ] DEBUG 2025/05/21 13:28:33 vehicle soc: 63%
[lp-1 ] DEBUG 2025/05/21 13:28:33 vehicle soc limit: 100%
[lp-1 ] DEBUG 2025/05/21 13:28:33 vehicle range: 215km
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

RenatusRo, 2025-05-21:
was a certificate update probably (13/5):

Dear Transparency Users,
Attention for API user and Subscribers via WS channel. Our service provider will renew the certificate of the ENTSO-E domain (*.tp.entsoe.eu) in IOP and PROD Transparency Platform on following dates:
IOP (*.tp-iop.entsoe.eu) -  06.05.2023 at 14:00 CEST
Production (*.tp.entsoe.eu) - 13.05.2023 at 15:00 CEST
As a result there may be issues with API and Subscriptions via WS channels. In order to avoid issues its recommended to download the new certificates from respective test and production public web sites:
https://web-api.tp-iop.entsoe.eu/api
https://web-api.tp.entsoe.eu/api
The SSL certificate can be downloaded via your web browser by clicking the Secure button in an address bar (left of the URL), as per screenshot bellow:

closed 2025-05-21 by naltatis

dlopesa, 2025-05-22:
its gone again no?

RenatusRo, 2025-05-22:
will come back ...
https://transparency.entsoe.eu/news/widget?id=682dacbd11e21d284aea2796

dlopesa, 2025-05-22:
That was yesterday, it seemed like today it went down without any type of warning or announcement...

tiemenbossijbe, 2025-05-23:
For me it still not works in EVCC. Works fine in HA still.