evcc-io/evcc #21371: v0.203.6 Does not start: cannot create charger type 'ocpp'

thomkal, 2025-05-21:
### Describe the bug

After Upgrade to v.0.203.6 EVCC does not start with following Message:

`charger [garage] cannot create charger 'garage': cannot create charger type 'template': cannot create charger type 'ocpp': timeout`

### Steps to reproduce

1. Upgrade from 0.203.5 to 0.203.6
2. Start Addon

### Configuration details

```yaml
## Log
# log settings
log: info
levels:
  site: debug
  lp-1: debug
  lp-2: debug
  cache: error
  db: error

## minimal configuration example

site:
  title: Home # display name for UI
  meters:
    grid: my_grid
    pv:
      - my_pv
    battery:
      - my_battery

# define your loadpoints according your needs
# see https://docs.evcc.io/en/docs/reference/configuration/loadpoints
loadpoints:
  - title: Garage # display name for UI
    charger: garage # charger
    meter: my_shelly #shelly 3em pro
    vehicle: my_car # default vehicle

# trial token, valid endless 15000 days
sponsortoken: xxxx
# meter definitions
# name can be freely chosen and is used as reference when assigning meters to site and loadpoints
# for documentation see https://docs.evcc.io/docs/devices/meters
meters:
  # replace with your real grid meter
  - name: my_grid
    type: template
    template: fronius-gen24
    usage: grid
    host: xxx # IP-Adresse oder Hostname
  #  port: 502 # Port (optional)
  # replace with your real pv system
  - name: my_pv
    type: template
    template: fronius-gen24
    usage: pv
    host: xxx # IP-Adresse oder Hostname
  #  port: 502 # Port (optional)
  # replace with your real battery
  - name: my_battery
    type: template
    template: fronius-gen24
    usage: battery
    host: xxx # IP-Adresse oder Hostname
  #  port: 502 # Port (optional)
  - name: my_shelly
    type: template
    template: shelly-pro-3em
    usage: charge
    host: xxx # IP-Adresse oder Hostname
    #user: # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.) (optional)
    #password: # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen) (optional)

# replace with your real charger
# see https://docs.evcc.io/docs/devices/chargers
chargers:
  - name: garage
    type: template
    template: elvi
    stationid: EVB-xxx
    connector: 1
  #  connecttimeout: 5m
  #  timeout: 2m
  #  meter: false # optional

  #- name: my_charger
  #  type: template
  #  template: demo-charger
  #  status: C # charging
  #  power: 1 # 2 kW charging power
  #  enabled: true # optional

# replace with your real vehicle (optional)
# see https://docs.evcc.io/docs/devices/vehicles
vehicles:
  - name: my_car
    type: custom
    title: Jaguar I-Pace
    capacity: 90 # in kWh

# enter your real grid tariff and feed-in price
# see https://docs.evcc.io/docs/tariffs
tariffs:
  currency: CHF
  grid:
    type: fixed
    price: 0.29 # CHF/kWh
  feedin:
    type: fixed
    price: 0.10 # CHF/kWh
```

### Log details

```text
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/21 17:37:13 evcc 0.203.6
[main  ] INFO 2025/05/21 17:37:13 using config file: /config/evcc.yaml
[main  ] INFO 2025/05/21 17:37:13 listening at :7070
[main  ] FATAL 2025/05/21 17:42:14 charger [garage] cannot create charger 'garage': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2025/05/21 17:42:14 will attempt restart in: 15m0s
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/05/21 17:53:03 evcc 0.203.6
[main  ] INFO 2025/05/21 17:53:03 using config file: /config/evcc.yaml
[main  ] INFO 2025/05/21 17:53:03 listening at :7070
[main  ] FATAL 2025/05/21 17:58:04 charger [garage] cannot create charger 'garage': cannot create charger type 'template': cannot create charger type 'ocpp': timeout
[main  ] FATAL 2025/05/21 17:58:04 will attempt restart in: 15m0s
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

mfuchs1984, 2025-05-21:
Tried rebooting the charger?

thomkal, 2025-05-21:
> Tried rebooting the charger?

Tried is the right question, looks like my evbox Elvi is out of order completely.

closed 2025-05-21 by thomkal