evcc-io/evcc #21376: Umschalten auf "MIN+PV laden" führt zumindest in der Nacht zu ungewünschtem "VOLLGAS LADEN" mit 2x 16A = 7,1 kW

typxxi, 2025-05-22:
### Describe the bug

Ich habe heute Nacht bei ausreichendem Ladestand das Laden starten wollen, was mit PV nicht geht, so dass ich MIN+PV nehmen muss(te).

Die Stromstärke zum LADEN - aktuell ist nur 1x Speicher mit 14 kWh und 3 kW Entladeleistung aktiv - ist von 6A bis 6A, wobei das Fahrzeug nur 2 Phasen hat und nur das Laden von 6A bis 16A unterstützt. 

[image]

Der Lader springt aber sofort auf "VOLLGAS LADEN" trotz der OBIGEN Voreinstellung "max. 6A" (screenshot oben), wie hier unter im Screenshot zu sehen ist.

[image]

* PULSAR PLUS WALLBOX mit EASTRON SDM630 Smartmeter
* GROWATT SPH 4600 WECHSELRICHTER mit EEL V4 BATTERY (JK BMS) 
* e-GOLF 2021
* unter HomeAssistant Addon

### Steps to reproduce

1. auf MiN + PV wechseln
2. Fahrzeug verbinden
3. Fahrzeug lädt sofort mit 2x 16 A = 7,2 kW, obschon 6 A als Limit eingestellt
...

### Configuration details

```yaml
funktioniert ansonsten
evcc dump --cfg klappt nicht mit homeassistant addon bzw. ich weiß nicht , wie ich es unter docker aufzurufen habe.
```

### Log details

```text
Nach "/l1: '15.7'" suchen, 
da lädt das Fahrzeug mit 15,7 und 15,9 A statt 6 A wie vorgegeben.
================

[ocpp  ] TRACE 2025/05/22 03:40:33 recv 251327: [2,"263065973","Heartbeat",{}]
[ocpp  ] TRACE 2025/05/22 03:40:33 send 251327: [3,"263065973",{"currentTime":"2025-05-22T01:40:33Z"}]
[mqtt  ] TRACE 2025/05/22 03:40:36 recv export/golf/soc_permanent: '46.4'
[mqtt  ] TRACE 2025/05/22 03:40:36 recv export/golf/reichweite_permanent: '115.0'
[mqtt  ] TRACE 2025/05/22 03:40:36 recv export/golf/km_stand_permanent: '69247'
[httpd ] TRACE 2025/05/22 03:40:36 GET /api/tariff/planner
[mqtt  ] TRACE 2025/05/22 03:40:40 recv export/golf/soc_permanent: '46.4'
[mqtt  ] TRACE 2025/05/22 03:40:40 recv export/golf/reichweite_permanent: '111.0'
[mqtt  ] TRACE 2025/05/22 03:40:40 recv export/golf/km_stand_permanent: '69247'
[ocpp  ] TRACE 2025/05/22 03:40:40 send 251327: [2,"1698976409","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp  ] TRACE 2025/05/22 03:40:40 recv 251327: [3,"1698976409",{"status": "Accepted"}]
[ocpp  ] TRACE 2025/05/22 03:40:40 recv 251327: [2,"935868504","MeterValues",{"meterValue": [{"timestamp": "2025-05-22T01:40:40Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "20047.3"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "7079.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "15.7"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "15.9"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1-N","measurand": "Voltage","unit": "V","value": "221.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2-N","measurand": "Voltage","unit": "V","value": "227.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "234.0"} ]} ],"connectorId": 1,"transactionId": 1747867757}]
[ocpp  ] TRACE 2025/05/22 03:40:40 send 251327: [3,"935868504",{}]
[site  ] DEBUG 2025/05/22 03:40:40 ----
[lp-2  ] DEBUG 2025/05/22 03:40:40 charge power: 0W
[lp-2  ] DEBUG 2025/05/22 03:40:40 charge currents: [0 0 0]A
[lp-1  ] DEBUG 2025/05/22 03:40:40 charge power: 7079W
[lp-1  ] DEBUG 2025/05/22 03:40:40 charge currents: [15.7 15.9 0]A
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/2/chargeCurrents: '0'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/1/chargeCurrents: '31.6'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/updated: '1747878040'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/2/chargePower: '0'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/2/chargeCurrents/l2: '0'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/2/chargeCurrents/l3: '0'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/2/chargeCurrents/l1: '0'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/1/chargePower: '7079'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/1/chargeCurrents/l3: '0'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/1/chargeCurrents/l1: '15.7'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/loadpoints/1/chargeCurrents/l2: '15.9'
[mqtt  ] TRACE 2025/05/22 03:40:40 send evcc/site/batteryGridChargeActive: 'false'
[site  ] DEBUG 2025/05/22 03:40:40 pv 2 power: 0W
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/22 03:40:40 modbus: send 01 04 03 fd 00 02 e0 7f
[... 67 lines cut ...]
[mqtt  ] TRACE 2025/05/22 03:40:41 send evcc/site/battery/1/power: '1220'
[mqtt  ] TRACE 2025/05/22 03:40:41 send evcc/site/battery/1/energy: '3115.9'
[lp-2  ] DEBUG 2025/05/22 03:40:41 charge voltages: [223 242 235]V
[mqtt  ] TRACE 2025/05/22 03:40:41 send evcc/site/battery/1/powers: ''
[mqtt  ] TRACE 2025/05/22 03:40:41 send evcc/site/battery/1/currents: ''
[lp-2  ] DEBUG 2025/05/22 03:40:41 detected connected phases: 3p
[mqtt  ] TRACE 2025/05/22 03:40:41 send evcc/site/battery/1/excessDCPower: ''
[mqtt  ] TRACE 2025/05/22 03:40:41 send evcc/site/battery/1/soc: '39'
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

detlefheese, 2025-05-22:
Also kurzzeitig ist das normal. Die Wallbox gibt "Vollgas" damit evcc merkt "was geht" ... später wird dann runtergeregelt. Je nach Batterieeinstellung oder PV-Leistung wird dann weitergeladen oder hold bis genug Sonne kommt.

typxxi, 2025-05-22:
Also tagsüber hab ich das so nicht gesehen. Da kam die Wallbox und lag bei Wolken oder hoher Last im Haus sofort bei 3,6 kW und 7 kW hab ich gar nicht gesehen.
Jetzt war es in der Nacht und ich mein das wäre bestimmt 2 Minuten später gewesen, weil ich in HA plötzlich eine hohe Last bemerkt habe und dann erst zu evcc zurück bin. 

Beim Protokoll / Log hab ich das Ganze aber von 10.000 Zeilen auf ich meine 10.000 Zeichen kürzen müssen, so dass man nix davon sehen kann, wie lange das aktiv war. Muss ich notfalls heut Nacht ohne PV wieder neu probieren.

mfuchs1984, 2025-05-22:
In deinen log fehlt das Kommando zum Ladestart durch evcc. Darin würde man sehen, welchen Strom evcc vorgibt.

typxxi, 2025-05-22:
> In deinen log fehlt das Kommando zum Ladestart durch evcc. Darin würde man sehen, welchen Strom evcc vorgibt.

Kannst Du mir sagen, wie das Kommando aussieht ?
Ich habe 10000 Zeilen gesichert und mir sagt das eher nix.

mfuchs1984, 2025-05-23:
`SetChargingProfile`

typxxi, 2025-05-23:
> Also kurzzeitig ist das normal. Die Wallbox gibt "Vollgas" damit evcc merkt "was geht" ... später wird dann runtergeregelt. Je nach Batterieeinstellung oder PV-Leistung wird dann weitergeladen oder hold bis genug Sonne kommt.

Ich hab das gestern Abend wieder getestet und das gleiches Ergebnis. 
1. Es wird immer beim Anstecken geladen und zwar mit 7,1 kW = 16 A Vollgas
2. Die Einstellung von 6 A bis 6 A wird ignoriert
3. Keine Regelung in den ersten 3 Minuten und ich habe dann abgebrochen
3. Danach die gleiche Prozedur bei der rechten, der 2. Pulsar Plus Wallbox (die hängen an der gleichen 11 kW Leitung und werden nur wechselweise, da die rechte Überschussladen macht und die linke im Winter "Octopus Intelligent Go Laden" von 0 Uhr bis 5 Uhr für 14 Cent, da Octopus eine ocpp Verbindung braucht) 

[image]

Hier sieht man beide Pulsar Plus Wallboxen, die rechte funktioniert sauber und hat sofort auf 6A geregelt ohne 16A / 7,1 kW "anzutesten". 
Die rechte ist eine Pulsar Plus 11 kW, die linke eine 22 kW Version, aber beide auf 16A begrenzt.

detlefheese, 2025-05-23:
Greift vielleicht ein Lastmanagement von der ZuLeitung (16A) oder kannst du beide full Power laufen lassen? Wie ist es mit der Prio-Einstellung?

typxxi, 2025-05-23:
> In deinen log fehlt das Kommando zum Ladestart durch evcc. Darin würde man sehen, welchen Strom evcc vorgibt.

Ich glaub, ich hab da jetzt etwas gefunden, also von heute 

```
[lp-1 ] DEBUG 2025/05/23 09:35:23 pv charge current: 6.56A = 15.6A + -9.04A (4160W @ 2p)
[lp-1 ] DEBUG 2025/05/23 09:35:24 set charge current limit: 6A	

```
aus 
```
[site ] DEBUG 2025/05/23 09:35:23 battery 1 soc: 69%
[site ] DEBUG 2025/05/23 09:35:23 site power: 4160W
[lp-1 ] DEBUG 2025/05/23 09:35:23 charge voltages: [222 231 235]V
[lp-1 ] DEBUG 2025/05/23 09:35:23 detected connected phases: 3p
[lp-1 ] DEBUG 2025/05/23 09:35:23 detected active phases: 2p
[lp-1 ] DEBUG 2025/05/23 09:35:23 charge total import: 34.324kWh
[lp-1 ] DEBUG 2025/05/23 09:35:23 charger status: C
[lp-1 ] DEBUG 2025/05/23 09:35:23 soc estimated: 93.78% (vehicle: 93.60%)
[lp-1 ] DEBUG 2025/05/23 09:35:23 vehicle soc: 94%
[lp-1 ] DEBUG 2025/05/23 09:35:23 vehicle range: 202km
[lp-1 ] WARN 2025/05/23 09:35:23 charger logic error: current mismatch (got 15.6A measured, expected 6A) - make sure your interval is at least 30s
[lp-1 ] DEBUG 2025/05/23 09:35:23 pv charge current: 6.56A = 15.6A + -9.04A (4160W @ 2p)
[lp-1 ] DEBUG 2025/05/23 09:35:24 set charge current limit: 6A	
[site ] DEBUG 2025/05/23 09:35:24 solar forecast: accumulated 11.639kWh, produced 8.100kWh, scale 0.696
[site ] DEBUG 2025/05/23 09:35:31 ----
[lp-2 ] DEBUG 2025/05/23 09:35:31 charge power: 0W
[lp-2 ] DEBUG 2025/05/23 09:35:31 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/05/23 09:35:31 charge power: 7074W
[lp-1 ] DEBUG 2025/05/23 09:35:31 charge currents: [15.6 15.7 0]A
[site ] DEBUG 2025/05/23 09:35:31 pv 2 power: 2799W
[site ] DEBUG 2025/05/23 09:35:31 grid power: 990W
[site ] DEBUG 2025/05/23 09:35:32 pv 1 power: 1080W
[site ] DEBUG 2025/05/23 09:35:32 battery 1 power: 2990W
[site ] DEBUG 2025/05/23 09:35:32 pv power: 3879W
[site ] DEBUG 2025/05/23 09:35:32 battery 1 soc: 69%
[site ] DEBUG 2025/05/23 09:35:32 site power: 4130W
[lp-2 ] DEBUG 2025/05/23 09:35:32 charge voltages: [221 242 233]V
[lp-2 ] DEBUG 2025/05/23 09:35:32 detected connected phases: 3p
[lp-2 ] DEBUG 2025/05/23 09:35:32 charge total import: 1950.105kWh
[lp-2 ] DEBUG 2025/05/23 09:35:32 charger status: A
[site ] DEBUG 2025/05/23 09:35:32 solar forecast: accumulated 11.656kWh, produced 8.100kWh, scale 0.695
```

Der Fehler , der da auftaucht, kann nicht stimmen, weil ich das Intervall schon vor Tagen auf 32 Sek. hochgesetzt habe
`[lp-1 ] WARN 2025/05/23 09:35:23 charger logic error: current mismatch (got 15.6A measured, expected 6A) - make sure your interval is at least 30s`

Für diesen Moment so um 09:35:32 habe ich auch noch die TRACE logs sichern können, falls es da noch etwas zu prüfen gäbe.

mfuchs1984, 2025-05-23:
Die Wallbox reagiert nicht auf das Kommando von evcc. 

- Haben beide die gleiche Firmware version?
- sind beide exakt gleich konfiguriert (nicht in evcc, sondern in den Einstellungen der Wallboxen selbst)?

andig, 2025-05-23:
> Es fehlt weiterhin ein brauchbares OCPP trace log.

@typxxi ich mache hier zu bis es das gibt.

closed 2025-05-23 by andig

typxxi, 2025-05-23:
Ich habe das hier aus TRACE geholt und manuell alle mqtt rausgeworfen.

```
[ocpp ] TRACE 2025/05/23 17:21:32 send 251327: [3,"2105256230",{}]
[site ] DEBUG 2025/05/23 17:21:34 ----
[lp-2 ] DEBUG 2025/05/23 17:21:34 charge power: 0W
[lp-2 ] DEBUG 2025/05/23 17:21:34 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/05/23 17:21:34 charge power: 7253W
[lp-1 ] DEBUG 2025/05/23 17:21:34 charge currents: [15.6 16 0]A
[site ] DEBUG 2025/05/23 17:21:34 pv 2 power: 6841W
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/23 17:21:34 modbus: send 01 04 03 fd 00 02 e0 7f
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/23 17:21:34 modbus: recv 01 04 04 00 00 68 b0 d5 f0
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/23 17:21:34 modbus: send 01 04 04 05 00 02 60 fa
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/23 17:21:34 modbus: recv 01 04 04 00 00 00 00 fb 84
[site ] DEBUG 2025/05/23 17:21:34 grid power: 2680W
[battery_1] TRACE 2025/05/23 17:21:34 modbus: send 01 04 03 f1 00 02 20 7c
[battery_1] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 04 00 00 00 00 fb 84
[pv_1_hybrid_wechselrichter] TRACE 2025/05/23 17:21:35 modbus: send 01 04 00 01 00 02 20 0b
[pv_1_hybrid_wechselrichter] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 04 00 00 43 f8 cb 36
[site ] DEBUG 2025/05/23 17:21:35 pv 1 power: 1740W
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/23 17:21:35 modbus: send 01 04 04 16 00 02 91 3f
[netz_meter_eastron_via_hybrid_wr] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 04 00 00 77 d8 dc 2e
[battery_1] TRACE 2025/05/23 17:21:35 modbus: send 01 04 03 f3 00 02 81 bc
[battery_1] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 04 00 00 00 00 fb 84
[site ] DEBUG 2025/05/23 17:21:35 battery 1 power: 0W
[pv_1_hybrid_wechselrichter] TRACE 2025/05/23 17:21:35 modbus: send 01 04 00 5b 00 02 00 18
[pv_1_hybrid_wechselrichter] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 04 00 00 d5 8d 65 71
[site ] DEBUG 2025/05/23 17:21:35 pv power: 8581W
[battery_1] TRACE 2025/05/23 17:21:35 modbus: send 01 04 04 1e 00 02 10 fd
[battery_1] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 04 00 00 7a 42 58 d5
[battery_1] TRACE 2025/05/23 17:21:35 modbus: send 01 04 03 f6 00 01 d1 bc
[battery_1] TRACE 2025/05/23 17:21:35 modbus: recv 01 04 02 00 63 f9 19
[site ] DEBUG 2025/05/23 17:21:35 battery 1 soc: 99%
[mqtt ] TRACE 2025/05/23 17:21:35 send evcc/site/batterySoc: '99'
[site ] DEBUG 2025/05/23 17:21:35 site power: 2830W
[lp-1 ] DEBUG 2025/05/23 17:21:35 charge voltages: [228 231 237]V
[lp-1 ] DEBUG 2025/05/23 17:21:35 detected connected phases: 3p
[lp-1 ] DEBUG 2025/05/23 17:21:35 charge total import: 34.858kWh
[lp-1 ] DEBUG 2025/05/23 17:21:35 charger status: C
[lp-1 ] DEBUG 2025/05/23 17:21:35 charge start energy: 34.858kWh
[lp-1 ] INFO 2025/05/23 17:21:35 start charging ->
[lp-1 ] DEBUG 2025/05/23 17:21:35 wake-up timer: stop
[db ] TRACE 2025/05/23 17:21:35 UPDATE `sessions` SET `created`="2025-05-22 21:48:17.421",`finished`="2025-05-23 10:25:11.421",`loadpoint`="1. DIREKT LADEN",`identifier`="evcc",`vehicle`="e-GOLF",`odometer`=69291,`meter_start_kwh`=30.145,`meter_end_kwh`=34.851,`charged_kwh`=4.631200000000007,`charge_duration`="1h4m12s",`solar_percentage`=80.62595626973284,`price`=0.5350338203648717,`price_per_kwh`=0.11552811806116577,`co2_per_kwh`=52.65588984764417 WHERE `id` = 84 1 <nil>
[mqtt ] TRACE 2025/05/23 17:21:35 send evcc/site/battery/1/icon: ''
[lp-1 ] DEBUG 2025/05/23 17:21:35 soc estimated: 95.25% (vehicle: 95.00%)
[lp-1 ] DEBUG 2025/05/23 17:21:35 vehicle soc: 95%
[mqtt ] TRACE 2025/05/23 17:21:35 send evcc/site/battery/1/power: '0'
[lp-1 ] DEBUG 2025/05/23 17:21:35 vehicle range: 204km
[lp-1 ] DEBUG 2025/05/23 17:21:35 pv charge current: 9.85A = 16A + -6.15A (2830W @ 2p)
[ocpp ] TRACE 2025/05/23 17:21:35 send 251327: [2,"2691750531","SetChargingProfile",{"connectorId":1,"csChargingProfiles":{"chargingProfileId":10,"stackLevel":3,"chargingProfilePurpose":"TxDefaultProfile","chargingProfileKind":"Absolute","chargingSchedule":{"startSchedule":"2025-05-23T15:20:35Z","chargingRateUnit":"A","chargingSchedulePeriod":[{"startPeriod":0,"limit":6}]}}}]
[ocpp ] TRACE 2025/05/23 17:21:36 recv 251327: [3,"2691750531",{"status": "Accepted"}]
[lp-1 ] DEBUG 2025/05/23 17:21:36 set charge current limit: 6A
[site ] DEBUG 2025/05/23 17:21:36 solar forecast: accumulated 88.086kWh, produced 78.500kWh, scale 0.891
[httpd ] TRACE 2025/05/23 17:21:38 GET /api/auth/status
[httpd ] TRACE 2025/05/23 17:21:38 GET /api/system/log
[httpd ] TRACE 2025/05/23 17:21:38 GET /api/system/log/areas
[ocpp ] TRACE 2025/05/23 17:21:41 send 161183: [2,"2049723131","TriggerMessage",{"requestedMessage":"MeterValues","connectorId":1}]
[ocpp ] TRACE 2025/05/23 17:21:41 recv 161183: [3,"2049723131",{"status": "Accepted"}]
[ocpp ] TRACE 2025/05/23 17:21:41 recv 161183: [2,"1888102948","MeterValues",{"meterValue": [{"timestamp": "2025-05-23T15:21:41Z","sampledValue": [{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Energy.Active.Import.Register","unit": "Wh","value": "1950105.3"},{"format": "Raw","location": "Outlet","context": "Trigger","measurand": "Power.Active.Import","unit": "W","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L1","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L2","measurand": "Current.Import","unit": "A","value": "0.0"},{"format": "Raw","location": "Outlet","context": "Trigger","phase": "L3","measurand": " [... cut ...] ocation": "Outlet","context": "Trigger","phase": "L3-N","measurand": "Voltage","unit": "V","value": "237.0"} ]} ],"connectorId": 1,"transactionId": 1747943065}]
[ocpp ] TRACE 2025/05/23 17:21:42 send 251327: [3,"660707158",{}]
[site ] DEBUG 2025/05/23 17:21:43 ----
[lp-2 ] DEBUG 2025/05/23 17:21:43 charge power: 0W
[lp-2 ] DEBUG 2025/05/23 17:21:43 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/05/23 17:21:43 charge power: 7191W
[lp-1 ] DEBUG 2025/05/23 17:21:43 charge currents: [15.5 15.9 0]A
```

mfuchs1984, 2025-05-23:
Nun, evcc schickt 6 Ampere, die wallbox akzeptiert das Kommando, macht aber mit 16 weiter. Du musst den Fehler wohl in den Einstellungen der Wallbox suchen.

mfuchs1984, 2025-05-23:
Wie gesagt, wallbox akzeptiert aber ignoriert das Limit. Einstellungen der Wallbox prüfen, ggf. factory reset machen, ggf. wallbox-fw5 template probieren.

mfuchs1984, 2025-05-24:
Zurück auf die 5er Firmware ist eigentlich nicht schlimm, dort funktioniert definitiv das wallbox-fw5 template.