evcc-io/evcc #21380: KIA connection not working

stickybyte, 2025-05-22:
### Describe the bug

Trying to get stats of the car in EVCC.

### Steps to reproduce

1. Configure
2. #evcc vehicle

### Configuration details

```yaml
vehicles:
  - name: EV6
    type: template
    template: kia
    user: 'XXXXXXXX' # Benutzerkonto (bspw. E-Mail Adresse, User Id, etc.)
    password: 'YYYYYYYY' # Passwort des Benutzerkontos (bei führenden Nullen bitte in einfache Hochkommata setzen)
    vin: ZZZZZZZZZZZZZZ # Erforderlich, wenn mehrere Fahrzeuge des Herstellers vorhanden sind (optional)
    title: KIA EV6 # Wird in der Benutzeroberfläche angezeigt (optional)
    capacity: 84 # Akkukapazität in kWh (optional)
    language: de # 'de' für Deutsch und 'en' für Englisch [en, de] (optional)
    icon: car # Icon in der Benutzeroberfläche [car, bike, bus, moped, motorcycle, rocket, scooter, taxi, tractor, rickshaw, shuttle, van, airpurifier, battery, bulb, climate, coffeemaker, compute, cooking, cooler, desktop, device, dishwasher, dryer, floorlamp, generic, heater, heatexchange, heatpump, kettle, laundry, laundry2, machine, meter, microwave, pump, smartconsumer, tool, waterheater] (optional)
    phases: 3 # Die maximale Anzahl der Phasen welche genutzt werden können (optional)
    mode: # Möglich sind Off, Now, MinPV und PV, oder leer wenn keiner definiert werden soll (optional)
    minCurrent: 6 # Definiert die minimale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    maxCurrent: 16 # Definiert die maximale Stromstärke pro angeschlossener Phase mit welcher das Fahrzeug geladen werden soll (optional)
    identifiers: # Kann meist erst später eingetragen werden, siehe: https://docs.evcc.io/docs/features/vehicle (optional)
    priority: # Priorität des Ladepunktes oder Fahrzeugs in Relation zu anderen Ladepunkten oder Fahrzeugen für die Zuweisung von PV-Energie (optional)
    cache: 10m # Zeitintervall nach dem Daten erneut vom Fahrzeug abgefragt werden (optional)
```

### Log details

```text
[main  ] ERROR 2025/05/22 12:36:57 creating vehicle EV6 failed: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: Post "https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/token": dial tcp: lookup prd.eu-ccapi.kia.com: i/o timeout
.Soc:     vehicle not available: cannot create vehicle type 'template': cannot create vehicle type 'kia': login failed: Post "https://prd.eu-ccapi.kia.com:8080/api/v1/user/oauth2/token": dial tcp: lookup prd.eu-ccapi.kia.com: i/o timeout
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

stickybyte, 2025-05-22:
> ` dial tcp: lookup prd.eu-ccapi.kia.com: i/o timeout`
> 
> da funktioniert wohl die Namensauflösung nicht.

Doch, die klappt gut in dem Netz. Davor passiert aber schon das login failed ...

RenatusRo, 2025-05-22:
hm, bis zur zeile

`[kia   ] TRACE 2025/05/22 12:36:57 code=OTEWYJA3NDATYWFMOC0ZODHKLTLIZMETZJQ1ODLLOTK2YJFL&grant_type=authorization_code&redirect_uri=https%3A%2F%2Fprd.eu-ccapi.kia.com%3A8080%2Fapi%2Fv1%2Fuser%2Foauth2%2Fredirect
`
sieht das bei mir genauso aus und hat 14:55 noch funktioniert.
und dann folgt bei mir  `{"access_token": ...) `

stickybyte, 2025-05-22:
Könnten Sonderzeichen im Passwort ein Problem sein?

RenatusRo, 2025-05-22:
wuerde denken, dass es bei passwortproblem schon frueher schief gehen sollte.
einfach mal ein zeichen mehr ins passwort und gucken, ob es dann anders failed.