evcc-io/evcc #21382: Octopus-API Tarrif Integration doesn't work

a-teece, 2025-05-22:
### Describe the bug

After the last Octopus-API bug I reported got fixed I've gone to try it again, but still get an error:

ERROR 2025/05/22 13:59:11 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'octopusenergy': Get "": unsupported protocol scheme ""

As this is the second bug which block the Octopus-API template from working at all I'm a little concerned that I'm the only person trying to use it.

### Steps to reproduce

1. Set the tarrif to use the octpus-api template.

### Configuration details

```yaml
tariffs:
  currency: GBP # three letter ISO-4217 currency code (default EUR)
  grid:
    type: template
    template: octopus-api
    apiKey: sk_live_############################
```

### Log details

```text
[octopus] ERROR 2025/05/22 13:59:11 Get "": unsupported protocol scheme ""
[main  ] ERROR 2025/05/22 13:59:11 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'octopusenergy': Get "": unsupported protocol scheme ""
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

duckfullstop, 2025-05-23:
Colour me confused because I tested this working last time. Investigating now...

duckfullstop, 2025-05-23:
I'm unable to replicate this, at least with my API key. Strangely the error you're giving implies that something isn't getting a valid URL which from what I can read in the code path is impossible?

Can you confirm whether this also happens with the `octopus-productcode` template, and if so, what your tariff details are?

a-teece, 2025-05-23:
@duckfullstop Will test with octopus-productcode shortly.
@andig I turned on trace logging and didn't get any further details

andig, 2025-05-23:
@duckfullstop I don't see why the client shouldn't log? @a-teece if @duckfullstop can't reproduce only way I see would be sharing your login for debugging.

duckfullstop, 2025-05-23:
> [@duckfullstop](https://github.com/duckfullstop) I don't see why the client shouldn't log?

To be fair it's lacking trace logging. I can look into adding detailed trace logging to a branch that they can use, or if @a-teece is happy to share their key with me I can try and fix the issue directly - likely won't be until tomorrow unfortunately.

andig, 2025-05-23:
I'd expect only request logging in trace- it seems you're already setting the logger, but maybe i misread.

a-teece, 2025-05-23:
Using octopus-productcode is working fine.

With trace on the only lines I see related to the octopus item are:
[octopus] TRACE 2025/05/23 17:15:28 GET 
[octopus] ERROR 2025/05/23 17:15:28 Get "": unsupported protocol scheme ""
[main  ] ERROR 2025/05/23 17:15:28 creating tariff grid failed: cannot create tariff type 'template': cannot create tariff type 'octopusenergy': Get "": unsupported protocol scheme ""

a-teece, 2025-05-23:
@duckfullstop I wonder if the problem is because my Octopus account has 2 addresses - my old house still shows even though it was from years ago. I certainly see both at https://octopus.energy/dashboard/new/accounts/personal-details/api-access.

In the source at https://github.com/evcc-io/evcc/blob/ede0663ef02cbdc36b21a42e31d8686f77c51541/tariff/octopus/graphql/api.go#L111 I can see it raises an error that multiple accounts isn't supported.... but if this is the root cause then there must still be a small issue that the error doesn't make it's way into the logs.

andig, 2025-05-25:
@a-teece anything left to do here? @duckfullstop we could just retire the non-productcode usage?

a-teece, 2025-05-25:
@duckfullstop reading the code (not my language), it does look like the error returned by the account code lookup highlighted above, used by the tarrif code method is just dropped and “” is returned for tarrif code.

I suspect this is an issue that my api key returning 2 accounts is not supported, but that the error handling is not working correctly. If that is the case I would suggest the error handling is fixed because there are other errors returned by parts of the code path that become difficult to debug.

duckfullstop, 2025-05-25:
Please let me know in this issue if this still doesn't work for you and I'll work on it as a priority. 🙇