evcc-io/evcc #21390: Ladelog: geladene kWh nicht plausibel

Mananix, 2025-05-22:
### Describe the bug

Im Modus PV werden Ladevorgänge bei zu wenig Überschuss pausiert. EVCC erfasst bei jeder Pause die Zählerstände und die geladenen kWh.

In der Datenbank werden diese pausierten Ladevorgänge allerdings zu einem Ladevorgang zusammengefasst. Ich vermute das EVCC hier dann die Zählerstände in der DB aktualisiert, aber die geladenen kWh addiert werden. Da in den Pausen die Wallbox natürlich einen minimalen Verbrauch hat passt dann jedoch der Wert für geladene kWh nicht zu den Zählerständen.

Logfile
[49686a9f_evcc_2025-05-22T17-35-00.814Z.log](https://github.com/user-attachments/files/20399842/49686a9f_evcc_2025-05-22T17-35-00.814Z.log)

Datenbankeintrag des Ladevorgangs:
[image]

### Steps to reproduce

1. Ladevorgang im Modus PV bei schwankender PV Leistung starten
2. Ladevorgang sollte Pausen über mehrere Minuten haben

### Configuration details

```yaml
EVCC in HA
PV = Huawei mit 2 WR
Wallbox = EVBox Elvi mit Zähler
```

### Log details

```text
Start Ladevorgang:
[lp-1  ] DEBUG 2025/05/22 16:30:17 charge total import: 13872.789kWh
[lp-1  ] DEBUG 2025/05/22 16:30:17 charger status: C
[lp-1  ] DEBUG 2025/05/22 16:30:17 charge start energy: 13872.789kWh
[lp-1  ] INFO 2025/05/22 16:30:17 start charging ->

Pause1:
[lp-1  ] DEBUG 2025/05/22 16:34:17 charge total import: 13873.045kWh
[lp-1  ] DEBUG 2025/05/22 16:34:17 charger status: B
[lp-1  ] DEBUG 2025/05/22 16:34:17 charge final energy: 0.256kWh
[lp-1  ] INFO 2025/05/22 16:34:17 stop charging <-

Ende Pause1
[lp-1  ] DEBUG 2025/05/22 16:38:17 charge total import: 13873.064kWh
[lp-1  ] DEBUG 2025/05/22 16:38:17 charger status: C
[lp-1  ] DEBUG 2025/05/22 16:38:17 charge start energy: 13873.064kWh
[lp-1  ] INFO 2025/05/22 16:38:17 start charging ->

Pause2:
[lp-1  ] DEBUG 2025/05/22 16:41:47 charge total import: 13873.289kWh
[lp-1  ] DEBUG 2025/05/22 16:41:47 charger status: B
[lp-1  ] DEBUG 2025/05/22 16:41:47 charge final energy: 0.481kWh
[lp-1  ] INFO 2025/05/22 16:41:47 stop charging <-

Ende Pause2:
[lp-1  ] DEBUG 2025/05/22 16:58:47 charge total import: 13873.310kWh
[lp-1  ] DEBUG 2025/05/22 16:58:47 charger status: C
[lp-1  ] DEBUG 2025/05/22 16:58:47 charge start energy: 13873.310kWh
[lp-1  ] INFO 2025/05/22 16:58:47 start charging ->

Pause3:
[lp-1  ] DEBUG 2025/05/22 17:24:17 charge total import: 13875.427kWh
[lp-1  ] DEBUG 2025/05/22 17:24:17 charger status: B
[lp-1  ] DEBUG 2025/05/22 17:24:17 charge final energy: 2.598kWh
[lp-1  ] INFO 2025/05/22 17:24:17 stop charging <-

Ende Pause3:
[lp-1  ] DEBUG 2025/05/22 17:27:17 charge total import: 13875.447kWh
[lp-1  ] DEBUG 2025/05/22 17:27:17 charger status: C
[lp-1  ] DEBUG 2025/05/22 17:27:17 charge start energy: 13875.447kWh
[lp-1  ] INFO 2025/05/22 17:27:17 start charging ->

Pause4 bzw. Ende für heute:
[lp-1  ] DEBUG 2025/05/22 17:31:17 charge total import: 13875.710kWh
[lp-1  ] DEBUG 2025/05/22 17:31:17 charger status: B
[lp-1  ] DEBUG 2025/05/22 17:31:17 charge final energy: 2.861kWh
[lp-1  ] INFO 2025/05/22 17:31:17 stop charging <-
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

andig, 2025-05-23:
Welcher Wert konkret passt hier nicht?

Mananix, 2025-05-23:
Die geladenen kWh passen am Ende im Log nicht zu den ebnfalls im Log aufgeführten Zählerständen, da der Verbrauch der WB während der Pausen nicht eingerechnet wird.

andig, 2025-05-23:
Die Unschärfe ist also der Standbyverbrauch?

premultiply, 2025-05-23:
Ich kann dir immer noch nicht folgen.

Es sollte die Gesamtenergie zwischen Anstecken und Abstecken einem Ladevorgang zugeordnet werden.

Tut es das nicht?

Wenn der Ladezähler den Eigenverbrauch der Wallbox mitzählt sitzt er ggf. an der falschen Stelle, die Box ist falsch verdrahtet oder es ist schlichtweg im Einzelfall nicht anders machbar (z.B. externer Zähler vor der Box).

Mananix, 2025-05-24:
Du sagst es ja selbst. Gesamtenergie zwischen anstecken und abstecken. 

Bei PV Überschuss wird halt pausiert, was ja kein abstecken ist. Entsprechend erfasst evcc hier auch nur den Anfangszählerstand und den beim abstecken im Log. 

Die Modi Min+PV und schnell haben keine Pausen und daher auch das Problem nicht. 

Ob irgendeine wallbox mit Zähler auf ihrem Zähler den Eigenverbrauch nicht mit erfasst würde ich auch anzweifeln, da es ja bei separat angebrachtem Zähler ebenfalls erfasst wird.

andig, 2025-05-24:
Gerne PR

Mananix, 2025-05-24:
Würde ich gern, aber beim programmieren hört es bei mir dann leider auf. 😢

premultiply, 2025-05-24:
Ok, dann machen wir hier zu.

Funktioniert offensichtlich ja genau so wie es soll.

closed 2025-05-24 by premultiply