evcc-io/evcc #21391: vehicle soc: Message: validation --- error with Polestar

njglauto, 2025-05-22:
### Describe the bug

The car is not sent the SOC data to charger / evcc.
System config is with a Polestar [car] and two Easee home box [Charger] in the system.  
The EVCC is version 0.203.6 (lower version x.3 and x.5 have the same failures) 
the config was not modified. Older configs backups are tested and produce the same failures. 

Since 20. May the following warnings are up. 

[lp-2  ] WARN 2025/05/22 21:20:47 charger logic error: disabled but charging
[lp-2  ] ERROR 2025/05/22 21:20:55 vehicle odometer: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-2  ] ERROR 2025/05/22 21:20:55 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []

Same error is up on the 1st Charger with the same car. 

### Steps to reproduce

When the car is connecting, failure will come up.

### Configuration details

```yaml
site:
  title: NJ-Home - Schweriner Straße 
  meters:
    grid: NJ-Grid
    pv:
    - NJ-SMA-Hybrid
    battery:
    - NJ-SMA-Home_Storage

meters:
- type: template
  template: sma-home-manager
  usage: grid
  host: ####IP-adresse
  name: NJ-Grid
- type: template
  template: sma-hybrid
  id: x
  host: ####IP-adresse
  port: xxx
  usage: pv
  modbus: tcpip
  name: NJ-SMA-Hybrid
- type: template
  template: sma-hybrid
  id: x
  host: ####IP-adresse
  port: xxx
  usage: battery
  modbus: tcpip
  name: NJ-SMA-Home_Storage

vehicles:
- name: Polestar
  type: template
  template: polestar
  user: #################
  password: #################
  vin: #################
  title: Polestar
  capacity: 78 
  icon: rocket
  phases: 3 
  mode: PV 
  minCurrent: 6 
  maxCurrent: 16 
  identifiers:
  priority: 
  cache: 15m 

chargers:
- type: template
  template: easee
  user: ###################
  password: ###################
  charger: #################
  timeout: 20s
  authorize:
  name: wallbox_black
- type: template
  template: easee
  user: ###################
  password: ###################
  charger: ###################
  timeout: 20s
  authorize:
  name: wallbox_white

loadpoints:
- title: Auffahrt
  charger: wallbox_white
  vehicle: Polestar
  mode: pv
  enable:
    delay: 2m
    threshold: -1250
  disable:
    delay: 10m
    threshold: 1250
- title: Garage
  charger: wallbox_black
  vehicle:
  mode: off
  enable:
    delay: 2m
    threshold: -1250
  disable:
    delay: 10m
    threshold: 1250
```

### Log details

```text
[lp-2  ] WARN 2025/05/22 21:20:47 charger logic error: disabled but charging
[lp-2  ] ERROR 2025/05/22 21:20:55 vehicle odometer: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-2  ] ERROR 2025/05/22 21:20:55 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-2  ] ERROR 2025/05/22 21:21:17 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-2  ] ERROR 2025/05/22 21:21:17 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-1  ] ERROR 2025/05/22 21:26:09 vehicle odometer: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-1  ] ERROR 2025/05/22 21:26:09 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-1  ] WARN 2025/05/22 21:26:09 charger logic error: disabled but charging
[lp-1  ] ERROR 2025/05/22 21:26:38 vehicle odometer: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
[lp-1  ] ERROR 2025/05/22 21:26:38 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

hjwsg, 2025-05-23:
Same for me with keba wallbox. and 0.203.6.

loebse, 2025-05-23:
I can confirm that Polestar changed the API again, I will have a look on the weekend.

loebse, 2025-05-23:
They are not using a different API they have just been faster to adopt to the change that Polestar did 😉

loebse, 2025-05-23:
> > I can confirm that Polestar changed the API again, I will have a look on the weekend.
> 
> As a Polestar 2 user: Do you need something from Amazon or something like that? Want to give you something back for your fun with the Polestar API Engineers 😁

A 2nd Polestar to test 😉 Jokes aside, this is just my hobby

loebse, 2025-05-23:
Issue is fixed an PR created. Major change:

Polestar changed their API to CarTelematicsV2

This new endpoint requires an array of VINs instead of a single VIN as before. As a consequence it also returns an array of Health, Battery, and Odometer data.

With this PR I am adopting to the new API assuming that we always have one VIN only (array with one entry). Probably this can be changed in the future but for now I assume even if you configure multiple Polestars in the vehicle section, they would be queried separately so I guess this is ok.

closed 2025-05-23 by andig

njglauto, 2025-05-24:
It is tested with the actual nightly and the Polestar SOC will be displayed. "Field test" successful 👍

wlanrouter, 2025-06-01:
Thank you so much @loebse