evcc-io/evcc #21392: Not able to add Charge Amps Luna wallbox to evcc

DonPalme, 2025-05-22:
### Describe the bug

I am unable to add Charge Amps Luna wallbox to evcc. It is configured in the installer app to ws://<my-evcc-ip>:8887 with station ID 020100001586L and it shows "connected"

As you can see in the log, there is also valid communication, but the charger will not appear in my webui of evcc.

The error messages that stand out in the log are:
```
[ocpp  ] TRACE 2025/05/22 23:46:31 send 020100001586L: [2,"3136633128","TriggerMessage",{"requestedMessage":"BootNotification"}]
[ocpp  ] TRACE 2025/05/22 23:46:31 recv 020100001586L: [3,"3136633128",{"status":"Rejected"}]
[020100001586L-1] DEBUG 2025/05/22 23:46:31 failed triggering BootNotification: Rejected
[020100001586L-1] DEBUG 2025/05/22 23:47:01 BootNotification timeout
```

and:
```
[ocpp  ] TRACE 2025/05/22 23:47:02 send 020100001586L: [2,"2449536744","ChangeConfiguration",{"key":"WebSocketPingInterval","value":"30"}]
[ocpp  ] TRACE 2025/05/22 23:47:02 recv 020100001586L: [3,"2449536744",{"status":"RebootRequired"}]
[020100001586L-1] DEBUG 2025/05/22 23:47:02 failed configuring WebSocketPingInterval: RebootRequired
```

Anyone has an idea how to solve this?

### Steps to reproduce

1. configure wallbox to ws://<my-evcc-ip>:8887 with station <my-station-id>
2. add configuration to evcc.yaml
3. start evcc docker
...

### Configuration details

```yaml
chargers:
  - name: wallbox5
    type: template
    template: ocpp
    stationid: 020100001586L
    connector: 1
```

### Log details

```text
[main  ] INFO 2025/05/22 23:46:30 evcc 0.203.6 (c7898f06)
[main  ] INFO 2025/05/22 23:46:30 using config file: /etc/evcc.yaml
[db    ] INFO 2025/05/22 23:46:30 using sqlite database: /root/.evcc/evcc.db
[db    ] TRACE 2025/05/22 23:46:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="settings" -1 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "settings" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT * FROM `settings` LIMIT 1 -1 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT * FROM `settings` 2 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT count(*) FROM sqlite_master WHERE type='table' AND name="configs" -1 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT sql FROM sqlite_master WHERE type IN ("table","index") AND tbl_name = "configs" AND sql IS NOT NULL order by type = "table" desc 1 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT * FROM `configs` LIMIT 1 -1 <nil>
[main  ] INFO 2025/05/22 23:46:30 listening at :7070
[db    ] TRACE 2025/05/22 23:46:30 SELECT * FROM `configs` WHERE `configs`.`class` = 5 0 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT * FROM `configs` WHERE `configs`.`class` = 2 0 <nil>
[db    ] TRACE 2025/05/22 23:46:30 SELECT * FROM `configs` WHERE `configs`.`class` = 1 0 <nil>
[020100001586L-1] DEBUG 2025/05/22 23:46:30 waiting for chargepoint: 5m0s
[ocpp  ] DEBUG 2025/05/22 23:46:31 charge point connected: 020100001586L
[ocpp  ] TRACE 2025/05/22 23:46:31 send 020100001586L: [2,"769791294","ChangeAvailability",{"connectorId":0,"type":"Operative"}]
[ocpp  ] TRACE 2025/05/22 23:46:31 recv 020100001586L: [3,"769791294",{"status":"Accepted"}]
[ocpp  ] TRACE 2025/05/22 23:46:31 recv 020100001586L: [2,"209eafee-fc32-4c3c-9462-6b01afbcc17b","StatusNotification",{"connectorId":1,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2025/05/22 23:46:31 send 020100001586L: [3,"209eafee-fc32-4c3c-9462-6b01afbcc17b",{}]
[ocpp  ] TRACE 2025/05/22 23:46:31 send 020100001586L: [2,"1089642037","GetConfiguration",{}]
[ocpp  ] TRACE 2025/05/22 23:46:31 recv 020100001586L: [2,"dc1bb712-bc8c-42b5-bee1-20588cfa15da","StatusNotification",{"connectorId":0,"errorCode":"NoError","status":"Available"}]
[ocpp  ] TRACE 2025/05/22 23:46:31 send 020100001586L: [3,"dc1bb712-bc8c-42b5-bee1-20588cfa15da",{}]
[ocpp  ] TRACE 2025/05/22 23:46:31 recv 020100001586L: [3,"1089642037",{"configurationKey":[{"key":"ACPhaseSwitchingSupported","readonly":true,"value":"false"},{"key":"AllowOfflineTxForUnknownId","readonly":false,"value":"false"},{"key":"AssignedCurrent","readonly":false,"value":"1.16"},{"key":"AssignedPhase","readonly":false,"value":"1.L123"},{"key":"AuthorizationCacheEnabled","readonly":false,"value":"true"},{"key":"AuthorizationKey","readonly":false,"value":""},{"key":"AuthorizeRemoteTxRequests","readonly":false,"value":"false"},{"key":"CableLock","readonly":false,"value":"true"},{"key":"ChargePointID","readonly":false,"value":""},{"key":"ChargeProfileMaxStackLevel","readonly":true,"value":"32"},{"key":"ChargingScheduleAllowedChargingRateUnit","readonly":true,"value":"Current,Power"},{"key":"ChargingScheduleMaxPeriods","readonly":true,"value":"100"},{"key":"ClockAlignedDataInterval","readonly":false,"value":"0"},{"key":"ConnectedWifiSSID","readonly":true,"value":"Kame"},{"key":"ConnectionTimeOut","readonly":false,"value":"30"},{"key":"ConnectorPhaseRotation","readonly":false,"value":"0.RST"},{"key":"ConnectorPhaseRotationMaxLength","readonly":true,"value":"1"},{"key":"ConnectorSwitch3to1PhaseSupported","readonly":true,"value":"true"},{"key":"EarthingSystem","readonly":false,"value":"TN"},{"key":"FirmwareVersion","readonly":true,"value":"1.7.1"},{"key":"FreeCharging","readonly":false,"value":"true"},{"key":"GetConfigurationMaxKeys","readonly":true,"value":"128"},{"key":"HardwareVersion","readonly":true,"value":"Main board,5.1,1424190128,Processor Board,4.0,1422940817,MID module, ,"},{"key":"HeartbeatInterval","readonly":false,"value":"60"},{"key":"HlSuperv","readonly":false,"value":"7200"},{"key":"InstallationChange","readonly":false,"value":""},{"key":"InstallationCurrent","readonly":false,"value":"0.16"},{"key":"InstallationOfflineCurrent","readonly":false,"value":"1.16"},{"key":"InstallationOfflinePhase","readonly":false,"value":"1.L123"},{"key":"InstallationPhases","readonly":false,"value":"3"},{"key":"LightIntensity","readonly":false,"value":"100"},{"key":"LocalAuthListEnabled","readonly":false,"value":"false"},{"key":"LocalAuthListMaxLength","readonly":true,"value":"200"},{"key":"LocalAuthorizeOffline","readonly":false,"value":"true"},{"key":"LocalPreAuthorize","readonly":false,"value":"false"},{"key":"LoggingStringDebug","readonly":false,"value":""},{"key":"LoggingStringError","readonly":false,"value":"*"},{"key":"LoggingStringInfo","readonly":false,"value":"*"},{"key":"LoggingStringWarning","readonly":false,"value":"*"},{"key":"LteAPN","readonly":false,"value":"internet.telia.iot"},{"key":"MaxChargingProfilesInstalled","readonly":true,"value":"6400"},{"key":"MaxCurrent","readonly":true,"value":"0.32,1.32"},{"key":"MaxPhases","readonly":true,"value":"3"},{"key":"MeterValueSampleInterval","readonly":false,"value":"10"},{"key":"MeterValuesAlignedData","readonly":false,"value":""},{"key":"MeterValuesAlignedDataMaxLength","readonly":true,"value":"1"},{"key":"MeterValuesSampledData","readonly":false,"value":"Energy.Active.Import.Register,Current.Offered,Power.Active.Import,Current.Import,Voltage"},{"key":"MeterValuesSampledDataMaxLength","readonly":true,"value":"10"},{"key":"NumberOfConnectors","readonly":true,"value":"1"},{"key":"PublicKey","readonly":true,"value":""},{"key":"ResetRetries","readonly":false,"value":"1"},{"key":"RetryBackOffRandomRange","readonly":false,"value":"60"},{"key":"RetryBackOffRepeatTimes","readonly":false,"value":"5"},{"key":"RetryBackOffWaitMinimum","readonly":false,"value":"30"},{"key":"RfidTagFreeCharging","readonly":false,"value":"00000000000000"},{"key":"SendLocalListMaxLength","readonly":true,"value":"200"},{"key":"Server","readonly":false,"value":"ws://192.168.178.13:8887"},{"key":"ServoRecalibration","readonly":false,"value":"false"},{"key":"StopTransactionOnInvalidId","readonly":false,"value":"false"},{"key":"StopTxnAlignedData","readonly":false,"value":""},{"key":"StopTxnAlignedDataMaxLength","readonly":true,"value":"0"},{"key":"StopTxnSampledData","readonly":false,"value":""},{"key":"StopTxnSampledDataDataMaxLength","readonly":true,"value":"0"},{"key":"SupportedFeatureProfiles","readonly":true,"value":"Core,FirmwareManagement,SmartCharging,LocalAuthListManagement,RemoteTrigger"},{"key":"TimeZone","readonly":false,"value":"UTC\n"},{"key":"TransactionMessageAttempts","readonly":false,"value":"3"},{"key":"TransactionMessageRetryInterval","readonly":false,"value":"60"},{"key":"UnlockConnectorOnEVSideDisconnect","readonly":false,"value":"true"},{"key":"UserCurrentLimit","readonly":false,"value":"1.16"},{"key":"WebS [... cut ...]
[pv2   ] TRACE 2025/05/22 23:47:54 modbus: recv 00 02 00 00 00 07 01 04 04 07 ea 00 03
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.6 (c7898f06)

premultiply, 2025-05-22:
Missing config.
Charger connected to a loadpoint?

closed 2025-05-22 by premultiply

DonPalme, 2025-05-23:
> Missing config. Charger connected to a loadpoint?

That was the issue, sorry I  am new to evcc. Now it is configured and running well! Thanks a lot!