evcc-io/evcc #21409: Publish `chargeVoltages` on API even on loadpoints with phaseswitching

deadrabbit87, 2025-05-23:
**Is your feature request related to a problem? Please describe.**
At the moment `chargeVoltages` are only published, if the loadpoint has no phase switching. 

In my point of view, this is caused by this: 
https://github.com/evcc-io/evcc/blob/ede0663ef02cbdc36b21a42e31d8686f77c51541/core/loadpoint.go#L1561-L1564

**Describe the solution you'd like**
Expose `chargeVoltages` even if the loadpoint has phase switching.

andig, 2025-05-23:
What's the use case?

deadrabbit87, 2025-05-24:
@iseeberg79 and i checked this: 

If we move this after
`chargeVoltages := []float64{u1, u2, u3}`

the voltages are published on the API. 

Should we provide a pr?