evcc-io/evcc #21416: Stündliche Tarife

andig, 2025-05-24:
### Discussed in https://github.com/evcc-io/evcc/discussions/21398

<div type='discussions-op-text'>

<sup>Originally posted by **Geisti01** May 23, 2025</sup>
Ich habe das Phänomen entdeckt, dass bei mehreren Stundeneinträgen Zeiten generiert werden, welche unnötig sind.
z.B.: 5:00 - 5:00
Habe bereits probiert, was passiert, wenn ich (beim preis für 5-6) hours lediglich "5" eintrage, aber da hat der Parser ganz offensichtlich auch keinen Bock drauf ;) .

[image]
[image]

Ist das ein Fehler meinerseits, weil ich die Konfig anders hätte schreiben müssen, ist das ein bekannter Bug ohne Auswirkungen oder gibt es hier tatsächlich etwas zu? Die Dokumentation erscheint bei den Möglichkeiten und Schreibweisen für meine Begriffe leider etwas dürftig.</div>

Wisley7, 2025-05-25:
Ich glaube das hatte ich auch mal und konnte das lösen, indem die Zeiten nicht überlappend waren. 
0-4
5-5
6-14
Aber bin seit einiger Zeit bei Tibber, und hab die alte config nicht mehr.

andig, 2025-05-25:
Das ist tatsächlich ein Bug, die Angabe doppelter Zeiten für Ende/Start muss möglich sein.

closed 2025-05-25 by andig