evcc-io/evcc #21419: SGReady Charger Does Not Send MQTT

hinge83, 2025-05-24:
### Describe the bug

When configuring a charger with type: sgready using MQTT control, EVCC does not publish the expected MQTT message during PV-based activation. The log reports:
`[lp-2 ] ERROR 2025/05/24 09:23:29 charger enable: not available`

### Steps to reproduce

1. Configure a charger in evcc.yaml with type: sgready and use value: false or true.
2. Start EVCC.
3. Activate PV-based control mode.
4. Observe that MQTT topic is not published and log shows a parsing error.

### Configuration details

```yaml
- name: wp_pool
  type: sgready
  setmode:
    source: switch
    switch:
    - case: 1
      set:
        source: const
        value: false
        set:
          source: mqtt
          topic: evcc/heatpump/sw
    - case: 2 #off
      set:
        source: const
        value: true
        set:
          source: mqtt
          topic: evcc/heatpump/sw
    - case: 3 #on
      set:
        source: error
        error: ErrNotAvailable
```

### Log details

```text
[lp-2 ] ERROR 2025/05/24 09:40:22 charger enable: not available
[site ] DEBUG 2025/05/24 09:40:22 solar forecast: accumulated 0.126kWh, produced 0.100kWh, scale 0.793
[lp-2 ] DEBUG 2025/05/24 09:40:34 set charge mode: off
[site ] DEBUG 2025/05/24 09:40:34 ----
[lp-1 ] DEBUG 2025/05/24 09:40:34 charge power: 0W
[lp-1 ] DEBUG 2025/05/24 09:40:34 charge currents: [0 0 0]A
[lp-1 ] DEBUG 2025/05/24 09:40:35 set charge mode: pv
[lp-2 ] ERROR 2025/05/24 09:40:43 charge power: outdated
[circuit-main] DEBUG 2025/05/24 09:40:43 power: -1339W
[circuit-main] DEBUG 2025/05/24 09:40:43 current: -1.69A
[site ] DEBUG 2025/05/24 09:40:43 grid power: -1339W
[site ] DEBUG 2025/05/24 09:40:43 pv 1 power: 1834W
[site ] DEBUG 2025/05/24 09:40:43 battery 1 power: 0W
[site ] DEBUG 2025/05/24 09:40:43 grid currents: [-1.69 -1.93 -2.33]A
[site ] DEBUG 2025/05/24 09:40:43 battery 1 soc: 28%
[site ] DEBUG 2025/05/24 09:40:43 site power: -1239W
[lp-2 ] DEBUG 2025/05/24 09:40:43 charger status: B
[site ] DEBUG 2025/05/24 09:40:43 solar forecast: accumulated 0.151kWh, produced 0.100kWh, scale 0.664
[site ] DEBUG 2025/05/24 09:40:43 ----
[lp-1 ] DEBUG 2025/05/24 09:40:43 charge power: 0W
[lp-1 ] DEBUG 2025/05/24 09:40:43 charge currents: [0 0 0]A
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

hinge83, 2025-05-24:
Hi andig
Thank you for quick reply
Can you help me a little bit more?
What do you mean by mode Numbers? Do you refer to case number?

andig, 2025-05-24:
https://github.com/evcc-io/evcc/pull/20747

hinge83, 2025-05-24:
Thanks. worked right away.