evcc-io/evcc #21421: Polestar 2 Abfrage des SOC des Fahrzeuges scheitert

maed72, 2025-05-24:
### Describe the bug

Die Verbindung zwischen EVCC und Polestar funktioniert nicht mehr.  Die Konfiguration von EVCC wurde im aktuellen Zeitraum nicht verändert. Der Fehler tritt seit 2 Tagen dauerhaft auf.

### Steps to reproduce

1. Fahrzeug anschließen
2. Ladung freigeben
3.
...

### Configuration details

```yaml
vehicles:
- type: template
  template: polestar
  user: *********
  password: *******
  vin: *************
  title: PS2.S
  icon: car
  capacity: 75
  mode: pv
  name: ev4
```

### Log details

```text
[lp-1 ] ERROR 2025/05/24 10:29:27 vehicle soc: Message: Validation error of type FieldUndefined: Field 'carTelematics' in type 'Query' is undefined @ 'carTelematics', Locations: [{Line:1 Column:35}], Extensions: map[], Path: []
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.203.6

StefanSchoof, 2025-05-24:
> I have verified that the issue is reproducible with the latest nightly build

Hast du? Das würde heißen https://github.com/evcc-io/evcc/pull/21413 funktioniert nicht

maed72, 2025-05-24:
Tatsächlich kann ich das nicht sagen, da der der Dockercontainer auf der Synology die Installation der nightly Version nicht zulässt. Das Häkchen wird aber zum senden gefordert. #21413   war mir bis zu deinem Kommentar nicht bekannt. Prüfe immer ob das Problem bekannt ist, habe diesmal aber nicht gefunden.

closed 2025-05-24 by andig

maed72, 2025-05-24:
Danke für den nochmaligen Hinweis. Es lag nicht in meiner Absicht unnötige Arbeit zu verursachen. Bedeutet es im Umkehrschluss niemand ein Problem melden kann, der die nightly Version nicht getestet hat?--Gesendet mit der 1&1 Mail App

Das Feld ist nicht ohne Grund da...

andig, 2025-05-24:
Das wäre sinnvoll. Warum Probleme melden, die schon keine mehr sind?