evcc-io/evcc #21422: Victron solar forecast does update

endurance1968, 2025-05-24:
### Describe the bug

In https://github.com/evcc-io/evcc/pull/21058 a new VRM based forecase was introduced. It works well after EVCC start but does not update data. Means after the initial 2-3 days initially retrieved the forecast is always 0.

Dont see any errors related to this in the logs

### Steps to reproduce

1. User the template solar:
    type: template
    template: victron
2. restart EVCC
3. Wait two days
...

### Configuration details

```yaml
solar:
    type: template
    template: victron
    idsite: <id>
    token: <token>
    interval: 1h
```

### Log details

```text
May 24 10:39:17 lx-evcc systemd[1]: Started evcc.service - evcc.
May 24 10:39:17 lx-evcc evcc[130988]: [main  ] INFO 2025/05/24 10:39:17 evcc 0.203.6
May 24 10:39:17 lx-evcc evcc[130988]: [main  ] INFO 2025/05/24 10:39:17 using config file: /etc/evcc.yaml
May 24 10:39:17 lx-evcc evcc[130988]: [db    ] INFO 2025/05/24 10:39:17 using sqlite database: /var/lib/evcc/evcc.db
May 24 10:39:18 lx-evcc evcc[130988]: [mqtt  ] INFO 2025/05/24 10:39:18 connecting evcc at tcp://192.168.1.113:1883
May 24 10:39:18 lx-evcc evcc[130988]: [eebus ] INFO 2025/05/24 10:39:18 Local SKI: 
May 24 10:39:18 lx-evcc evcc[130988]: [eebus ] DEBUG 2025/05/24 10:39:18 starting websocket server on :4712
May 24 10:39:18 lx-evcc evcc[130988]: [eebus ] DEBUG 2025/05/24 10:39:18 mdns: announce
May 24 10:39:18 lx-evcc evcc[130988]: [eebus ] DEBUG 2025/05/24 10:39:18 mdns: using zeroconf
May 24 10:39:18 lx-evcc evcc[130988]: [main  ] INFO 2025/05/24 10:39:18 listening at :7070
May 24 10:39:18 lx-evcc evcc[130988]: [bmw   ] DEBUG 2025/05/24 10:39:18 identity.Login - database token found
May 24 10:39:18 lx-evcc evcc[130988]: [bmw   ] TRACE 2025/05/24 10:39:18 
May 24 10:39:20 lx-evcc evcc[130988]: [gsi   ] ERROR 2025/05/24 10:39:20 unexpected status: 503 (Service Unavailable)
May 24 10:39:20 lx-evcc evcc[130988]: [main  ] ERROR 2025/05/24 10:39:20 creating tariff co2 failed: cannot create tariff type 'grünstromindex': unexpected status: 503 (Service Unavailable)
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] WARN 2025/05/24 10:39:31 vehicle '-3' does not support automatic detection
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] WARN 2025/05/24 10:39:31 vehicle '' does not support automatic detection
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31 site config:
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31   meters:      grid ✓ pv ✓ battery ✓
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     grid:      power ✓ energy ✓ currents ✓
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     pv 1:      power ✓ energy ✗ currents ✗
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✗
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31   vehicles:
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     vehicle 2: range ✓ finish ✗ status ✓ climate ✓ wakeup ✓
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     vehicle 3: range ✓ finish ✗ status ✗ climate ✗ wakeup ✗
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31   tariffs:
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     grid:      priceforecast
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     feed-in:   pricestatic
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     co2:       ✓
May 24 10:39:31 lx-evcc evcc[130988]: [site  ] INFO 2025/05/24 10:39:31     solar:     ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-1  ] INFO 2025/05/24 10:39:31 loadpoint 1:
May 24 10:39:31 lx-evcc evcc[130988]: [lp-1  ] INFO 2025/05/24 10:39:31   mode:        minpv
May 24 10:39:31 lx-evcc evcc[130988]: [lp-1  ] INFO 2025/05/24 10:39:31   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
May 24 10:39:31 lx-evcc evcc[130988]: [lp-1  ] INFO 2025/05/24 10:39:31   meters:      charge ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-1  ] INFO 2025/05/24 10:39:31     charge:    power ✓ energy ✓ currents ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-2  ] INFO 2025/05/24 10:39:31 loadpoint 2:
May 24 10:39:31 lx-evcc evcc[130988]: [lp-2  ] INFO 2025/05/24 10:39:31   mode:        pv
May 24 10:39:31 lx-evcc evcc[130988]: [lp-2  ] INFO 2025/05/24 10:39:31   charger:     power ✓ energy ✓ currents ✓ phases ✓ wakeup ✗
May 24 10:39:31 lx-evcc evcc[130988]: [lp-2  ] INFO 2025/05/24 10:39:31   meters:      charge ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-2  ] INFO 2025/05/24 10:39:31     charge:    power ✓ energy ✓ currents ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-3  ] INFO 2025/05/24 10:39:31 loadpoint 3:
May 24 10:39:31 lx-evcc evcc[130988]: [lp-3  ] INFO 2025/05/24 10:39:31   mode:        minpv
May 24 10:39:31 lx-evcc evcc[130988]: [lp-3  ] INFO 2025/05/24 10:39:31   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
May 24 10:39:31 lx-evcc evcc[130988]: [lp-3  ] INFO 2025/05/24 10:39:31   meters:      charge ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-3  ] INFO 2025/05/24 10:39:31     charge:    power ✓ energy ✓ currents ✓
May 24 10:39:31 lx-evcc evcc[130988]: [lp-1  ] INFO 2025/05/24 10:39:31 vehicle updated: unknown -> 
May 24 10:39:32 lx-evcc evcc[130988]: [lp-2  ] INFO 2025/05/24 10:39:32 vehicle updated: unknown -> 
May 24 10:39:32 lx-evcc evcc[130988]: [lp-3  ] INFO 2025/05/24 10:39:32 vehicle updated: unknown --3
May 24 10:39:32 lx-evcc evcc[130988]: [circuit-waermepumpe-stromkreis] TRACE 2025/05/24 10:39:32 validate current: 0A + (0A -> 11A) <= 16A ok
May 24 10:39:32 lx-evcc evcc[130988]: [circuit-haus-verteiler-stromkreis] TRACE 2025/05/24 10:39:32 validate current: 0A + (0A -> 11A) <= 32A ok
May 24 10:39:32 lx-evcc evcc[130988]: [circuit-main] TRACE 2025/05/24 10:39:32 validate current: 0A + (0A -> 11A) <= 34A ok
May 24 10:39:32 lx-evcc evcc[130988]: [circuit-waermepumpe-stromkreis] TRACE 2025/05/24 10:39:32 validate power: 0W + (0W -> 7590W) <= 11520W ok
May 24 10:39:32 lx-evcc evcc[130988]: [circuit-haus-verteiler-stromkreis] TRACE 2025/05/24 10:39:32 validate power: 0W + (0W -> 7590W) <= 20000W ok
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

evcc version 0.203.6

endurance1968, 2025-05-24:
What do I need to enable to see? Debugs for all several days will create a monster file.
With Info I dit not see any VRM victron solar output except that it is enabled at the start.

endurance1968, 2025-05-25:
Looks like start end is constant after restart. It should increase by an hour after each hourly request.

`
May 24 18:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/24 18:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 24 19:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/24 19:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 24 20:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/24 20:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 24 21:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/24 21:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 24 22:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/24 22:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 24 23:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/24 23:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 00:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 00:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 01:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 01:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 02:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 02:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 03:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 03:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 04:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 04:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 05:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 05:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 06:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 06:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 07:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 07:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 08:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 08:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 09:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 09:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 10:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 10:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
May 25 11:55:16 lx-evcc evcc[308834]: [solar ] TRACE 2025/05/25 11:55:16 GET https://vrmapi.victronenergy.com/v2/installations/423190/stats?start=1748105716&end=1748278516&interval=hours&type=forecast
`

jlegen, 2025-05-25:
> Did you try the pr?

Yes, but then ‚make test‘ for me does fail with:
render_testing.go:70: yaml: line 3: did not find expected key

In fact, i tried this imo a bit weired looking double-double bracelets-thing before - since i found that the ‚nordpool‘ tariff template does use that as well.
I then assumed that this might be related to nordpool-template not using ‚http‘ as source, but type ‚go‘. 
Could that be?

jlegen, 2025-05-25:
Ok, fixed the ‚make test‘ - i had to remove the double-quotes from the uri. Now it looks like this:
uri: https://vrmapi.victronenergy.com/v2/installations/{{ .idsite }}/stats?start={{ `{{ now | unixEpoch | int }}` }}&end={{ `{{ dateModify "48h" now | unixEpoch | int }}` }}&interval=hours&type=forecast

It‘s running now, will find out if it fixed the issue in an hour… :)

closed 2025-05-25 by andig