evcc-io/evcc #21459: Ladevorgang bricht immer mal wieder ab und starte dann neu

userwithoutpassword, 2025-05-26:
### Describe the bug

Der Ladevorgang stoppt in meinen Augen ohne ersichtlichen Grund. Nach einer Weile startet er dann einfach wieder erneut um dann wieder abzubrechen

### Steps to reproduce

siehe oben

### Configuration details

```yaml
siehe oben
```

### Log details

```text
[lp-1  ] DEBUG 2025/05/26 14:12:10 vehicle range: 248km
[lp-1  ] DEBUG 2025/05/26 14:12:10 pv charge current: 5.62A = 6A + -0.384A (88W @ 1p)
[lp-1  ] DEBUG 2025/05/26 14:12:11 set charge current limit: 5A
[site  ] DEBUG 2025/05/26 14:12:11 solar forecast: accumulated 228.888kWh, produced 159.860kWh, scale 0.698
[site  ] DEBUG 2025/05/26 14:12:25 ----
[lp-1  ] DEBUG 2025/05/26 14:12:25 charge power: 0W
[lp-1  ] DEBUG 2025/05/26 14:12:25 charge currents: [0.008 0.014 0.011]A
[site  ] DEBUG 2025/05/26 14:12:25 grid power: -1423W
[site  ] DEBUG 2025/05/26 14:12:25 pv 1 power: 2104W
[site  ] DEBUG 2025/05/26 14:12:25 battery 1 power: 6W
[site  ] DEBUG 2025/05/26 14:12:25 battery 1 soc: 97%
[site  ] DEBUG 2025/05/26 14:12:25 site power: -1417W
[lp-1  ] DEBUG 2025/05/26 14:12:25 charge total import: 7342.824kWh
[lp-1  ] DEBUG 2025/05/26 14:12:25 charger status: B
[lp-1  ] INFO 2025/05/26 14:12:25 stop charging <-
[lp-1  ] DEBUG 2025/05/26 14:12:25 charge total import: 7342.824kWh
[lp-1  ] DEBUG 2025/05/26 14:12:25 soc estimated: 59.00% (vehicle: 59.00%)
[lp-1  ] DEBUG 2025/05/26 14:12:25 vehicle soc: 59%
[lp-1  ] DEBUG 2025/05/26 14:12:25 vehicle soc limit: 95%
[lp-1  ] DEBUG 2025/05/26 14:12:25 vehicle range: 248km
[lp-1  ] DEBUG 2025/05/26 14:12:25 pv charge current: 6.16A = 0A + 6.16A (-1417W @ 1p)
[lp-1  ] DEBUG 2025/05/26 14:12:27 set charge current limit: 6A
[site  ] DEBUG 2025/05/26 14:12:27 solar forecast: accumulated 228.902kWh, produced 159.860kWh, scale 0.698
[site  ] DEBUG 2025/05/26 14:12:40 ----
[lp-1  ] DEBUG 2025/05/26 14:12:40 charge power: 0W
[lp-1  ] DEBUG 2025/05/26 14:12:40 charge currents: [0.008 0.014 0.011]A
[site  ] DEBUG 2025/05/26 14:12:40 grid power: -1305W
[site  ] DEBUG 2025/05/26 14:12:40 pv 1 power: 2014W
[site  ] DEBUG 2025/05/26 14:12:40 battery 1 power: 6W
[site  ] DEBUG 2025/05/26 14:12:40 battery 1 soc: 97%
[site  ] DEBUG 2025/05/26 14:12:40 site power: -1299W
[lp-1  ] DEBUG 2025/05/26 14:12:40 charge total import: 7342.824kWh
[lp-1  ] DEBUG 2025/05/26 14:12:40 charger status: B
[lp-1  ] DEBUG 2025/05/26 14:12:40 pv charge current: 5.65A = 0A + 5.65A (-1299W @ 1p)
[lp-1  ] DEBUG 2025/05/26 14:12:42 set charge current limit: 5A
[site  ] DEBUG 2025/05/26 14:12:42 solar forecast: accumulated 228.916kWh, produced 159.860kWh, scale 0.698
[site  ] DEBUG 2025/05/26 14:12:55 ----
[lp-1  ] DEBUG 2025/05/26 14:12:55 charge power: 0W
[lp-1  ] DEBUG 2025/05/26 14:12:55 charge currents: [0.008 0.014 0.011]A
[site  ] DEBUG 2025/05/26 14:12:55 grid power: -1326W
[site  ] DEBUG 2025/05/26 14:12:55 pv 1 power: 2057W
[site  ] DEBUG 2025/05/26 14:12:55 battery 1 power: 6W
[site  ] DEBUG 2025/05/26 14:12:55 battery 1 soc: 97%
[site  ] DEBUG 2025/05/26 14:12:55 site power: -1320W
[lp-1  ] DEBUG 2025/05/26 14:12:55 charge total import: 7342.824kWh
[lp-1  ] DEBUG 2025/05/26 14:12:55 charger status: B
[lp-1  ] DEBUG 2025/05/26 14:12:55 pv charge current: 5.74A = 0A + 5.74A (-1320W @ 1p)
[site  ] DEBUG 2025/05/26 14:12:55 solar forecast: accumulated 228.928kWh, produced 159.860kWh, scale 0.698
[site  ] DEBUG 2025/05/26 14:13:10 ----
[lp-1  ] DEBUG 2025/05/26 14:13:10 charge power: 0W
[lp-1  ] DEBUG 2025/05/26 14:13:10 charge currents: [0.008 0.014 0.011]A
[site  ] DEBUG 2025/05/26 14:13:10 grid power: -1913W
[site  ] DEBUG 2025/05/26 14:13:10 pv 1 power: 2869W
[... 507 lines cut ...]
[site  ] DEBUG 2025/05/26 14:20:25 battery 1 soc: 97%
[site  ] DEBUG 2025/05/26 14:20:25 site power: -1198W
[lp-1  ] DEBUG 2025/05/26 14:20:25 charge total import: 7342.824kWh
[lp-1  ] DEBUG 2025/05/26 14:20:25 charger status: B
[lp-1  ] INFO 2025/05/26 14:20:25 stop charging <-
[lp-1  ] DEBUG 2025/05/26 14:20:25 charge total import: 7342.824kWh
[lp-1  ] DEBUG 2025/05/26 14:20:25 soc estimated: 60.63% (vehicle: 59.00%)
[lp-1  ] DEBUG 2025/05/26 14:20:25 vehicle soc: 61%
[lp-1  ] DEBUG 2025/05/26 14:20:25 vehicle soc limit: 95%
[lp-1  ] DEBUG 2025/05/26 14:20:25 vehicle range: 250km
[lp-1  ] DEBUG 2025/05/26 14:20:25 pv charge current: 5.21A = 0A + 5.21A (-1198W @ 1p)
[site  ] DEBUG 2025/05/26 14:20:25 solar forecast: accumulated 229.326kWh, produced 160.227kWh, scale 0.699
```

### What type of operating system or environment does evcc run on?

Linux

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

userwithoutpassword, 2025-05-26:
Habe ich auch schon. Ändert aber an dem Problem nichts.

userwithoutpassword, 2025-05-26:
okay dann beobachte ich das nochmal. Gibt es eine möglichkeit die Werte unter 6A zu sperren oder auszublenden im GUI wenn die eh nicht funktionieren?

VolkerK62, 2025-05-26:
>Gibt es eine möglichkeit die Werte unter 6A zu sperren oder auszublenden im GUI wenn die eh nicht funktionieren?

ist in Arbeit  https://github.com/evcc-io/evcc/issues/21179

andig, 2025-05-26:
Warum sollte man sowas einstellen wenns ne klare Warnung gibt? Und wenn man schon Warnungen ignoriert, warum dann Issues eröffnen?

closed 2025-05-26 by andig

userwithoutpassword, 2025-05-26:
[image]

Ich muss die Warnung an dieser Stelle im GUI glatt übersehen haben.