evcc-io/evcc #21524: Automatic phase changing for Victron evse

DILLONR123, 2025-05-30:
Currently no support for automatic phase changing for Victron evse

While not officially doccumented by victron. Their evse can be modified slightly to enable automatic phase changing from 1 to 3 phase. This is already supported by the victron software and just needs to be enable by sending a modbus command.
Victron has also stated this feature will be standard in v2 of their evse.

https://communityarchive.victronenergy.com/questions/215013/single-phase-charging-on-three-phase-setup.html

https://community.victronenergy.com/t/charging-an-ev-with-excess-solar-switching-between-1p-and-3p/21181/16

https://communityarchive.victronenergy.com/storage/attachments/evcs-phasenumschaltung.pdf

premultiply, 2025-06-11:
PR welcome.
Should be possible to autodetect the modification / new version by register 5100 == 1.

andig, 2025-06-19:
No furthwr insights,  losing.

closed 2025-06-19 by andig