evcc-io/evcc #21551: evcc crasht nach Update auf 0.204.0

mclane, 2025-05-31:
### Describe the bug

Nach Update auf 0.206.0 crasht evcc beim Neustart:

### Steps to reproduce

1. Update docker container to 0.206.0
2.Restart -> crash
3.
...

### Configuration details

```yaml
network:
  schema: http
  host: evcc.local # .local suffix announces the hostname on MDNS
  port: 7070

influx:
  url: http://ursamajor:9087
  database: evcc-rawdata # InfluxDB v2.x uses term `bucket` but for compatibility still named `database` here
  token: ***
  org: Altenburg

log: debug
levels:
  cache: error
  lp-1: trace
  modbusproxy: info

# unique installation id
plant: ***

interval: 10s # control cycle interval

meters:
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.159
  port: 502
  usage: grid
  modbus: tcpip
  name: grid1
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.159
  port: 502
  usage: pv
  modbus: tcpip
  name: pv2
- type: template
  template: sungrow-hybrid
  id: 1
  host: 192.168.0.159
  port: 502
  usage: battery
  modbus: tcpip
  name: battery3
- type: template
  template: sunspec-inverter
  id: 1
  host: 192.168.0.40
  port: 502
  usage: pv
  modbus: tcpip
  name: pv4
- name: charge
[... 165 lines cut ...]
github.com/spf13/cobra.(*Command).ExecuteC(0x624e4a0)
	github.com/spf13/cobra@v1.9.1/command.go:1148 +0x46f
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:120 +0x1a
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x2d
```

### What type of operating system or environment does evcc run on?

Docker container

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.206.0

mfuchs1984, 2025-05-31:
https://github.com/evcc-io/evcc/discussions/21550#discussioncomment-13327780

andig, 2025-05-31:
Closed in 4d2ba9ba0b93d03072844c4a8377821ba11312c2

closed 2025-05-31 by andig