evcc-io/evcc #21563: EVCC crashes on start up

dentecde, 2025-06-01:
### Describe the bug

Since Update to version 0.204.0 evcc crashes with "panic: runtime error: index out of range [1] with length 1"

### Steps to reproduce

1.
2.
3.
...

### Configuration details

```yaml
# open evcc at http://evcc.local:7070
network:
  # schema is the HTTP schema
  # setting to `https` does not enable https, it only changes the way URLs are generated
  schema: http
  # host is the hostname or IP address
  # if the host name contains a `.local` suffix, the name will be announced on MDNS
  # docker: MDNS announcements don't work. host must be set to the docker host's name.
  host: evcc.local
  # port is the listening port for UI and api
  # evcc will listen on all available interfaces
  port: 7070

# unique installation id
plant: **removed**

interval: 10s # control cycle interval

meters:
- type: template
  template: solarlog 
  usage: pv  
  host: 192.168.172.189  
  port: 502  
  name: HausPV
- type: template
  template: victron-energy 
  usage: battery  
  host: 192.168.172.166  
  port: 502  
  capacity: 10  
  name: battery1
- type: template
  template: victron-energy
  usage: grid
  host: 192.168.172.166 # IP-Adresse oder Hostname
  port: 502
  name: grid1
- type: template
  template: hoymiles-opendtu 
  name: GaragePV
  usage: pv
  host: 192.168.172.182 # IP-Adresse oder Hostname 

  
#chargers:
#  - name: wallbox1
#    type: template
#    template: phoenix-em-eth
#    id: 180
#    host: 192.168.172.84 # Hostname
#    port: 502 # Port 

vehicles:
- type: template
  template: tesla
  title: Model Y
  icon: car
  clientId: ** removed **
  accessToken: ** removed **
  refreshToken: ** removed **
  vin: ** removed **
  capacity: 79
  phases: 3
  mode: pv
  minCurrent: 5
  maxCurrent: 16
  priority: 2
  name: ev2

loadpoints:
- title: Carport
  charger: wallbox1
  mode: pv

site:
  title: Zuhause

  meters:
    grid: grid1
    pv:
    - HausPV
    - GaragePV
    battery:
    - battery1
  residualPower: 100

tariffs:
  currency: EUR # (default EUR)
  grid:
    # static grid price
    type: fixed
    price: 0.394 # [currency]/kWh

  feedin:
    # rate for feeding excess (pv) energy to the grid
    type: fixed
    price: 0.15 # [currency]/kWh
```

### Log details

```text
goroutine 1 [running]:
encoding/binary.bigEndian.Uint16(...)
	encoding/binary/binary.go:156
github.com/volkszaehler/mbmd/encoding.Uint16(...)
	github.com/volkszaehler/mbmd@v0.0.0-20250524200024-23178706aaf1/encoding/number.go:10
github.com/evcc-io/evcc/charger.(*PhoenixEMEth).Enabled(0x2a0be60?)
	github.com/evcc-io/evcc/charger/phoenix-em-eth.go:109 +0x80
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0x400098cd88, {0x3526db8?, 0x4000c59448?}, 0x4000f0ae50?, 0x2?, 0x2?)
	github.com/evcc-io/evcc/core/loadpoint.go:676 +0xe08
github.com/evcc-io/evcc/core.(*Site).Prepare(0x4000c59448, 0x40005b2690, 0x40005f6cb0)
	github.com/evcc-io/evcc/core/site.go:1025 +0x160
github.com/evcc-io/evcc/cmd.runRoot(0x5d19aa0, {0x2a48f83?, 0x7?, 0x2a3c6c8?})
	github.com/evcc-io/evcc/cmd/root.go:320 +0x18f0
github.com/spf13/cobra.(*Command).execute(0x5d19aa0, {0x400004c2e0, 0x2, 0x2})
	github.com/spf13/cobra@v1.9.1/command.go:1019 +0x82c
github.com/spf13/cobra.(*Command).ExecuteC(0x5d19aa0)
	github.com/spf13/cobra@v1.9.1/command.go:1148 +0x384
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:120 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38
Using config file: /config/evcc.yaml
starting evcc: 'EVCC_DATABASE_DSN=/data/evcc.db evcc --config /config/evcc.yaml'
[main  ] INFO 2025/06/01 00:43:30 evcc 0.204.0
[main  ] INFO 2025/06/01 00:43:30 using config file: /config/evcc.yaml
[db    ] INFO 2025/06/01 00:43:30 using sqlite database: /data/evcc.db
[main  ] INFO 2025/06/01 00:43:31 listening at :7070
[main  ] ERROR 2025/06/01 00:43:31 creating vehicle db:8 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': unexpected status: 429 (Too Many Requests)
[main  ] ERROR 2025/06/01 00:43:31 creating vehicle ev1 failed: cannot create vehicle type 'template': cannot create vehicle type 'bmw': unexpected status: 429 (Too Many Requests)
[site  ] WARN 2025/06/01 00:43:32 vehicle 'BMW I3' does not support automatic detection
[site  ] WARN 2025/06/01 00:43:32 vehicle 'MG4' does not support automatic detection
[site  ] WARN 2025/06/01 00:43:32 vehicle 'I3 120Ah' does not support automatic detection
[site  ] INFO 2025/06/01 00:43:32 site config:
[site  ] INFO 2025/06/01 00:43:32   meters:      grid ✓ pv ✓ battery ✓
[site  ] INFO 2025/06/01 00:43:32     grid:      power ✓ energy ✗ currents ✗
[site  ] INFO 2025/06/01 00:43:32     pv 1:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/06/01 00:43:32     pv 2:      power ✓ energy ✓ currents ✗
[site  ] INFO 2025/06/01 00:43:32     battery 1: power ✓ energy ✗ currents ✗ soc ✓ capacity ✓
[site  ] INFO 2025/06/01 00:43:32   vehicles:
[site  ] INFO 2025/06/01 00:43:32     vehicle 1: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/06/01 00:43:32     vehicle 2: range ✓ finish ✓ status ✓ climate ✗ wakeup ✓
[site  ] INFO 2025/06/01 00:43:32     vehicle 3: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/06/01 00:43:32     vehicle 4: range ✗ finish ✗ status ✗ climate ✗ wakeup ✗
[site  ] INFO 2025/06/01 00:43:32   tariffs:
[site  ] INFO 2025/06/01 00:43:32     grid:      pricestatic
[site  ] INFO 2025/06/01 00:43:32     feed-in:   pricestatic
[site  ] INFO 2025/06/01 00:43:32     co2:       ✗
[site  ] INFO 2025/06/01 00:43:32     solar:     ✓
[lp-1  ] INFO 2025/06/01 00:43:32 loadpoint 1:
[lp-1  ] INFO 2025/06/01 00:43:32   mode:        pv
[lp-1  ] INFO 2025/06/01 00:43:32   charger:     power ✗ energy ✗ currents ✗ phases ✗ wakeup ✗
[lp-1  ] INFO 2025/06/01 00:43:32   meters:      charge ✗
panic: runtime error: index out of range [1] with length 1
goroutine 1 [running]:
encoding/binary.bigEndian.Uint16(...)
	encoding/binary/binary.go:156
github.com/volkszaehler/mbmd/encoding.Uint16(...)
	github.com/volkszaehler/mbmd@v0.0.0-20250524200024-23178706aaf1/encoding/number.go:10
github.com/evcc-io/evcc/charger.(*PhoenixEMEth).Enabled(0x2a0be60?)
	github.com/evcc-io/evcc/charger/phoenix-em-eth.go:109 +0x80
github.com/evcc-io/evcc/core.(*Loadpoint).Prepare(0x4000c36488, {0x3526db8?, 0x4000cc4b48?}, 0x4000908c40?, 0x2?, 0x2?)
	github.com/evcc-io/evcc/core/loadpoint.go:676 +0xe08
github.com/evcc-io/evcc/core.(*Site).Prepare(0x4000cc4b48, 0x4000537c00, 0x40005cbab0)
	github.com/evcc-io/evcc/core/site.go:1025 +0x160
github.com/evcc-io/evcc/cmd.runRoot(0x5d19aa0, {0x2a48f83?, 0x7?, 0x2a3c6c8?})
	github.com/evcc-io/evcc/cmd/root.go:320 +0x18f0
github.com/spf13/cobra.(*Command).execute(0x5d19aa0, {0x40001822b0, 0x2, 0x2})
	github.com/spf13/cobra@v1.9.1/command.go:1019 +0x82c
github.com/spf13/cobra.(*Command).ExecuteC(0x5d19aa0)
	github.com/spf13/cobra@v1.9.1/command.go:1148 +0x384
github.com/spf13/cobra.(*Command).Execute(...)
	github.com/spf13/cobra@v1.9.1/command.go:1071
github.com/evcc-io/evcc/cmd.Execute()
	github.com/evcc-io/evcc/cmd/root.go:120 +0x24
main.main()
	github.com/evcc-io/evcc/main.go:44 +0x38
```

### What type of operating system or environment does evcc run on?

HomeAssistant Add-on

### External automation

- [x] I have made sure that no external automation like HomeAssistant or Node-RED is active or accessing any of the mentioned devices when this issue occurs.

### Nightly build

- [x] I have verified that the issue is reproducible with the latest nightly build

### Version

0.204.0

StefanSchoof, 2025-06-01:
> I have verified that the issue is reproducible with the latest nightly build

Did you?

https://github.com/evcc-io/evcc/issues/21551

dentecde, 2025-06-01:
As you can only post the issue by activating this, i did so.
but i have no plan how to install a nightly build in HomeAssistant. But i will try to do this now. Thanks for answering and thanks for fixing the issue !

closed 2025-06-01 by andig